UltimateHackingKeyboard/firmware #269: USB full-speed and "Not enough bandwidth for new device state."

Agrendalath, 2020-01-21:
**Firmware**: 8.7.0, but noticed this on earlier versions as well.

# Context and the problem
I've been using the UHK with Gigabyte Z370P and Intel i5-8400 without any issues. It's also working perfectly fine with ThinkPad W520 and X1C6. Recently I switched my desktop to Ryzen 7 3800x and the keyboard still worked fine via USB hub built in my monitor. However, when I tried connecting it directly to the motherboard (to update the firmware), the keyboard didn't work (the LED screen was working, but the keystrokes were not registered via BIOS). I replugged it, logged into Linux and checked `dmesg` to see "Not enough bandwidth for new device state." after plugging it back directly to motherboard.

# Testing
Since then I've tried the following boards:
- ASUS TUF GAMING X570-PLUS - not working - the manufacturer claims it's working fine when I sent them the board for the warranty claim; I also tested it with "Xbox 360 Wireless Receiver for Windows" (which is also a `full-speed` device) and it seems to be having the same problem,
- Gigabyte X570 AORUS ULTRA - works fine - unfortunately cooling settings are incompatible with my chassis' fan hub, so keeping this board was not an option for me, because it was pretty noisy and the CPU temperatures were high on idle,
- MSI MEG X570 ACE - not working - I'm currently testing this one; UHK worked with the front panel of my chassis (which didn't work for ASUS), but I still cannot update the firmware via it, as it claims to be a USB hub; the board supports "Xbox 360 Wireless Receiver for Windows", though.

I'm going to check one more motherboard - ASRock X570 Phantom Gaming X for testing this and I should have it on Saturday. I've also considered whether the CPU is faulty, but I'm not sure if this could be a case here.

# Conclusions
I'm suspecting that not all manufacturers care enough about proper implementation of `xhci_hcd` support for `full-speed` devices, but I'm definitely not an expert in this matter, so I'm just judging by my observations and some research. It also doesn't seem to be OS-related, as the keyboard doesn't work in BIOS (nor in Windows) as well.

# Logs
My logs from `dmesg` (along with some comments) are here:
https://privatebin.net/?7d14d583cccf58e4#GKFpvLrDL7wz29dZmW8VFwBbo8MFDAwTNnj3upuhNNmL

I included 4 scenarios there:
- plugging UHK to the rear USB (of the motherboard),
- plugging UHK to the USB hub (in the monitor),
- plugging UHK to the front USB panel (USB 3.0),
- booting with UHK plugged to the front USB panel (USB 2.0).

mondalaci, 2020-01-21:
Thank you very much for the fabulously detailed report!

We've received some reports like yours. According to these reports, the USB 3.x ports of some motherboards are affected. This issue can be worked around by connecting the UHK to a USB 2.x port or by using it via a USB hub with a USB 3.x port.

The UHK exposes an unusually sophisticated USB interface, and I think some chipsets / BIOSes can't deal with it. It's questionable whether we can address this issue via a firmware update, but we'll make a try nevertheless because of the reoccuring nature of such reports.

This will likely lead us deep down the rabbit hole, and given that I haven't been able to get a hold of our USB expert for a while, chances are I have to find someone else, so don't expect a follow-up soon.

Agrendalath, 2020-01-21:
> This issue can be worked around by connecting the UHK to a USB 2.x port

I just wanted to clarify this one. The [manual](https://download.msi.com/archive/mnu_exe/mb/M7C35v1.1-EURO.pdf) for MSI MEG X570 ACE (page 12) states that it has USB 2.0 Type-A ports, but they have the same issue as described above. I just checked them and the output from  `dmesg` is the same:
https://privatebin.net/?3459d0e5d7b17df9#2S7q2TAjz81BNphPqyQjXjkXeteqp21cukBsKTcErEmN

mondalaci, 2020-01-21:
@Agrendalath Thanks for the clarification! We'll dig further later.

Jopie01, 2020-02-05:
I have a GigaByte B450 AORUS PRO WIFI which has the same issue. This seems like a (kernel) problem with SuperSpeed USB 3.1. I also got the warnings below, before the `Not enough bandwidth for new device state.`
```
xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
```
You can take a look at https://bugzilla.kernel.org/show_bug.cgi?id=202541 where also a bug is reported with this warnings.
My motherboard has red (USB 3.1 Gen 2 Type-A) and blue (USB 3.1 Gen 1) USB connectors. The red ones are working, the blue ones are not. The red ones comes from the B450 chipset where the blue ones comes from the CPU. My CPU is also an AMD Ryzen (Ryzen 5 3600).

mondalaci, 2020-02-05:
@Jopie01 Thanks for the valuable feedback!

@Agrendalath Can you only reproduce this issue with Linux?

Agrendalath, 2020-02-05:
@mondalaci, I can reproduce this with BIOS and Windows as well.

laxu, 2020-02-19:
Tested my Gigabyte X470 AORUS Gaming 7 Wifi ports. I tested by simply plugging into different ports and seeing if the keyboard is recognized and types stuff in Windows 10.

- Four USB 3.1 gen 1 ports closest to Wifi antenna connectors do not work, UHK is not recognized at all. Other devices like my Focusrite Scarlett audio interface or my Steam Controller receiver work perfectly fine in these ports.
- USB 3.1 gen 2 port (red) works fine.
- Two USB 2.0/1.1 ports port fine.
- Two USB 3.1 gen 1 ports closest to audio I/O ports work fine.

According to the manual the four USB 3.1 gen 1 ports are tied to CPU whereas all the others are tied to the chipset. I don't know how this really works but that seems to be the main difference.

hanwencheng, 2020-04-07:
I have ASUS X570 plus Motherboard with Ryzen chips, unfortunately it does not work on all my USB port, though I have updated the latest chipset driver from ASUS today.

mondalaci, 2020-04-07:
@hanwencheng Very sorry to hear! So far, I think the UHK worked in USB 2.0 ports. Unfortunately, the best workaround I can suggest now is to connect your UHK via a USB hub.

daenney, 2020-04-18:
I have the same issue. To add to the list:

Desktop:
* CPU: AMD Ryzen 7 3700X
* Mobo/chipset: ROG STRIX E GAMING, X570
* USB controller: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller

Laptop 1:
* CPU: AMD Ryzen 7 2700U with Radeon Vega Mobile Gfx
* Mobo/chipset: HP Envy x360
* USB controller: Advanced Micro Devices, Inc. [AMD] Raven USB 3.1

Laptop 2:
* CPU: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
* Mobo/chipset: Lenovo X1 Carbon Gen6
* USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)

Both my laptops are fine, not a single port on my desktop will work, I have to go through the USB hub in my screen to get the UHK to work.

My desktop logs the same issues @Agrendalath has in their dmesg output:

```
usb X-Y: Not enough bandwidth for new device state.
usb X-Y: can\'t set config #1, error -28
```

daenney, 2020-04-18:
If I'm reading all the reports right so far, the issue seems prevalent on desktop AMD Ryzen systems for some reason.

mondalaci, 2020-04-18:
@daenney Thanks for your detailed report! A contractor of ours will look into this issue.

garro95, 2020-04-23:
I can confirm keyboard not working on ASUS TUF GAMING X570-PLUS

organsnyder, 2020-04-24:
I'm encountering the same issue on my new Gigabyte X570 UD motherboard. The keyboard isn't usable in the BIOS nor in Linux (Kubuntu 20.04), unless it is plugged in through a hub. Every other peripheral works fine.

garro95, 2020-04-24:
I wonder if a USB 3.0 HUB will work or it needs to be USB 2.0

mondalaci, 2020-04-26:
I wish I knew. I'd like to hear about this, too.

phlpl, 2020-04-27:
I have a Gigabyte Aorus x570-i and just tried this and can confirm that the UHK works with a USB 3.0 hub using the blue ports.  This is the hub I am using https://www.anker.com/products/variant/4port-usb-30-ultra-slim-data-hub/A7516011

Agrendalath, 2020-04-27:
@garro95, I have the UHK connected to the USB 3.0 hub built into my monitor and it works fine. The only downside is that it's not possible to upgrade firmware this way.

fcasco, 2020-05-16:
I can confirm on Asus Prime x570-pro, connected to the usb ports on the back, the keyboard does not work and show "Not enough bandwidth for new device state" "can't set config #1, error -28" (in dmesg).
Using it with a cheap usb hub works fine.

xxxajk, 2020-06-20:
Patch coming soon to fix the issue.

xxxajk, 2020-06-21:
Please see if #301 fixes this.
Working here, same setup.

Agrendalath, 2020-06-21:
@xxxajk, thank you for creating this! I built your `master` branch locally, flashed it on my keyboard and now it's working fine with direct USB 3.x connection to `ASUS TUF GAMING X570-PLUS` motherboard.
cc: @mondalaci

xxxajk, 2020-06-21:
Don't thank me totally. @mondalaci found me and hired me to fix it.
Thank YOU for verification that it works, and thank YOU for the excellent bug report!

closed 2020-06-21 by mondalaci

Jopie01, 2020-06-22:
Just installed the newest version (8.10.1) of the firmware and it works perfectly!

xxxajk, 2020-06-23:
@Jopie01 Thanks for the extra verification.

stphnlwlsh, 2020-06-23:
I see this was released for firmware. Any idea on when it might make its way over to Agent?

organsnyder, 2020-06-23:
I was able to flash it using the existing Agent release by downloading the firmware release from GitHub.

xxxajk, 2020-06-23:
@organsnyder thank you for testing.
@stphnlwlsh get the latest release of agent.

mondalaci, 2020-06-23:
@stphnlwlsh We can't currently release a new Agent version due to the standing bug, but you can flash the latest firmware revision that contains the fix with the latest Agent regardless.

Jopie01, 2020-06-23:
@stphnlwlsh Go to https://github.com/UltimateHackingKeyboard/firmware/releases and download the most recent version of the firmware (https://github.com/UltimateHackingKeyboard/firmware/releases/download/v8.10.1/uhk-firmware-8.10.1.tar.gz). Start the Agent, go to 'firmware' and click on the button 'Choose firmware file and flash it'. Select your just downloaded firwmare file and of you go.

stphnlwlsh, 2020-06-23:
I got it. Used current agent to download the zip to my UHK. Thanks all!

PR #301 (2020-06-21): Fix for USB3.x issues
Fixes #269.

This eliminates the bandwidth issue. Linux no longer complains.
This should fix USB3.x on Mac too.

I'll have to see if the following are happier:

1. Ancient Mac that I have
2. BIOS on the ASUS machine