UltimateHackingKeyboard/firmware #270: Add more metadata to firmware builds

mondalaci, 2020-01-22:
Currently, the firmware version number gets added to compiled firmware binaries and package.json, and Agent displays it.

Unfortunately, the version number poorly identifies the many firmware builds floating around, so additional metadata would be very useful to identify them explicitly, such as the GitHub username of the author, and a tag. By tag, I don't mean a Git tag, just a plain alphanumeric string that would give an idea about how the build differs from master.

Agent would display the above as ${user}:${version}:${tag}. The first colon would only be displayed if ${user} is a non-empty string. The second colon would only be displayed if ${tag} is a non-empty string.

Imagine the following metadata displayed in Agent:

* kareltucek:8.5.4:25
* p4elkin:8.5.3:187-4
* 8.7.0:diagonal-mouse-slow-movement
* 8.7.0:909e30433fb07b4f4146a4b4889ce3d5377ce6f4

I think the above metadata would be sufficient for the use cases we've encountered so far.

laxu, 2020-02-19:
Easiest way to do this would be just adding an object to package.json with this info, e.g.

uhk_metadata: { user: 'laxu', build: 'some_commit', tag: 'impressive-feature' }

mondalaci, 2020-02-19:
There's more to this than just a JSON as the firmware has to expose these values via USB. I'll write a more detailed spec later.

mondalaci, 2022-03-19:
Done via https://github.com/UltimateHackingKeyboard/firmware/pull/460

closed 2022-03-19 by mondalaci