UltimateHackingKeyboard/firmware #274: Building problems - make flash and npm.

kareltucek, 2020-01-27:
I have been having some problems with building and flashing lately:

- first, `./make-release.js` started failing on 8.7.1. 
- so I upgraded agent in lib/agent. This time I needed to:

      rm -rf node_modules
      sudo npm cache clean -f
      sudo npm install -g n
      sudo n 12.4.0
      (then the standard npm install && npm run build && npm run electron)

- So I ended up with npm 6.9.0 and node 12.4.0. Agent builds and runs fine. `make-release.js` works fine. But `make flash` in `right` started failing with the following (spammed again and again indefinitely):

      NODE_MODULE_VERSION 57. This version of Node.js requires
      NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
      the module (for instance, using `npm rebuild` or `npm install`).
          at Object.Module._extensions..node (internal/modules/cjs/loader.js:815:18)
          at Module.load (internal/modules/cjs/loader.js:641:32)
          at Function.Module._load (internal/modules/cjs/loader.js:556:12)
          at Module.require (internal/modules/cjs/loader.js:681:19)
          at require (internal/modules/cjs/helpers.js:16:16)
          at bindings (/opt/data/UHK/firmware/lib/agent/packages/usb/node_modules/bindings/bindings.js:112:48)
          at loadBinding (/opt/data/UHK/firmware/lib/agent/packages/usb/node_modules/node-hid/nodehid.js:19:46)
          at Object.showdevices [as devices] (/opt/data/UHK/firmware/lib/agent/packages/usb/node_modules/node-hid/nodehid.js:127:5)
          at Timeout._onTimeout (/opt/data/UHK/firmware/lib/agent/packages/usb/uhk.js:213:37)
          at listOnTimeout (internal/timers.js:531:17)
      (node:2070867) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 11)
      (node:2070867) UnhandledPromiseRejectionWarning: Error: The module '/opt/data/UHK/firmware/lib/agent/packages/usb/node_modules/node-hid/build/Release/HID-hidraw.node'
      was compiled against a different Node.js version using
      NODE_MODULE_VERSION 57. This version of Node.js requires
      NODE_MODULE_VERSION 72. Please try re-compiling or re-installing

- So I tried to `rm -rf node_modules`, `npm install` and `npm rebuild` in `scripts`. Didn't help.
- So I tried to google and to `npm rebuild node-hid --update-binary` in both `lib/agent` and `scripts` folders. Ended up with node-hid@0.7.9. Didn't help.
- Next try was `rm -f /node_modules && rm -f package-lock.json && npm install` (again in both paths) (adviced by google). Then added `npm run build` in agent's directory. Didn't help.

Any ideas?

kareltucek, 2020-01-27:
(Also, if `build_make` directories in `trackball` and `trackpoint` are not created manually, the `make-release.js` won't work for some reason.)

kareltucek, 2020-01-27:
Thinking about it a bit more, missing `build_make` was maybe the only problem, and upgrading agent            just broke flashing machinery.

So the question is - which node version is correct? The thing claims that required is >=12.4.0 and <13.0

kareltucek, 2020-01-27:
So, we have two problems now:

- Upgrading node to 12.9 didn't help and I still cannot flash.

- I am a bit suspictious why `make-release.js` needs the `build_make` directories, since I assume that it builds somewhere in `scripts/<version>` and therefore should not take anything from `{trackball|trackpoint}/build_make`. If it does, it packs incorrect firmware versions, does it not?

mondalaci, 2020-01-27:
@ert78gb Would you please help @kareltucek with this?

kareltucek, 2020-01-27:
Alright, solved. Deleting entire Agent, reseting to pure git files and installing again finally helped.

I was not aware that there are other npm/node related files than those in node_modules. Not sure what the proper way of dealing with this should have been, but does not matter I guess. 

Thanks guys, even if just for your moral support! :-)

closed 2020-01-27 by kareltucek

ert78gb, 2020-01-28:
Sorry for the late response.
use the `npm run clean` to delete all `node_modules` in the `packages/*` subdirectories.

kareltucek, 2020-01-28:
No problem and thanks for proper instructions!