UltimateHackingKeyboard/firmware #276: Possible typo in buspal

kareltucek, 2020-01-30:
https://github.com/UltimateHackingKeyboard/firmware/blob/14f1fd2ad478e5e05641570e69f436078b1e15a1/right/src/buspal/command.c#L179

@mondalaci would you take a look at this?

mondalaci, 2020-01-30:
I'm unsure about this, don't have time to look into it, and not sure if I can get a hold of Erich who customized BusPal for our needs.

Would you possibly look into it based on the BusPal documentation?

kareltucek, 2020-01-30:
Well, I hoped you would have some larger image of where and how is the BusPal used, and therefore could make a qualified guess of possible consequences and their urgency... 

I can take a look at the docs, but I don't think I can arrive at a reasonably reliable conclusions within a reasonable amount of time. All I can tell atm is that if the line is replaced by `if(status!=kStatus_Success)`, UHK firmware seems to work as fine as before (so far).

luteijn, 2020-01-30:
There seems to be an #ifdef that might exclude that part of the code. Also, if status usually (happy-flow) IS kStatus_Success, then the flow usually will end up the same - condition false, skip then-clause.

kareltucek, 2020-01-30:
The `ifdef` includes the part (line 10 defines it). 

It seems to take care of  something related to flashing firmware and does not seem to be executed during normal operation (i.e., when using keyboard regularly) at all. 

If Erich is not available to give a comment, I would probably leave it as is. I am afraid that the `Failed` branch might be untested by the author, so "correcting" it without deeper understanding of the code might not be the wisest thing to do.

kareltucek, 2020-01-30:
Or well,  it really just confirms the failed status in case of fail... That probably can be fixed without any special research...

mondalaci, 2020-01-30:
BusPal is the Kinetis bootloader protocol translator. It routes the bootloader commands received via USB by the right half to the left half and modules via I2C.

PR #277 (2020-01-31): Fix typo in buspal.
Closes #276
closed 2020-01-31 by mondalaci