UltimateHackingKeyboard/firmware #279: Make some keys sticky

Cretezy, 2020-02-24:
I use Ctrl + Tab to switch between files in my IDE, and I would like to make that sticky.

I see that only Tab/arrows + alt/super is allowed to be sticky, but it would be nice to be able to modify the stickiness though the agent as well (should I move this ticket there?).

mondalaci, 2020-02-24:
Which IDE treats Ctrl+Tab the way OSes treat Alt+Tab?

Cretezy, 2020-02-24:
[image]

IntelliJ has a switcher using Ctrl + Tab (or at least that's what mine is set as) to switch between files. If it's not sticky, it would only switch to the last opened file

mondalaci, 2020-02-24:
Gotcha, thanks!

Please this make this shortcut sticky, too, @kareltucek.

kareltucek, 2020-02-24:
Sure.

Cretezy, 2020-02-24:
Wouldn't it be better to be able to configure which are sticky? I can imagine others using different binds for different programs

mondalaci, 2020-02-24:
@Cretezy I'd much rather make sticky keys hardcoded and make things work without configuring anything.

PR #280 (2020-02-26): Add ctrl+tab and ctrl+arrows shortcuts into the sticky set.
I am adding "ctrl + arrows" too. The reason to remain as consistent and intuitive as possible. If it turns out to be a problem, we can always revert...

Closes #279.
closed 2020-02-26 by mondalaci