UltimateHackingKeyboard/firmware #281: Segmentation fault attempting to flash latest firmware

ebr, 2020-03-01:
```
cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.10
DISTRIB_CODENAME=eoan
DISTRIB_DESCRIPTION="Pop!_OS 19.10"
```

`UHK.Agent-1.4.2-linux-x86_64.AppImage`

```
16:49:29.932 › [UhkOperations] Waiting for bootloader
16:49:30.938 › [UhkOperations] Flash security disable
16:49:31.106 › [UhkOperations] Flash erase region
16:49:32.418 › [UhkOperations] Read RIGHT firmware from file
16:49:32.476 › [UhkOperations] Write memory
16:49:35.692 › [UhkOperations] Reset bootloader
16:49:35.795 › [UhkOperations] Close communication channels
16:49:35.795 › [UhkOperations] Right firmware successfully flashed
16:49:35.796 › [UhkOperations] Start flashing left module firmware
16:49:35.796 › [UhkHidDevice] Start reenumeration, mode: NormalKeyboard
../../third_party/tcmalloc/chromium/src/tcmalloc.cc:337] Attempt to free invalid pointer 0x5a3639372e35333a 
[1]    4559 segmentation fault  ~/Downloads/UHK.Agent-1.4.2-linux-x86_64.AppImage
```

After the update attempt:
```
Firmware 8.5.3 is running on the left keyboard half.
Firmware 8.8.1 is running on the right keyboard half.
```

ebr, 2020-03-01:
Update: apparently this started with 8.7.1. I successfully flashed each firmware version starting with 8.6.0, but once I got to 8.7.1, the above segfault presented.

I was able to downgrade to 8.7.0 on both keyboard halves.

mondalaci, 2020-03-01:
Closing as a duplicate of https://github.com/UltimateHackingKeyboard/agent/issues/1175

closed 2020-03-01 by mondalaci