UltimateHackingKeyboard/firmware #282: Polling rate question

mondalaci, 2020-03-04:
Just got a question whether the UHK supports 1000 Hz polling rate, and I'm unsure of it. We should figure this check this on Linux, Mac, and Windows.

kareltucek, 2020-03-04:
Polling itself is hardcoded to 4ms. I am not sure about decreasing it - if there is no hardware limit, then it might work, and on average the update loop is fast enough, but there is not much margin...

But I am unsure of whether UHK actually sends reports in response to polls, because interrupt handling seems to allow devices to send data as soon as they are available, and in that case I believe that the actual response time depends on the hardware USB subsystem and host system.

mondalaci, 2020-03-04:
It'd be worth to count the reports that the UHK creates, expose the counter via the USB debug buffer, and read it with agent/packages/usb/get-debug-info.js. This would not measure the polling rate but the report generation because the latter may be the bottleneck.

kareltucek, 2020-03-04:
The actual sending of a report - e.g., call to UsbBasicKeyboardAction() takes just 75 us (micro seconds). Then it takes up to 4000 us to receive the response interrupt.  During this time, UHK's report generation/creation is inactive, so you will not get any meaningful number.

kareltucek, 2020-03-04:
(But on average one update cycle takes between 200 and 700 ms, (spikes are up to some 2000), so the bottleneck should not be in *sending* data.)

kareltucek, 2020-03-04:
Anyway, setting USB_BASIC_KEYBOARD_INTERRUPT_IN_INTERVAL 1 seems to work as expected - then update cycle timer shows 1.0ms with the usb test mode enabled, which means that one report is dispatched every 1ms, polling runs at 1ms *and* all the state information is updated properly too(, at least in average case).

It would be worth to give it a bit more testing (with all the *_INTERVALs set to 1ms) and check with a usb expert, but I think it can be done without much effort.

mondalaci, 2020-03-04:
Setting every *_INTERVAL to 1ms may be overkill. I can't imagine gamers practically needing it for anything other than the basic keyboard interface.

I'm a little bit lost regarding the USB test mode. How did you exactly measure this?

kareltucek, 2020-03-04:
I employed the stopwatch mentioned in the secondary role PR. I measure time between 1000 calls to `updateActiveUsbReports` (not `UpdateUsbReports`), then divide the number by 100 and (on every 1000th report update) show the result on Led Display.

mondalaci, 2020-03-04:
Can you check it on all the three major OS?

kareltucek, 2020-03-04:
> Setting every *_INTERVAL to 1ms may be overkill. I can't imagine gamers practically needing it for anything other than the basic keyboard interface.

Note that even though all interfaces can send their reports concurrently (if all transfers are initiated in the same cycle), once one report is sent, all other interfaces are blocked until success is confirmed. This means that if you do not decrease mouse interval too, it will slow down basic reports too in the test mode.

> Can you check it on all the three major OS?

Nope. But here is the firmware. Just hit fn+1 to show the stopwatch (it will update approximately every one second, in s*10^-4). Should show `010`.

[uhk-fw-8.8.1-polling_v1.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/4286877/uhk-fw-8.8.1-polling_v1.tar.gz)

--------------------------------------------------
Unfortunately this is still only the USB part of the thing - what is/should be the update rate of modules? My measurements yield 2.5ms on average, 4ms on spikes, which obviously doesn't guarantee the sought 1ms response.

Where is the module update rate configured? And is it safe to decrease it, especially when considering the two more modules which will have to fit into the update cycle?

mondalaci, 2020-03-04:
Are you sure that we'd have to set the polling rate of every USB interface to 1ms, otherwise 4ms stays in effect? If so, please explain why you think this is the case.

What do you mean by the "test mode"?

The display shows mostly 006 with occasional 007 peaks on all three OSes.

The polling rate of the modules and the left half depends on data rate of the main I2C bus and the optimization of the I2C protocol scheduler. I plan to optimize it further, but it's not a priority yet.

kareltucek, 2020-03-04:
> Are you sure that we'd have to set the polling rate of every USB interface to 1ms, otherwise 4ms stays in effect? If so, please explain why you think this is the case.

No, no. For normal use it is mostly fine. 

The problem occurs only when another interface sends reports at the same time as the basic keyboard reports. E.g., if  I press and hold mouse movement (let's assume that its speed is so high that it sends reports whenever it is allowed to), then the sending rate of basic keyboard reports drops to 4ms.

The reason is the logic of `UsbUpdateSemaphore`. It basically makes the reports be sent in "batches" - if there are multiple pending reports, they are all sent at the same time. Then the keyboard waits until *all* interfaces have confirmed before allowing another report.

I think that this logic tries to prevent one endpoint from usurping the resources from others. But honestly I think it could be possibly dropped.

> What do you mean by the "test mode"?

The fn+tilde mode which spams a/del and left+right mouse movement.

> The display shows mostly 006 with occasional 007 peaks on all three OSes.

Sorry, I was apparently unclear on this - the timer just measures the update cycle interval. In this case it means that update cycle takes 0.6-0.7 ms when idle.

In order to make it measure the actual time needed to send a report, you need to make the keyboard send reports continuously - e.g., by the testing mode. So instructions should have been "Press fn+tilde (you do not need to activate it via agent first in this version), and then fn+1."

> The polling rate of the modules and the left half depends on data rate of the main I2C bus and the optimization of the I2C protocol scheduler. I plan to optimize it further, but it's not a priority yet.

So then it is quite clear that even though that we can support 1 ms polling, we cannot support actual 1ms response at the moment. (As long as my measurements are correct.)

mondalaci, 2020-03-04:
Gotcha, thanks for explaining! The `UsbUpdateSemaphore` logic makes mouse pointer movement smooth. Without it, movement speed is irregular.

With the test mode enabled, the display mostly shows 010 with occasional bursts in the 011-013 range on Linux and Windows, and it shows 020 on Mac, so this does seem to be an OS (or OS settings) limitation on Mac.

Which function call does Fn+1 measure without the test mode enabled exactly?

Have you also noticed that this change makes mouse movement much faster? It's probably due to the increased polling rate.

kareltucek, 2020-03-04:
> Which function call does Fn+1 measure without the test mode enabled exactly?
 
Just as explained above - In both cases it measures the time between calls to updateActiveUsbReports and averages them. (I.e., the stopwatch snippet is placed right under the usb test mode snippet, and every 1000 calls it recalculates the value.)

> Have you also noticed that this change makes mouse movement much faster? It's probably due to the increased polling rate.

Interesting. Haven't noticed it...

kareltucek, 2020-03-04:
(The call does not happen when the semaphore is on, so then the interpretation is slightly different.)

mondalaci, 2020-03-04:
Actually, the mouse speed is not only faster but mouse speed became irregular. This wasn't an issue for for firmware 8.8.1 and neither for your secondary role branch. I can notice it quite easily on Linux if I use a homogenous background and keep moving the pointer horizontally leftward and rightward, back and forth.

kareltucek, 2020-03-05:
I guess I should not be surprised. While measuring the times, I repeatedly encountered suspictiously high spikes - e.g., from the regular 600us of the update cycle to up to 2000us. They are probably also the reason for the occurences of the freeze bug.

However, I don't have any good idea how to investigate these occurences. I guess you do not have any tracing/profiling tool that could run in real time on UHK_at hand?

mondalaci, 2020-03-05:
I think it should be possible to hook up an SWD probe to the UHK, and profile this, but I don't have related experience.

Alternatively, it should be possible to expose measurements via the USB debug buffer.

kareltucek, 2020-03-05:
Mmm, I am afraid I do not have much experience with setting these things up either... Of course I guess I can place my own "trace/log" hooks around the place... 

Should I add it onto my todo list? (I would run the traces on my fork onnce it is up-to-date with master...)

mondalaci, 2020-03-05:
Sure, it'd be nice to trace this from code. It's not a priority.

mondalaci, 2020-06-30:
@xxxajk Is this resolved by #301?

xxxajk, 2020-06-30:
It might be. Ping the reporting party and have them test.

mondalaci, 2020-06-30:
I wish I remembered who's asked the question.

Could you possibly check this one, @kareltucek?

kareltucek, 2020-07-01:
Not sure what is here to check - the above testing should still apply as well as other observations.

The way I understand the problem - yes it is solved (, but I am not a usb expert). But there are still the bottom lines about actual response time being higher than 1 ms, about uneven mouse movement and about desirability of further digging into I2C optimizations and general profiling.

mondalaci, 2020-07-01:
I'm only interested about USB interface polling rate regarding this issue. There should be some way to check this in OSes.

kareltucek, 2020-07-01:
https://wiki.archlinux.org/index.php/Mouse_polling_rate looks promising. Will look into it in the evening or tomorrow..

kareltucek, 2020-07-02:
Confirming - my `/sys/kernel/debug/usb/devices` lists all UHK interfaces at 1ms:

```
T:  Bus=01 Lev=01 Prnt=01 Port=02 Cnt=01 Dev#= 20 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d50 ProdID=6122 Rev= 1.01
S:  Manufacturer=Ultimate Gadget Laboratories
S:  Product=Ultimate Hacking Keyboard
C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=02(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
E:  Ad=83(I) Atr=03(Int.) MxPS=   8 Ivl=1ms
I:* If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=84(I) Atr=03(Int.) MxPS=   8 Ivl=1ms
I:* If#= 3 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=85(I) Atr=03(Int.) MxPS=   1 Ivl=1ms
I:* If#= 4 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
E:  Ad=86(I) Atr=03(Int.) MxPS=   8 Ivl=1ms
```

mondalaci, 2020-07-02:
Awesome, thanks for confirming!

closed 2020-07-02 by mondalaci

kareltucek, 2020-09-11:
> Have you also noticed that this change makes mouse movement much faster? It's probably due to the increased polling rate.

The reason seems to be that the timer does not measure time always at the same pace - when reports are being sent continuously, it measures time somewhat slower. 

I assume that timer gets incremented only when interrupts are not globally disabled. Which means that four times more reports sent makes interrupts be disabled for four times longer time.

mondalaci, 2020-09-12:
Thanks for the explanation! In any case, the polling rate of the most recent firmware is 1000 Hz, and nobody has complained about mouse speed changes, so I consider this a non-issue.