UltimateHackingKeyboard/firmware #285: Fix coding style

mondalaci, 2020-04-11:
#265 has introduced loads of underscored_variable_names which should be written as lowerCamelCase according to [our coding standards](https://github.com/UltimateHackingKeyboard/firmware/blob/master/CONTRIBUTING.md). Please fix this, and let's adhere to our coding standards as much as possible.

kareltucek, 2020-04-11:
Oops. I will need to take care of reverse merge of the functionality first. Sorry for the inconvenience!

daenney, 2020-04-11:
Since there's already a `.clang-format` in this repository, why not have CI check this on PRs?

Something like:

```yaml
addons:
  apt:
    sources:
      - llvm-toolchain-xenial-10
    packages:
      - clang-format-10
```
And then as one of the jobs

```
find . -type f \( -name '*.c' -o -name '*.h' \) -print0 | xargs -r0 clang-format
```

mondalaci, 2020-04-12:
I love the idea! I wasn't aware that C coding style verification is possible this way. I'd love to see this getting implemented assuming it really works.

PR #290 (2020-04-21): Fix naming conventions
Closes #285 .
closed 2020-04-21 by mondalaci