UltimateHackingKeyboard/firmware #287: Rearrange USB interfaces for better BIOS compatibility

mondalaci, 2020-04-15:
@xxxajk has just advised that we should change the current USB interface order:

UsbGenericHidCallback
UsbBasicKeyboardCallback
UsbMediaKeyboardCallback
UsbSystemKeyboardCallback
UsbMouseCallback

Let's use the following order instead:

UsbBasicKeyboardCallback
UsbMouseCallback
UsbMediaKeyboardCallback
UsbSystemKeyboardCallback
UsbGenericHidCallback

The assumption is that the USB descriptor parser of the BIOS can claim only the first two interfaces this way instead of having to parse all the interfaces, potentially resulting in better BIOS compatibility.

An unwanted side effect is that Agent is hardcoded to use interface zero as the communication interface which gets renumbered, making Agent unable to communicate with the UHK. To combat this issue, we can change the USB PID, make Agent branch based on the PID, and use the proper interface number. But first, let's implement the above in a PR, and check out if it truly improves BIOS compatibility.

xxxajk, 2020-04-15:
Actually, no need for a new PID, just check the first interface descriptor. :-)

mondalaci, 2020-04-15:
Great idea! Although I'm wondering whether it's possible with node-hid. We'll look into it for sure!

borje, 2021-03-16:
Is this implemented?

mondalaci, 2021-03-16:
@borje This issue is not implemented yet. If it is, the issue will be closed like every other issue.

steamraven, 2021-03-25:
Here is some untested code.  Careful, since the agent is not configured, the agent will stop talking to it.  You will need to enter emergency mode to fix.

https://github.com/steamraven/firmware/tree/287-reorder-interfaces

mondalaci, 2021-04-14:
@steamraven Thanks for your contribution! Unfortunately, I can't test it because the actual USB interface order works with the BIOS of my Gigabyte B550M DS3H motherboard.

Please let me know if anyone can reproduce this issue.

jonhermansen, 2021-04-14:
Will the proposed changes help at all with KVM compatibility?

I found that the UHK v1 would not work with my StarTech DisplayPort 2-way KVM at all.

mondalaci, 2021-04-14:
@jonhermansen The proposed changes might help with KVM compatibility. We can make a firmware build eventually, but then Agent and further firmware updates won't work due to USB interface reordering until [shorting these pads](https://ultimatehackingkeyboard.com/repair/unbricking-the-right-keyboard-half), so testing this will be painful.

xxxajk, 2021-04-14:
Actually, I question why the firmware uploader has to have a certain order at all. You should be able to determine this from the device descriptors, that's what they are there for. I would do this, but I'm unfamiliar with the language it is written in. 

What I do in my projects is instead of having the special wormhole backdoor for programming, I have a dedicated trigger. Let me give you some examples.
Let's say we have a device operating like CDC-ACM, and you need to trigger it to go into bootloader mode. The easy way to do that is have the software watch for the baud rate to change to 110baud, and toggle DTR and RTS is a special sequence. That causes the MCU to reboot into the bootloader only, and doesn't interfere with the operation of CDC-ACM.
The same is possible with HID devices. You send custom HID packets, trigger the bootloader. No need for the bootloader to always show up that way :-)

steamraven, 2021-04-14:
I think I have a solution to the agent.   The node-hid through libusb and hidraw supports querying HID usage for the interfaces.  This makes it trivial to find the management interface, no matter what interface it is on.   

However, on SOME linux boxes,  the usage can come back as zero.  Since we are already filtering on Vendor ID, we know this is some type of UHK keyboard and we can do the following in the case it is zero.  We can send commands to the possible interfaces and check for a response.   Again, this is only used if usage comes back zero, which is only in some weird configurations of linux (like mine).

xxxajk, 2021-04-14:
Exactly, thus you really shouldn't have to make anything special first, with exception to compatibility.
From all the composite devices I have seen in the wild, the order has always been:

1. Keyboard
2. Mouse
3. everything else, (other HID, serial ports, audio, video, adnausium)
4. Vendor specific interfaces

Yes this shouldn't matter, but due to poor choices in implementations that aren't as smart, it does, and solves this issue.
I'm even willing to bet that it solves all kinds of issues, from BIOS to different OS.
I've even seen where multiple serial, which has 2 descriptors per physical interface, mess up if the two descriptors for that same port aren't in the "usual order", although according to the spec, this shouldn't matter at all.

jonhermansen, 2021-04-17:
Just catching up on this thread. @mondalaci Thank you for your answer. I am curious to try the proposed patch with the StarTech KVM, so I may give it a go soon.

Shorting the contacts should not be a problem for me, and the value of having it work seamlessly with KVM outweighs not being able to use it. I'll report back soon if I try it out.

steamraven, 2021-04-18:
Put in a merge request on the agent side so the agent can still communicate
https://github.com/UltimateHackingKeyboard/agent/pull/1488

mondalaci, 2021-04-18:
@steamraven I've just tested your 287-reorder-interfaces branch, my UHK enumerated via USB, and I could use it. Very impressed! Anyone interested, please test [uhk-firmware-steamraven-287-reorder-interfaces-8.10.8.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6332132/uhk-firmware-steamraven-287-reorder-interfaces-8.10.8.tar.gz), but be aware that Agent versions released up until this point won't be able to talk to this firmware, so the bootloader has to be physically activated by [shorting the pads](https://ultimatehackingkeyboard.com/repair/unbricking-the-right-keyboard-half) for Agent to flash the stock firmware.

Even though your firmware enumerated properly, many of my mappings didn't function as intended, so there's some related bug. Please test the above firmware regardless, @jonhermansen, and everyone interested.

I'll follow up about Agent in the mentioned PR in a sec.

xxxajk, 2021-04-18:
I'll check it out, does it fix the remaining Apple wakeup problem?

mondalaci, 2021-04-18:
I can't check the Apple wakeup issue because I don't have a new MacBook. Anyone with a new MacBook, please test!

steamraven, 2021-04-20:
@mondalaci 
> many of my mappings didn't function as intended, so there's some related bug.
That is pretty weird.  The code change is pretty small:
https://github.com/steamraven/firmware/commit/9ad9650f0fbb721486e2fedccf51da8c8ff4fd67

One thing that pops to mind is that the original branch was based on a 2 week old branch.  I updated it to master, but I don't know if that is included in the firmware you complied.   But really the only thing that has changed in the last two weeks related to mappings is caret mode and the isModuleActive bug.  

The only other thing that uses the changed variables is UsbReportUpdateSemaphore, and the order the bits is different, but as long as its consistent, it shouldn't affect anything.

mondalaci, 2021-04-20:
@steamraven Thanks for explaining! I'll retest your branch as soon as I get confirmation from others that it solves KVM / BIOS issues.

xxxajk, 2021-11-01:
Guess what else refuses to work with bootp EPs not first?
U-Boot.

AkechiShiro, 2022-03-11:
This looks useful for many use cases, especially at boot time, I'd like it that the PR land in master when the code is stable and the agent can be used successfully.

mondalaci, 2023-06-09:
@Restraint I'm not sure how well the mentioned firmware is working, but it's surely outdated, and after flashing it, you won't be able to update your configuration via recent Agent versions unless updating your firmware.

If your UHK is bricked, shorting the pins longer, possibly during the whole firmware update process, might help.

mondalaci, 2023-06-09:
@Restraint I can't promise to prioritize this issue, but [this is the solution](https://github.com/UltimateHackingKeyboard/firmware/issues/536#issuecomment-1304267903) we plan to implement, and the referenced issue is more relevant to your problem.

mondalaci, 2023-06-10:
@Restraint I stand corrected. Our USB firmware expert is sporadically available, and we have other priorities now, so it may take some time, but we'll look into this. Thanks for the heads-up and patience!

xxxajk, 2023-06-10:
@mondalaci  who? me? :-) I'm watching... Just super-duper busy.

mondalaci, 2023-06-10:
@xxxajk I was actually thinking of @benedekkupper. I don't expect you ever to have free time :)

xxxajk, 2023-06-10:
@mondalaci Yeah, things are kind of crazy here. I can at least do testing if needed. Ping me if you need testing. I have a KVM, various PC's, android tablets, some Rasp-pi's, etc etc etc...

sunghospark, 2023-11-21:
I just tested out @steamraven 's [firmware](https://github.com/UltimateHackingKeyboard/firmware/files/6332132/uhk-firmware-steamraven-287-reorder-interfaces-8.10.8.tar.gz) with its [matching agent](https://github.com/steamraven/agent/tree/interface-reorder) fixed KVM compatibility issue with tesmarts latest KVM switch PKS0402A20.

I think this would be a better fix to try to address [the firmware comptability issue](https://github.com/UltimateHackingKeyboard/firmware/issues/536) than the 6key rollover change

mondalaci, 2023-11-21:
@sunghospark Thanks for sharing your findings!

It's useful to know that interface order profoundly affects this issue. We'll reorder the USB interfaces accordingly, but we can't merge the relevant commit yet because the USB-related codebase is in flux and expected to change.

sunghospark, 2023-11-21:
@mondalaci no worries, I'm just glad I could find a working configurations for now.
There was an issue with [uhk-firmware-steamraven-287-reorder-interfaces-8.10.8.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6332132/uhk-firmware-steamraven-287-reorder-interfaces-8.10.8.tar.gz) where key mapping change was not working.

I compiled from the latest commit of steamraven's branch and that sorted out the key mapping issue. Sharing here to save some other peoples time to install all the firmware dev setup.
[uhk-firmware-steamraven-usb-reorder-8.10.9.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/13428678/uhk-firmware-steamraven-usb-reorder-8.10.9.tar.gz)

mondalaci, 2023-12-12:
@kareltucek Can you recreate the relevant PR? A fair number of users seem to be affected by KVM compatibility, and this could help them. We'd modify Agent accordingly.

kareltucek, 2023-12-27:
@mondalaci I will see. 

Where should the new gamepad interface go?

mondalaci, 2023-12-27:
@kareltucek Yes, the end should work.

closed 2024-02-26 by mondalaci

jonhermansen, 2024-02-26:
> Just catching up on this thread. @mondalaci Thank you for your answer. I am curious to try the proposed patch with the StarTech KVM, so I may give it a go soon.
>
> Shorting the contacts should not be [...]

Sorry, I never got to try out this configuration again. Now that the changes have landed, I may give it a try soon! Thank you all for your excellent work on UHK

commit 8f25f54d (2024-02-26): Fixes #287 Reorder USB Interfaces
commit f794a563 (2024-02-26): Merge pull request #738 from kareltucek/rebased_interface_reorder

Fixes #287 Reorder USB Interfaces