UltimateHackingKeyboard/firmware #288: Fast key combination not recognized (must holding first key for a while and then hit the second key)

lohnicky, 2020-04-17:
I re-mapped the Mouse key to the Cmd. When hitting shortcut combinations like Cmd+C, Cmd+V, Cmd+F etc. many times I get only the second key and the combination is not recognized. The issue is most likely in the speed, when I hit the combination pretty fast it fails quite often, if I hold the Mouse key for a bit and then additionally hit the second key it works.
- Running macOS 10.15.4 (19E287)
- I tried 8.8.1, 8.7.0 and 8.7.1 versions of the Firmware and the issue persist. 
- It might be that updating a Firmware helps for a while but then it's the same. However this might be only placebo effect so not sure about this one.

kareltucek, 2020-04-17:
- We are talking about regular Cmd key right and regular C/V/F keys right? (No secondary roles involved, no layer switching involved.)
- How fast are we talking about?
- Are you sure it is not a finger issue? UHK should be able handle everything as long as you (as a human being) can be sure you really did hit them in correct order. 
- Are you sure this is not a switch issue? 
- Is this left/right half specific?
- How reliably can you test this? If I send you a custom firmware, how fast can you tell me "the issue persists".

lohnicky, 2020-04-17:
- We are talking about regular Cmd key right and regular C/V/F keys right? (No secondary roles involved, no layer switching involved.)
  - LCmd mapped to Mouse key and regular C/V/F keys. So for copy which is LCmd+C I use the "re-mapped Mouse"+C
- How fast are we talking about?
  - Quite fast, I hit this combinations many times per day for years :).
- Are you sure it is not a finger issue? UHK should be able handle everything as long as you (as a human being) can be sure you really did hit them in correct order.
  - I bought the keyboard recently, using this speed on many other keyboards, including laptops work. Also I used UHK on Linux PC and it worked without issue, when I switched to macbook it started to be an issue.
- Are you sure this is not a switch issue?
  - How can I check. Physically I don't feel any issue.
- Is this left/right half specific?
  - Left
- How reliably can you test this? If I send you a custom firmware, how fast can you tell me "the issue persists".
  - We can chat/call and debug together if it helps.

mondalaci, 2020-04-17:
@lohnicky You can test if it's a switch issue by binding LCmd to another key. Given that nobody else has this issue, chances are high it's a switch issue.

luteijn, 2020-04-17:
If it (still) works fine on the linux box, it might also be that something is running on the mac that does something special with the ctrl/cmd key, so that it needs to be pressed a little longer before hitting the next key.

lohnicky, 2020-04-17:
I re-mapped LCmd to LCtrl with a very short test it looks like that it helped. 

I also re-mapped LShift to the Mouse key and holding shift and clicking the F key I get this results (each case is hit Mouse, hold it and keep clicking the F key):
- fffffffffFfFFFFFFFF
- fFFfFFFFFFFFFFFF
- FFFFFFFFFFFFFFF
- FFFFFFFFFFFFFFF
- FFFFFFFFFFFFFFF
When re-mapping LCtrl to Shift I get:
- FFFFFFFFFFFFFFF
- FFFFFFFFFFFFFFF
- FFFFFFFFFFFFFFF
- FFFFFFFFFFFFFFF
So indeed it looks like the switch issue.

kareltucek, 2020-04-17:
> So indeed it looks like the switch issue.

Yep, seems to be the culprit. 

Frankly I am happy to hear this since I am quite paranoid about this type of firmware + IC2 communication related problems. I plan to give it some more investigation regardless as part of #282 .

mondalaci, 2020-04-17:
[Contact cleaner spray](https://ultimatehackingkeyboard.com/repair-and-modding/fixing-misbehaving-keys) should fix this issue.

closed 2020-04-17 by mondalaci