UltimateHackingKeyboard/firmware #289: Pre-Windows 7 Support

davidpgil, 2020-04-18:
What would be involved in order to support Windows versions before Windows 7? I have an offline Windows XP computer that I would love to use with the UHK.

mondalaci, 2020-04-18:
This issue may end up being a duplicate of #287.

In any case, can you check in the meantime whehter you can reproduce this on multiple Windows XP computers, preferably with different hardware configurations? I have doubts that it's universally reproducible on Windows XP.

davidpgil, 2020-04-20:
I just realized that if I enable USB in my Award BIOS, the UHK works and Windows XP detects it. My issue is resolved. I do believe that I have tried other BIOS and have had issues with the keyboard being recognized... Perhaps the real issue is to make UHK work with BIOS that don't really like USB. Probably not on the menu! :) Anyways sorry for the trouble.