UltimateHackingKeyboard/firmware #299: Add support for making arbitrary keys "sticky"

matthewwardrop, 2020-05-26:
Hi all!

I'm not sure if "sticky" is the correct technical term for this, but when you press <Mod> + <Key mapped to <Tab>>, and then release the <Tab> key while still holding <Mod>, the window switcher remains open (on most desktop environments). I would like the ability to have the same behaviour for any key. In particular, <Mod> + <key mapped to <alt> + <`>>, which on the GNOME desktop allows you to cycle between windows of the same application. When using <Alt> + <`>, things work as expected when you release the <`> key (the window switcher remains open); but I cannot seem to configure the <Mod> + <key mapped to <`>> to have the same behaviour. The switcher is not sticky when holding down <Mod>.

Perhaps I am missing something?

mondalaci, 2020-05-26:
Hey there!

The set of sticky modifiers are [hardcoded](https://github.com/UltimateHackingKeyboard/firmware/blob/2f7837d83927695ec1e742abb9050e8adac1e1b5/right/src/usb_report_updater.c#L103-L123) in the firmware for the sake of simplicity.

@kareltucek Would you make this configurable in your [UHK firmware work](https://github.com/kareltucek/firmware)?

kareltucek, 2020-05-26:
Well, I sure can distinguish three global strategies (none, smart, all) rather than the current two (none and official). 

Allowing it on a per-key basis is also possible, but definitely more complicated and probably with too much of the "benefit to user-unfriendliness" tradeoff.

Matthew, what do you think about it? Would an option to enable stickiness on all non-base-layer modifiers suite your usecase well?

Also, could you please clarify the usecase? The un-intentional markdown has made the original post pretty hard to interpret.

mondalaci, 2020-05-26:
@kareltucek I'd like to be able to transfer such issues to https://github.com/kareltucek/firmware

Wondering if I could do it if you gave me write access to your repo.

kareltucek, 2020-05-27:
Sure! Try now.

matthewwardrop, 2020-05-27:
@kareltucek Sorry. I fixed the markdown rendering in the initial post (can't believe I missed it after posting initially!). Let me know if the initial post remains unclear.

From a configurability perspective, I really like the idea of making things sticky on a per-key basis (there are other combinations which I would rather not be sticky; such as volume commands or arrow keys).

mondalaci, 2020-05-27:
@kareltucek For some reason, GitHub doesn't offer me your repo regarding the transfer, even after accepting your invitation. Feel free to open an issue in your repo and continue this conversation there.

Given @matthewwardrop's answer and the reoccuring nature of such requests, I think the per-key stickiness configurability is justified, so feel free to develop such a feature, @kareltucek.

On a related note, I'm thinking about enabling Agent to use a specified repo for firmware updates and automatically offer updates of new firmware releases. This way, one would be able to pick the Karel repo, and keep using newer versions of your fork.

matthewwardrop, 2020-05-27:
@mondalaci The Agent feature sounds great :).

On a semi-related note, is @kareltucek work a staging ground for features that might eventually hit this repo; or is it decidedly outside of the scope of the official firmware?

mondalaci, 2020-05-27:
@kareltucek's firmware contains loads of advanced features that I don't plan to merge into the official firmware because these features are rather niche and harder to configure than simpler features. Lucky us, he also works on parts of the official firmware that are really hard to comprehend, such as secondary role resolution. :)

kareltucek, 2020-05-28:
> Feel free to open an issue in your repo and continue this conversation there.

Sure, lets continue at kareltucek/firmware#50 .

kareltucek, 2020-05-28:
I am now browsing the shortcuts https://help.gnome.org/users/gnome-help/stable/shell-keyboard-shortcuts.html.en, which features multiple alt+tab like shortcuts, which are probably quite standard in some parts of linux world, so maybe it might (or might not) make sense to extend the default set of sticky shortcuts too.

(Of course, adding this option to the fork is a reasonable request all the same.)

mondalaci, 2020-05-28:
Thanks for the tip, but I rather wouldn't extend the set of sticky shortcuts unless people really want them because I'm afraid that we end up sticking shortcuts that many don't want to be sticky.

I'm closing this issue, and feel free to continue the discussion in the new issue.

closed 2020-05-28 by mondalaci