UltimateHackingKeyboard/firmware #300: Save Mouse Speed as part of the Layout

m42e, 2020-06-16:
I have the keyboard sometimes connected to my mac, sometimes to my windows device.
I saw a huge difference in how the mouse speed is interpreted in both. 
Would it be possible to save or overwrite the mouse speed in the current active keymap? This would really help me a lot.

Thanks.

mondalaci, 2020-06-16:
@m42e How about adjusting the mouse speed on Windows / Mac instead?

m42e, 2020-06-16:
@mondalaci This would affect all other mouse-devices, too. and then either all are very slow or very fast. I mean it is not that much of a deal, but it would be anice feature.

mondalaci, 2020-06-16:
Gotcha. I think it's a valid use case which was requested by others too, so we'll make it possible to override global mouse speed settings on a per keymap basis. This will require substantial development regarding Agent too, and will take a while to implement.

m42e, 2020-06-16:
I may be able to provide a PR for the change of the firmware as a proposal. But I am not too much into the agents stuff.

mondalaci, 2020-06-16:
Thanks, but the firmware side is much dependent on Agent, and the on the updated configuration structure in specific. We'll handle this.

steamraven, 2021-04-29:
I like the idea of moving a lot of the configuration into the keymap settings. This allows you to easy switch between preset settings.

mondalaci, 2021-04-29:
This is a relatively niche feature, but it'll be possible via [smart macros](https://github.com/UltimateHackingKeyboard/firmware/issues/351).

mondalaci, 2022-03-19:
Implemented via smart macros, will be included in the next firmware release.

closed 2022-03-19 by mondalaci