UltimateHackingKeyboard/firmware #302: Feature request: Toggle key state

justinian, 2020-06-22:
When gaming with my UHK, I often find myself holding down certain keys for extended periods (e.g. Shift and/or W) which is not very friendly to my RSI. I've tried adding macros to hold down keys for me, but the best I can do is: Press key, delay N seconds. The key will be released when the macro ends, even without a Release Key action.

Would it be possible with the current firmware design to have a macro toggle or invert the state of a key such that the UHK reports the key as being held down until it is again toggled off?

mondalaci, 2020-06-22:
This is not planned, but please take a look at https://github.com/kareltucek/firmware which may implement something like what you're looking for.

closed 2020-06-22 by mondalaci

justinian, 2020-06-22:
Thanks, I'll take a look!

kareltucek, 2020-06-22:
There is no straightforward way to do that even on the fork, since in order to keep a key pressed you need to keep its activation macro alive. To do that, you can use an active wait loop in combination with register signalling, but it will not be trivial to set up...

justinian, 2020-06-22:
Thanks @kareltucek! I'll take a look at the code in your fork and try to get an understanding for it first.