UltimateHackingKeyboard/firmware #304: Layer switching too quick

robertofarrell, 2020-06-24:
I map a layer change to the right space key when held and I find that when I'm typing quickly that I accidentally trigger keys on the function layer.  There used to be a longer delay when triggering layers by holding keys which avoided this issue but now that its quite short this has become an issue.

Is there any way this delay could be exposed in the agent?

mondalaci, 2020-06-24:
@robertofarrell I don't recall shortening the layer switch delay. Which was the latest firmware version which worked according to your expectation in this respect?

robertofarrell, 2020-06-24:
This is a great question and I attempted to determine this yesterday without any luck.  I rolled back as far as 8.7.1 without being able to replicate it.  I'll investigate further today.

kareltucek, 2020-06-24:
If we are talking about really prehistoric releases, then there have been two significant changes in 8.4.3 and 8.3.3 (the first was a fix of timer, which incidentally broke doubletap locks (and possibly other) delays; the second one fixed them again)

robertofarrell, 2020-06-24:
Reverting back to 8.6.0 seems to resolve the issue so I wonder if its related to the debouncing changes that were introduced in 8.7.0.

kareltucek, 2020-06-24:
I doubt that. But I see there a change in secondary role behaviour, which might theoretically be involved.

What is your exact scenario? And what is the mapping - is it a secondary role switcher?

mondalaci, 2020-06-30:
Closing due to the lack of replies.

closed 2020-06-30 by mondalaci