UltimateHackingKeyboard/firmware #306: Modifiers as secondary role don't trigger with external devices

jceb, 2020-06-29:
I'm running firmware version 8.10.0 and I like the space cadet shift key overloading functionality. I have both Shift keys configured to produce an open/closed brace when tapping the key and Shift when holding the key. This setup works great with the UHK, however it breaks down when combining it with an external mouse.
When I try to create a Shift-MouseClick just the mouse click is registered while the secondary role Shift key is not recognized by my Linux OS. I debugged the issued with the qmk keyboard tester (https://config.qmk.fm/#/test) and found that the secondary role is only triggered when I press another key on the UHK. While this behavior matches the description it's unfortunate. Qmk manages to get the job done in a different way - this issue doesn't exist there. It would be great to see UHK solve this issue.

kareltucek, 2020-06-29:
Well, secondary roles can work two ways:

- issue mod always, and produce primary action if no other key was pressed
- wait to see if another key is pressed and then issue the mod if another action was actually pressed

Uhk chooses the second one. The problem is that the mod itself can have a side effect in the OS/application (e.g., win/gui key have a side effect in most window managers), which is sometimes desired and sometimes not. I.e., both defaults are broken in some cases, and UHK and QMK just use different default. 

With UHK, you can get your usecase working with this fork https://github.com/kareltucek/firmware using the `ifInterrupted` or `ifNotInterrupted` condition.

mondalaci, 2020-06-29:
@jceb I don't plan to make this feature an integral part of Agent, but chances are fair we'll eventually merge @kareltucek's glorious achievements into the official UHK firmware.

I'm closing this issue, but feel free to follow up regardless.

closed 2020-06-29 by mondalaci

jceb, 2020-07-02:
@kareltucek thanks a lot for the hint. I didn't make the connection in my mind that your firmware offers this functionality. It works great!
@mondalaci :+1: