UltimateHackingKeyboard/firmware #308: Asus ROG X570 doesn't work with the keyboard

ANtlord, 2020-08-13:
Hello! I've got a new desktop. It has motherboard Asus ROG strix X570-I gaming. The problem is that the keyboard doesn't work in BIOS as well as in live USB of Linux (Fedora 32) and Windows 10. The keyboard works fine with my laptops in Linux, Windows, OSX and in BIOS.

I tried my old keyboard with the new machine then it works fine. After reading a couple of issues I got that a cause could be related to secure boot. I tried to disable it but doesn't help. Then I updated firmware of the keyboard to 8.6.0. The version is available in my UHK agent so it doesn't help too.

When the keyboard is connected to the machine its lights are on and it is seen in USB device list in Linux (lsusb command I mean). The BIOS recognises the board too as it show 2 keyboards when I connect the UHK and my old one. I'm able to change layouts and get indication of the active layout but nothing happens when I write text or use arrows.

How can get my UHK to work within the new machine? Please feel free to ask any additional technical information as I have no idea what to write else.

mondalaci, 2020-08-13:
Hi there! Does [this knowledgebase query](https://ultimatehackingkeyboard.com/knowledgebase?q=ryzen) help?

ANtlord, 2020-08-13:
Updating firmware to 8.10.1 fixes the issue. I looked for a motherboard in the knowledge base but not a CPU. Thank you very much and thank you for the keyboard!

closed 2020-08-13 by ANtlord