UltimateHackingKeyboard/firmware #309: Occasional delay when typing on MacOS

suneetshah, 2020-09-20:
I purchased the UHK in May and I am using it with a Mac.
There are some apps, likes  Microsoft Teams where I frequently see a delay before the typed characters appear

Would you know what is causing this and how it can be resolved?
I was using a DAS 5Q earlier did not have similar issues.

mondalaci, 2020-09-20:
Please flash firmware [8.7.0](https://github.com/UltimateHackingKeyboard/firmware/releases/download/v8.7.0/uhk-firmware-8.7.0.tar.gz) and [8.7.1](https://github.com/UltimateHackingKeyboard/firmware/releases/download/v8.7.1/uhk-firmware-8.7.1.tar.gz) via Agent, and let me know which one resolves this issue, if any.

suneetshah, 2020-09-20:
I am running 8.10.1 which can make agent 1.5.4

Should we move to an older version?

Firmware 8.10.1 is running on the left keyboard half.
Firmware 8.10.1 is running on the right keyboard half

mondalaci, 2020-09-20:
Yes, please do.

mondalaci, 2020-10-13:
Closing issue due to the lack of follow-ups.

closed 2020-10-13 by mondalaci

theguly, 2020-11-09:
hi there, i'm reopening this issue because i have a similar one that could be related. let me know if you prefer a new issue.

i'm running macos 10.15, running with a displaylink (https://displaylink.com/) external adapter to run the third monitor.

i had issue with "unordered, missing or repeated" keys lately and delay was just a secondary issue to me (i experience a delay of ~1sec not so often,. these issues are mostly, but not only, present when i type on monitor handled by displaylink
let me explain better: writing on an app running on main monitor this happens rarely, but writing on an app running on the displaylink connected one she receive wrong keys at a rate of i think 10% (in the "problematic" keyspace range, as a rule of thumb, this mostly happens with e r t d keys, mostly with r (could be a failure with the key itself?)).

an example of wrong keys is: typing "prova prova" i sometimes get a "prorva prova" or "sembra" become "semba" and again "tastiera" is "tastierar".

i specified that i'm also using displaylink because i feel that this issue is more present with their legacy driver and looks like more present on the monitor connected via displaylink (like 2 times more frequent than on the monitor connected using builtin thunderbolt).

another thing i observed, i could be wrong, is that when i keep the keyboard unpowered for some time wrong keys are way less, the rate increase after some (2?) hours of working.

i tested the keyboard also on another laptop running linux, with no UHK installed, and i don't have any issue but i tested it for 10/15 minutes or so.

i tried to downgrade firmware from 8.10.1 to 8.7.1 and 8.7.0 with apparently no benefit.

i also tried to change usb cable with no improvement, i don't have a spare RJ cable to test but if you think it could be a valid test i'll find one (i observed that mostly problematic keys are on the left side of the keyb, who knows).

i'm going to try new release 8.10.2 to see what happens, let me know how i can further debug this issue.

thanks

mondalaci, 2020-11-09:
@theguly Thanks for elaborating on this issue!

@kareltucek Any ideas about the solution?

kareltucek, 2020-11-09:
Not much - maybe to have xxxajk have a look.

To me, this looks as a usb/host specific issue. 

--------------------------------
Some irrelevant thoughs:

> i tested the keyboard also on another laptop running linux, with no UHK installed, and i don't have any issue but i tested it for 10/15 minutes or so.

Testing some more could rule out that it is actual internal failure of the keyboard.  If you cannot reproduce any of the problems with another PC, you can be quite sure that internal logic of the firmware is fine, as well as are your switches and RJ cable/I2C.

> (could be a failure with the key itself?)).

Yes ( https://ultimatehackingkeyboard.com/repair-and-modding/fixing-misbehaving-keys ), except if it was a failure with the key itself, it would have occured on all your monitors and all your PCs.

> as a rule of thumb, this mostly happens with e r t d keys, mostly with r

These are all left-half keys, which might mean IC2 fails caused by elmag interference on the bridge cable. You can try using the UHK joined without the bridge cable. But again, if this was UHK fault, occurences would be stable across PCs.

> i observed that mostly problematic keys are on the left side of the keyb, who knows

Most, or all?

> another thing i observed, i could be wrong, is that when i keep the keyboard unpowered for some time wrong keys are way less, the rate increase after some (2?) hours of working.

Reminds me of one issue I encountered when soldering switches - some leftover resin shortcircuited the board and apparently caused some capacitator to charge slowly over time (days, not hours), which at some point caused some keys to act as if stuck. But again, any hardware fault if this sort would probably be stable across different PCs...

theguly, 2020-11-09:
> To me, this looks as a usb/host specific issue.

this scares me :)

> Some irrelevant thoughs:
>
>
> Testing some more could rule out that it is actual internal failure of the keyboard. If you cannot reproduce any of the problems with another PC, you can be quite sure [...]

I'll try to use another PC for some hours this week/next week end.

> > i observed that mostly problematic keys are on the left side of the keyb, who knows
> 
> Most, or all?

mostly left side (also o has issues, not as frequent as r key).
and, to be clear, mostly on displaylink connected monitor,  but also on thunderbolt connected one (looks like 8.10.2 makes both monitor to act similar in term of double/missing key hit, adding more entropy...).
of course based on my observation, far from being "scientific".

> Reminds me of one issue I encountered when soldering switches - some leftover resin shortcircuited the board and apparently caused some capacitator to charge slowly over time (days, not hours), [...]

resin should be left on lot of keys, at least 5, i would be very unlucky if this happened to my UHK.

I'll try to make sure that i can't replicate this issue on another PC before any other move my side. will update this issue as soon as i can, probably not before the next monday.
I'll also try to remove displaylink driver to make sure it's not she causing this issue.

meanwhile let me know if you need more information

thanks

kareltucek, 2020-11-09:
Anyway, can you try to debug the problem in your OS? E.g., checking that system sees the keyboard, there are no driver failures, etc.. Not sure what is macos's counterpart of `dmesg`. 

Also, if the agent led lights up for a second at random (without Agent running), it might mean that keyboard is getting resetted for some reason (power supply?).

theguly, 2020-11-14:
after some more thoroughly test (on another laptop, with/without usb hub, with/without thunderbolt hub, with different firmware, having the keyb powered for hours/days, etc) i looks like that canned air breeze solved the problem.
i'm surprised because, for example, using my UHK on another laptop this week showed this very issue, using the keyb 3/4 weeks ago didn't, and (this never made sense to be honest) external monitor had a higher rate of "failure".

coincidence or maybe the dust or dirt wasn't stuck, who knows, i'll never know :)

anyway thanks for your help, and sorry for having used this issue as scratchpad

mondalaci, 2020-11-14:
@theguly Thanks for sharing, and super glad your UHK works again!