UltimateHackingKeyboard/firmware #310: Switching windows of application in Gnome Shell 3.36

jsporna, 2020-10-12:
I have configured:
Mod + D as Super + Tab (switching application)
Mod + \` as Super + \` (switching windows of application)

Description of keyboard behaves:
I pressed Mod and pressing D until I reach desire application next keep having pressed Mod button I'm pressing \` to switch windows of application. I've observed keyboard is sending scancode of release of Super button after release of \` button.

Screen of `showkey`.
[image]

Screen of `configuration`
[image]

Agent: 1.5.4
Firmware: 8.10.1 & 8.10.2
OS: Ubuntu 20.04 5.4.0-48-generic

mondalaci, 2020-10-12:
@kareltucek I guess we can make Super+` sticky, although I'm wondering how prevalent this shortcut is.

jsporna, 2020-10-12:
@mondalaci 
Super + \` is default shortcut in Gnome Shell for `Switch windows of an application`

[image]

mondalaci, 2020-10-12:
@jsporna Thanks for letting us know!

@kareltucek Please let's make Super+backtick sticky unless you have objections.

luteijn, 2020-10-12:
I'd still really like to see a general solution to this instead of yet another 'sticky' exception to keep in mind. Some way to leave keys explicitly held or explicitly release them when wanted. I know the USB HID interface doesn't really use key down/key up messages in the same way that ps/2, AT, XT or even MIDI serial keyboard interfaces do (making proper n-key roll-over possible for those protocols), but software and wet-ware still expects it to more or less work that way. Having a way for the UHK to emulate this behavior to at least as much as can be done within the USB protocol still feels really useful.

PR #311 (2020-10-13): Make mod plus backtick sticky
Closes #310
closed 2020-10-13 by mondalaci