UltimateHackingKeyboard/firmware #313: Holding Shift key is intermittent

proullon, 2020-10-23:
Before submitting a new issue, make sure to do the following:

1. On Windows 10
2. Agent v1.5.4
3. Firmware 8.10.2 on both half
4. The issue: Shift key not working properly.

Hi, I have an issue with my shift key not working properly. The "holding" mechanism is alternating on/off several time per second. As an example, here is the output where I hold shift key and press w repeatedly:
WWWwwwwwwwWWWwwWWwwwWWwwWWWwWWWwWWWW

The issue is clearly visible when playing a videogame with the 'crouch' set on Shift key, the character change position several times per second.

It's obviously a crippling issue since I have to sometimes try 3/4 times to enter a single character, I don't know how to address it, any idea ?

No issue with 'fn' or 'mod' though.

Cheers

kareltucek, 2020-10-23:
This looks like a switch issue. (If you want to test whether it is indeed a switch issue, try swapping the shift with another key in Agent and use that.)

Following https://ultimatehackingkeyboard.com/repair-and-modding/fixing-misbehaving-keys should help

proullon, 2020-10-23:
ah yes it looks like a physical switch issue, my bad.

Thanks, and sorry for the noise

closed 2020-10-23 by proullon

kareltucek, 2020-10-23:
No problem!