UltimateHackingKeyboard/firmware #314: USB reset happens frequently on linux with firmware 8.10.1

lo48576, 2020-11-01:
## Environment
* Gentoo Linux, kernel 5.9.2
    + Also with 5.8.17, same problem happens.
* UHK is connected to USB 2.0 port, and XHCI driver is used (xhci_pci, xhci_hcd)
* UHK firmware 8.10.1

## Expected result
UHK works normally.

## Actual result
With linux kernel built with `CONFIG_HIGH_RES_TIMERS=y` **OR** `CONFIG_NO_HZ_IDLE=y` (or both), UHK is frequently disconnected (i.e. key inputs are ignored intermittently) and `dmesg` reports errors:

> [  286.223211] usb 1-10.3: reset full-speed USB device number 7 using xhci_hcd

[image]

## Other thoughts

### Workaround
If you are suffering from this, build the linux kernel with `CONFIG_HIGH_RES_TIMERS=n`, `CONFIG_NO_HZ_IDLE=n`, **AND** `CONFIG_HZ_PERIODIC=y`.

### Cause
I'm almost sure #269 caused this problem.
(I tried to confirm that, but I couldn't. The USB reset happened during firmware flashing, and left half of the keyboard seems to be bricked :rofl:)

lo48576, 2020-11-01:
I plugged the keyboard to another laptop, and then the error does not seem to happen, even with `CONFIG_HIGH_RES_TIMERS=y` and `CONFIG_NO_HZ_IDLE=y`...
I don't know what to check more.

mondalaci, 2020-11-02:
What happens if you connect your UHK via a USB 3 port or a powered USB hub?

lo48576, 2020-11-03:
Powered USB 2 port (connected to USB 2 port of M/B): problem still happens.
USB 3 port: The problem doesn't seem to happen.

Unplugging from USB 3 port, this warning is reported by `dmesg`, but I don't know this is related.

> [74359.445614] xhci_hcd 0000:02:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.

[image]

mondalaci, 2020-11-03:
Are you using a Ryzen PC?

lo48576, 2020-11-03:
Yes, using Ryzen 7 3800X.

mondalaci, 2020-11-03:
@xxxajk Shouldn't #301 fix USB 2 issues like this, too? Is there any way to fix this?

Jopie01, 2021-02-22:
Just to let you all know, I came across this: https://www.reddit.com/r/Amd/comments/lnmet0/an_update_on_usb_connectivity_with_500_series/. So it seems this issue is a problem from AMD and not UHK or Linux.

xxxajk, 2021-02-22:
Update your bios, which should have the microcode fixes. Solved a lot of
the problems for the USB and other bugs.

lo48576, 2021-02-22:
I know this is hardware problem.
I googled about that AFTER the left half of the UHK was broken), and I noticed this sometimes (infrequently) happens on other USB devices.

However, the disconnection of UHK became quite frequent (a few times in a minute!) at some point, so I think this could possibly be mitigated by the Linux kernel or UHK firmware side.
(I'm sorry I don't know when and why the disconnection became frequent on my machine.)

I alreday tried UEFI updates, but they haven't solved the problem.
I'll try newer UEFI firmwares from now on...

Thank you for your information!

xxxajk, 2021-02-23:
One other thing to consider is a newer kernel version. Currently I'm running 
Linux devil 5.10.12-051012-generic #202101301330 SMP Sat Jan 30 14:07:49 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
which fixes a ton of stuff too. My UPS would drop USB constantly, now it is much better with new BIOS and kernel.

mondalaci, 2021-08-03:
A lot of improvements have been made to the USB handling of the firmware recently, so chances are this issue has been fixed, assuming that it was a firmware issue and not a BIOS-related one.

I'm closing this issue but if anyone thinks that it still persists, feel free to comment and I'll reopen it if needed.

closed 2021-08-03 by mondalaci