UltimateHackingKeyboard/firmware #315: Changing brightness via key combination

polarkac, 2020-11-11:
It would be quite useful to change brightness without requiring to start Agent and change it there. 

Is this feature possible or at least planned?

mondalaci, 2020-11-11:
Thanks for explaining! This is more like a firmware feature.

@kareltucek Does your firmware fork support this use case?

kareltucek, 2020-11-11:
No - I don't even remember anyone asking for this before. But adding it should not be a problem...

polarkac, 2020-11-11:
I guess its not that requested feature because UHK v1 is not backlit and many users will not solder the LEDs. With v2 it could be requested more but I am just guessing.

If I am the first to ask this I think its pointless to implement it. Maybe just add it to TODO list as very low priority feature?

mondalaci, 2020-11-11:
Good point. We'll see how popular this feature will get when it comes to backlit UHKs. In the meantime, it'd be great if [Karel's UHK firmware fork](https://github.com/kareltucek/firmware) supported, which I plan to merge into the main UHK firmware eventually.

kareltucek, 2020-11-11:
>  guess its not that requested feature because UHK v1

I was wondering which of the three controls (led display, layer indicators, backlight) we were talking about, but figured it had to be about those that are already on the market. Still, w.r.t. support in the fork, its all the same.

So since you don't care about current LEDs, then I guess I will be dealing with backlight-related stuff once the hardware and relevant agent/firmware support are released.

(@mondalaci Just to make sure - am I right in assuming that Agent-side support for backlight is being developed on some private branch in the meantime, and therefore the single backlight slider (visible in 1.5.5. agent) is not all we are going to get? (-: )

> Good point. We'll see how popular this feature will get when it comes to backlit UHKs. 

Still, it might be a good idea to think about serialization formats now, because that's what makes the difference between possible and impossible.

mondalaci, 2020-11-11:
The UHK 60 v2 firmware which was forked from v1 lives in a private repo that I'll merge into the firmware repo soon. Both the v1 and v2 firmware uses the same codebase and branches with ifdefs to build two separate firmware images.

As for Agent, no support is required for backlighting because backlighting simply works based on the key actions. We will make the colors configurable though, so a relevant feature will be added to Agent and the configuration format will be extended.

mhantsch, 2020-11-13:
> Good point. We'll see how popular this feature will get when it comes to backlit UHKs.

I own some backlit keyboards. All of them allow you to adjust the backlight through some keyboard shortcuts.

I think this is a very valid and useful use case. The ambient light changes, and you want to adjust the backlight intensity. Right from the keyboard, while you are sitting and typing. Should be as easy as a few key presses. I don't always run Agent. Heck, it doesn't even run on my Chromebook. I have to unplug the keyboard, plug it into a Windows laptop, then reconfigure. Then I can plug it back into the Chromebook and continue my work. All of this just to adjust the backlight?

mhantsch, 2020-11-13:
Oh, one additional comment: all of my Laptops that have backlit keyboards also allow me to adjust the intensity through some Fn-Key combinations. I think this is a must.

mondalaci, 2020-11-13:
Do you guys expect the UHK to remember the brightness setting when setting it via shortcut keys?

mahmoudhossam, 2020-11-14:
@mondalaci As a future UHK v2 user (I pre-ordered mine) I'd expect my keyboard to remember its own settings every time, regardless of how it was set up.

kareltucek, 2020-11-14:
@mondalaci My strongly biased but not particularly strong expectation is: no.

mondalaci, 2020-11-14:
@kareltucek I'm unsure about how it should behave, but making the UHK not remember the backlight setting would be so much easier.

TorC8, 2020-11-21:
Backlight brightness may be constantly adjusted.  I think mostly by how much the room brightness changes.  Someone who uses the UHK on an office desk probably doesn't change it much at all, but someone who uses it in various places will likely change backlight brightness every other use.

If I could design the brightness settings however I'd like, trying to maintain a balance between simplicity and power users:

Have the backlight brightness adjustable on the UHK.  The current setting should be remembered across a power cycle.  Agent knowledge of backlight brightness setting not needed.  Info LED brightness should have floor and ceiling levels set by agent, UHK auto-scales between these boundaries according to backlight brightness.

If, at a future time, agent is capable of detecting screen brightness and/or has access to an ambient light meter, then it may be of some use to have agent able to adjust backlight brightness.  Until then it's just a complication.

If backlight brightness is exposed to agent, resetting backlight on config save to keyboard may seem odd, but isn't a big deal.

BastiOfBerlin, 2021-12-09:
*Pushing this request again*

Also, or me, it would be okay to have a shortcut to toggle on/off the backlight.

My use case is that I am using a KVM switch which always keeps the keyboard powered (for emulation reasons, I suppose). Being able to toggle off the backlight in the evenings would be a useful feature.

bzed, 2022-03-27:
Same here, although I'd prefer to ba able to change the brightness. During the day I need/want a bright keyboard, so keys are properly visible in daylight - but when I'm on call and I have to get up in the middle of the night, the bright keyboard just burns in the eyes ;) A scriptable way would be fine for me, too, so I can dim the keyboard automatically.

mondalaci, 2022-03-27:
Closing as a duplicate of #437.

closed 2022-03-27 by mondalaci