UltimateHackingKeyboard/firmware #316: Feature request: same mouse speed on diagonal

vmauery, 2020-11-16:
Would it be possible to change the speed of the vertical and horizontal speeds to be 1/sqrt(2) * X and 1/sqrt(2) * Y so that the composite speed will track the same as only X or only Y?

I feel like when I am pressing both X and Y at the same time the mouse is moving too fast. This could default to the current behavior but in the mouse keys dialog, there could be a setting or something that allows diagonal speed to be the same speed as vertical and horizontal instead of 41% faster.

kareltucek, 2020-11-16:
This has already been discussed in  #256 . TL;DR: Until someone persuades Laci to add it officially and until it actually gets done, you can try out https://github.com/kareltucek/firmware (for details see https://github.com/UltimateHackingKeyboard/firmware/issues/256#issuecomment-619526425 ).

vmauery, 2020-11-16:
didn't see that one. Thanks for the response. I will just go ahead and close this since it is a duplicate.

closed 2020-11-16 by vmauery

vmauery, 2020-11-16:
@kareltucek - I tried out your firmware with this feature and like it. Thanks for making that available!

@mondalaci - The way that @kareltucek has this feature implemented is nice. If you don't like it, you don't set the flag. I like it, so I set it.