UltimateHackingKeyboard/firmware #317: Duplicate keypresses

loiclacombe, 2020-11-19:
I have been getting random duplicate keypresses for a few days on Windows 10 20H2. To reproduce, I just have to type in any text field.
I am using firmware version 8.10.2, but I had the saame problem with the previous version in Agent

mondalaci, 2020-11-19:
Please see https://ultimatehackingkeyboard.com/knowledgebase?q=duplicate and let us know if the issue persists.

closed 2020-11-19 by mondalaci

loiclacombe, 2020-11-19:
I tested when the two parts of the keyboard were joined, and it worked right away. I could not reproduce the issie. Then I split back the keyboard and got some more duplicated key input issues.
Then. I rolled back to firmware 8.9.3, with split keyboard and I am not getting any issues anymore. It could be just a random correlation if it is a connectivity/mechanocal issue. So I will tell you if any funky issue arises again.

mondalaci, 2020-11-19:
Thanks for elaborating on this! I don't think the firmware version should affect this issue. Please shoot us an email to support at ultimatehackingkeyboard dot com if you encounter further issues.