UltimateHackingKeyboard/firmware #327: Accidental layer toggling

mondalaci, 2021-01-19:
If I double tap a layer switcher key, which has the "lock layer when double tapping this key" option enabled, and I hold it for a while, then release it, it doesn't get toggled, which is correct behavior.

However, If I double tap such a layer switcher key, hold it for a while, but at the same time press some other keys, and then release it, it gets toggled, which is not desired, and from time to time, I toggle layers accidentally this way.

kareltucek, 2021-01-20:
This applies to simple layer "toggle" too. 

Is the behaviour considered correct for toggle, or should it be changed too?

mondalaci, 2021-01-20:
I'd expect single-tap toggle actions to toggle layers under all circumstances.

PR #357 (2021-04-20): Fix interruptions of doubletap toggle.
Closes #327.
closed 2021-04-20 by mondalaci