UltimateHackingKeyboard/firmware #329: Keyboard locks up and Key press sticks

richClubb, 2021-01-27:
While typing they keyboard will lock up and the last key pressed before the lock-up will stick on for a period of time.
Can't input any other key presses during this time, and can't seem to cancel it. It lasts for about 1-2 seconds.
Running firmware 8.10.3 on both halves.

richClubb, 2021-01-27:
I've aaaaaaaaaaaaaaaaaaao (that actually happened while typing this, was trying to type "I've actually noticed") I noticed that the agent LED lights randomly even when the agent is not open. Not sure if this is related.

mondalaci, 2021-01-27:
I'm sorry to hear about this issue! Please give a try to [these instructions](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys).

richClubb, 2021-01-29:
Cleaning the link cable seems to have helped, sorry for reporting this as an issue.
Thanks for the help :)

mondalaci, 2021-01-29:
So glad to hear! :)

closed 2021-01-29 by mondalaci