UltimateHackingKeyboard/firmware #330: Per-assignment mouse speed adjustments

akanter, 2021-02-16:
One feature I absolutely _love_ is controlling my mouse via keypresses and being able to adjust the speed is crucial.

It would, however, be even more amazing if I could have different keys adjust the speeds different amounts. I'd love to be able to have "+speed" and "++speed" key assignments, and similarly "-speed" and "--speed" key assignments. To that point, it seems as though I should be able to set an arbitrary boost/reduction amount on a per-assignment basis.

mondalaci, 2021-02-16:
I'm sorry, but we don't plan this feature because it's not widely requested. The closest we plan is #300.

closed 2021-02-16 by mondalaci

akanter, 2021-02-27:
@mondalaci thanks for the response! So if I find enough other people to request it then will you consider it? =D Where would such requests be made? Would y'all be open to a pull request?

mondalaci, 2021-02-27:
You'll be able to create let's say 3 keymaps with different speed settings, and switch between them with an arbitrary shortcut such as Fn +/-

akanter, 2021-03-01:
The functionality I'm looking for is more like:
mousing around, press decelerate key for precision, then shift+decelerate (or a different key) for yet more precision.

If I could bind the keymap switch events to macros and/or keydown+keyup events then I think the keymaps would work but since that's not the case I'm dubious.  It seems to me that both of these features might be more straightforward to implement if you could just bind the acceleration values per key instead of per keyboard or keymap.

mondalaci, 2021-03-02:
I'm sorry, but we don't plan to implement your use case because it's very niche.

Feel free to check out [Karel's UHK firmware fork](https://github.com/kareltucek/firmware) which might contain a solution.

kareltucek, 2021-03-02:
It does not, but PRs and generally contributors are welcome :-). 

(I can give pointers whereto look and explain how stuff works - just don't hesitate ask ;-).)