UltimateHackingKeyboard/firmware #332: Tab backlight

kareltucek, 2021-03-04:
This does not look entirelly right...
[image]

mondalaci, 2021-03-04:
I knew about this bug, and I've just partially fixed it with b1f858c28692f41a7ada7d5dc969be3530129107, but for some reason, the Tab and Backspace LEDs are still affected on the Mouse and Fn layers.

kareltucek, 2021-03-05:
I think I see the problem, will confirm later and hopefully also post a PR.

steamraven, 2021-03-14:
LedMap: line 79
        { .red=0, .green=16, .blue=29 }, // Tab

I think this should be

        { .red=0, .green=16, .blue=32 }, // Tab

since 29 is also used with G green.

Also, 
        { .red=152, .green=168, .blue=0 }, // Backspace
Based on the pattern it looks like it should be 
        { .red=152, .green=168, .blue=40 }, // Backspace

kareltucek, 2021-03-14:
Indeed! Works with both custom backlight as well as with the functional backlight, and does not mess up led display text!

Will you create a PR so that the contribution rightly remains logged as yours? :-)

steamraven, 2021-03-14:
Can we confirm if backspace is also incorrect?

kareltucek, 2021-03-14:
Yes, and your proposed fix works as well.

commit ede30ed2 (2021-03-14): fixes #332 tab backlight (#337)
closed 2021-03-14 by mondalaci