UltimateHackingKeyboard/firmware #342: use of LVFS (fwupd)

dustymabe, 2021-04-05:
I'm a very happy user of my Ultimate Hacking Keyboard (one of the original to fund the development). Thank you for the hard work in making a great keyboard!

Considering the rise of the [Linux Vendor Firmware Service](https://fwupd.org/) could we consider (or have we considered) enabling firmware updates of the keyboard through the LVFS as well? If this is already supported and I missed it please forgive me.

Thanks again!

mondalaci, 2021-04-07:
Thank you for the nice words, and so glad you're happy!

If the UHK had a single firmware, I think LVFS was feasible, but it has two keyboard halves and multiple modules, each running their own firmware, making the firmware update process so much more complicated. Agent will offer per-module firmware updates when encountering modules that run outdated firmware. I can't see a way to solve this scenario with LVFS.

closed 2021-04-07 by mondalaci

dustymabe, 2021-04-09:
> Thank you for the nice words, and so glad you're happy!

Thank you for the excellent keyboard and for making it open source!

> If the UHK had a single firmware, I think LVFS was feasible, but it has two keyboard halves and multiple modules, each running their own firmware, making the firmware update process so much more [...]

Interesting. Thanks for the insight. Would you be open to working with someone who knows more about LVFS/fwupd? Even if it doesn't lead to you using LVFS it might help them understand a new problem space and work toward supporting it in the future.

mondalaci, 2021-04-09:
Yes, I'm open to it, but our use case is quite niche, and we don't plan to support LVFS because our firmware support mechanism is already complex.

hughsie, 2021-04-12:
> I can't see a way to solve this scenario with LVFS.

Ohh we have that kind of thing already. Something like a single Lenovo dock is really about 12 different devices _in a trenchcoat_, each with intra-device and inter-system dependencies. I'd be happy to listen about the low level details on how you currently deploy updates if you can share.

mondalaci, 2021-04-12:
Upon firmware updates, Agent directly updates the right keyboard half via USB, then it updates the left half and modules via I2C. The right half acts as a USB to I2C proxy. Agent keeps polling the list of attached modules via USB, and it'll update newly attached modules that run outdated firmware.

hughsie, 2021-04-12:
> Agent directly updates the right keyboard half via USB

Using what protocol? How does the host known the installed firmware version?

> then it updates the left half and modules via I2C

Got any details on the low level protocol between the two halves? Are the images signed?

> it'll update newly attached modules that run outdated firmware.

Also using the i2c bridge?

mondalaci, 2021-04-12:
KBOOT (Kinetis bootloader) is used for updating the right half. Right half firmware version query is implemented via the custom-developed UHK USB protocol. Left half and module firmware version query is implemented via the custom UHK I2C protocol. The UHK protocols are undocumented and can only be understood by reading the source. The firmware images are not signed. BusPal is used to implement the USB to I2C bridge but it requires USB reenumeration which introduces extra delay during the firmware update process, so we'll replace it with a custom USB protocol to talk KBOOT via I2C.

hughsie, 2021-04-12:
@mondalaci it sounds like a good chunk of work, but I'm getting the impression using LVFS and fwupd isn't super interesting for you -- which is fine. I'm guessing government / large companies are not super interesting as a revenue source to you, and that's quite understandable (which means the LVFS requirement of those places isn't a problem). I guess the only other thing we can offer is someone else maintaining the firmware update client for you, the fact you don't have install anything new on Linux and that we mirror the firmware for ever at no cost. @dustymabe isn't the first one to ask about UHK and it would be really good to work together if that's an option.

mondalaci, 2021-04-12:
In all honesty, I think the effort it'd take to get LVFS working for the UHK would hardly worth it, even if someone would help us. Maybe we'll look into it in the far future, but there's way too much on our plates nowadays. If the time comes, I know who to contact. ;)

Nice to know that LVFS is a requirement for governments and large companies. I wasn't aware of that.

dustymabe, 2021-04-13:
Thanks @hughsie and @mondalaci for having the conversation! Super insightful.