UltimateHackingKeyboard/firmware #349: Touchpad tap action

mondalaci, 2021-04-09:
Currently, the tap action of the touchpad is hardcoded to left click. The relevant action of the configuration should be used.

steamraven, 2021-04-09:
The current code applies left click directly.  It could instead set the HardwareSwich state of the approriate key (left) in the module.
I believe this is keyID 1 in slot 3.  Then the applyKeyAction code will pick up the configurable action.  Also, this can change based on active layer or keymap

mondalaci, 2021-04-10:
Sounds good to me. I think touchpad tapping should be implemented in the same manner as the tap key macro action which might be the same implementation as the one you're advising.

piratk, 2021-12-09:
I would like to be able to set actions based on zones (either as an X, or + or perhaps as # shapes) on the touchpad, so that it could end up as an improvised cursor key pad, or have multiple functions, depending on location of tap.

mondalaci, 2021-12-09:
@piratk I'm sorry, but yours is quite a niche use case. We will only implement it if we see substantial demand in the future.

Felixoid, 2022-05-02:
Hi Laszlo, I see that touchpad should process taps with FN and MOD

mondalaci, 2022-05-04:
I have left click mapped to the base layer and right click mapped to the mod layer on my touchpad, and they're working as expected. I'm using the latest development firmware.

Does the above configuration not work for you as expected? If so, which firmware are you using?

Felixoid, 2022-05-04:
[image]

I have mapped left on the base, middle on the mod, and right on the fn layers. And it's left-click everywhere

mondalaci, 2022-05-04:
Then this must have been fixed in the meantime. Please give a try to the next firmware release which should resolve this issue.

closed 2022-05-04 by mondalaci

Felixoid, 2022-05-04:
I'd like to try a dev version, but it doesn't build currently. Drone builds fail since April

mondalaci, 2022-05-04:
@kareltucek How much of a build expert you are? It'd be so handy to have releases generated for every commit. drone.io may or may not be the solution for this.

Felixoid, 2022-05-04:
It complains about unused code https://cloud.drone.io/UltimateHackingKeyboard/firmware/179/1/3 in line 130+

kareltucek, 2022-05-04:
> Hi Laszlo, I see that touchpad should process taps in FN and MOD layers as different mouse clicks. But it doesn't seem working.
> 
> Is it an interface bug or a known issue?

That is exactly what this issue is about. Yes, it is known issue.

-------

Laszlo is probably using development version, which already contains proper handling.

-----

Drone does not compile because of unreachable code disabled via preprocessor directive. It is intentionall, and results in a warning, not an error. The drone just considers all warnings as errors. However, the firmware compiles fine.

------

If you just want to try more up to date code without the inconvenience of building it, feel free to give a try to my fork. It can be more or less considered a staging ground for stock firmware - http://www.github.com/kareltucek/firmware 

---

@mondalaci, not a ci/cd expert. In order to make the drone work again, you just need to add some flag so that it does not consider unused code warning as an error. Or figure a way to disable  a piece of c code without generating the warning

Felixoid, 2022-05-04:
Thanks, just installed https://github.com/kareltucek/firmware/releases/tag/v8.10.12.kt.2, the issue was there until the configuration was updated. But after I've updated touchpad config it's solved.

The issue described in https://github.com/UltimateHackingKeyboard/agent/issues/1794 is not fixed BTW

kareltucek, 2022-05-04:
> the issue was there until the configuration was updated

@mondalaci any idea why the issue might have persisted until configuration update? Flashing right half and restart of the firmware itself should have resulted in reloading the user config.

mondalaci, 2022-05-06:
@kareltucek I don't know. I wondering whether the latter firmware version uses the updated configuration format. Even then, the firmware should branch properly making this a non-issue.