UltimateHackingKeyboard/firmware #351: Smart macros

mondalaci, 2021-04-11:
The binary configuration format of the UHK is very space-efficient, but it’s hard to change, and changing it in the firmware breaks compatibility with Agent. A more flexible mechanism is needed that doesn’t affect Agent and allows for rapidly introducing additional configuration options purely on the firmware side.

When thinking about this issue, the macro-based solution of [Karel’s UHK firmware fork](https://github.com/kareltucek/firmware) immediately came to my mind. I was eager to merge Karel's fork into the stock firmware, but [he discouraged me](https://github.com/UltimateHackingKeyboard/firmware/pull/325#discussion_r584122532), and I think rightly so. The mechanism he used, however, is ideal for this purpose.

This specification uses the present tense for the sake of simplicity, but this feature set has yet to be implemented.

# Macro commands

Text macro actions that start with `$` are treated as macro commands. For example, the `$set module.trackball.navigationMode.base caret` text macro action changes the trackball’s navigation mode on the base layer to caret mode. Commands start with a name, such as `set`, followed by multiple, space-delimited arguments whose number depends on the command name.

Currently, only the `set` macro command is planned.

set ${variable} ${value}

I can think of the following variables:

## Module navigation modes

module.${module}.navigationMode.${layer} = ${navigationMode}
${module} = keycluster | trackball | trackpoint | touchpad
${layer} = base | mod | fn | mouse
${navigationMode} = cursor | scroll | caret | media

## Module speed settings

module.${module}.${speedType} = ${float}
${speedType} = baseSpeed | speed | acceleration

module.${module}.scrollSpeedDivisor = ${uint8}
module.${module}.caretSpeedDivisor = ${uint8}
module.${module}.caretSkewStrength = ${uint8}

## Secondary roles

secondaryRoles.strategy = {simple | timeout}
secondaryRoles.doubletapTime = ${uint16}
secondaryRoles.triggerTime = ${uint16}
secondaryRoles.safetyMargin = ${uint16}
secondaryRoles.allowTriggerByRelease = {1 | 0}

## Mouse keys speed settings

mouseKeys.initialPointerSpeed = ${uint8}
mouseKeys.basePointerSpeed = ${uint8}
mouseKeys.initialPointerAcceleration = ${uint8}
mouseKeys.deceleratedPointerSpeed = ${uint8}
mouseKeys.acceleratedPointerSpeed = ${uint8}

mouseKeys.initialScrollSpeed = ${uint8}
mouseKeys.baseScrollSpeed = ${uint8}
mouseKeys.initialScrollAcceleration = ${uint8}
mouseKeys.deceleratedScrollSpeed = ${uint8}
mouseKeys.acceleratedScrollSpeed = ${uint8}

mouseKeys.[axisSkew](https://github.com/UltimateHackingKeyboard/agent/issues/748#issuecomment-725601903) = ${float}

# Macro events

Macros whose names start with `$` are invoked on specific events. For example, a macro with the name of `$onKeymapChange QWR` is executed upon changing to the keymap whose abbreviation is `QWR`. Event names, such as `onKeymapChange`, are followed by multiple space-delimited arguments whose number depends on the event name.

`$onInit` - Runs on UHK powerup and after saving keyboard configuration, allowing to override default values, such as module configuration.

`$onKeymapChange ${keymapName}` - Runs on keymap changes, allowing for [advanced keymap-specific configuration scenarios](https://github.com/UltimateHackingKeyboard/agent/issues/748).

# Lifecycle and versioning

A feature should have a decent interest to be considered for inclusion as a smart macro.

Once a smart macro is included, it can be promoted to be moved to the user configuration if it’s popular enough. In this case, Agent will expose the relevant configuration value in a user-friendly manner via its user interface.

Alternatively, if a smart macro proves to be largely unused or superseded by another feature, it can be removed.

The smart macro feature is versioned, and versioning is governed by [semver](https://semver.org/). 

Individual firmware releases will feature the smart macro version they implement, and the firmware changelog will include smart macro changes. A well-written, versioned documentation should be provided for smart macros.

mondalaci, 2021-04-15:
@kareltucek Any feedback on your part? You're the king of the hill when it comes to this stuff. :)

kareltucek, 2021-04-15:
Well.

- Generally, it sounds good. However:
- Hijacking your own data fields looks a bit questionable. Ideally, you should extend the macro actions by a new action for commands (that's just a new enum value - space for that is already reserved, no arguments not needed). (And get rid of the ridiculous `$` notation.)
- I like the idea of macro events. The proposed notation is logical, however the spaces are going to make those macros incompatible with fork's "macro calling" features. (Just an observation.)
- Behaviour w.r.t. multiline commands is unclear, and possibly complicated. 
- Plans regarding future compatibility of this proposal with the fork are unclear. If you start implementing your own macro language, you are very likely going to disrupt compatibility of the two to the point where you will effectively loose the benefits which the (currently fully backwards-compatible) fork brings you now. 
  
   I know I have scared you a bit w.r.t. merging of the fork, but I didn't try to imply that you should not merge the fork - more that you should be aware of risks and implications, necessary refactors, etc. Both paths have quite serious consequences, and I am not sure whether the fork should be abandoned. 
- Future plans regarding extensions of this thing. I mean stuff like language control flow or scoping.

mondalaci, 2021-04-19:
We can create a macro command action quite easily in Agent, indeed, and it seems like the Right Thing to do. It'd be stored just like a macro text action but with a different enum value and interpreted by the firmware specially. (The `$` notation, however, would be used in macro event names regardless.)

Can you make your fork's macro calling compatible with macro names containing spaces? I'm not sure if it'd be worth it, so it's your call.

I don't think we should use multiline commands for the sake of simplicity.

If the stock firmware uses macro command actions and your fork would use text macro actions, then the two could coexist. I'm not suggesting it, but it's a possibility. Or would you prefer to use macro command actions in your fork, too?

A part of me still wants to merge your fork, but I'm a bit afraid that its feature set is not fully justified and adds unnecessary complexity from a user and developer standpoint. I guess I'd be less stressed if we could merge the commands one by one and think hard about their use cases and whether they're truly useful.

kareltucek, 2021-04-19:
> We can create a macro command action quite easily in Agent, indeed, and it seems like the Right Thing to do. It'd be stored just like a macro text action but with a different enum value and [...]

+1

> Can you make your fork's macro calling compatible with macro names containing spaces? I'm not sure if it'd be worth it, so it's your call.

Yeah I could, but I guess it is not a big deal either way, and not worth at the moment.
 
> I don't think we should use multiline commands for the sake of simplicity.

Yeah, simplicity-wise it makes sense, but think of the pain of copy pasting macros forth and back when you have to copy/insert every single action separately...

> If the stock firmware uses macro command actions and your fork would use text macro actions, then the two could coexist. I'm not suggesting it, but it's a possibility. Or would you prefer to use macro command actions in your fork, 

Well, it sounds as the clean way. On the other hand, your implication of leaving the firmware on the "stolen" text action sound quite reasonable.

Still the compatibility part of the thing stretches to the macro engine itself. Unless the efforts are either coordinated, or done from your side specifically to be non-conflicting, maintaining the fork alongside a new macro language might be quite laborious.

> A part of me still wants to merge your fork, but I'm a bit afraid that its feature set is not fully justified and adds unnecessary complexity from a user and developer standpoint. I guess I'd be [...]

Sounds fair. Still, I should elaborate a bit more on implications of this.

The thing is that in order for the engine to be useful, it needs the ability to run multiple macros in parallel (with other keys and with each other). Unfortunately, this is exactly the part of the engine which needs the refactor/rewrite. I guess, a good further goal might be to rebuild this thing anew and then start porting a reasonable feature set on top of it.

mondalaci, 2021-04-19:
If we introduce the macro command action in Agent, would you be interested in 1) extending the configuration parser of the firmware to handle it in a way to not conflict with your fork, and 2) possibly making it easy to port commands from your fork?

I agree that we should coordinate to not diverge the macro language of the stock firmware and your fork. You could even use the macro command action as long as we don't diverge. I only introduced the set command above which is not used in your fork, so there are no conflicts yet.

mondalaci, 2021-05-02:
@kareltucek I'd like to introduce smart macros in the near future, and I'm interested in your answer. I don't want to bug you, I'd just like to make sure it's not forgotten.

kareltucek, 2021-05-02:
@mondalaci honestly I am now very confused, and it seems like taking a week to think about it didn't help. So, let's break down my thought process. Following text is therefore mostly just a mind map, which does not always need to be considered relevant to this issue.

If you have better idea about what this issue is actually about implementation-wise, please do elaborate.

----

I have a feeling that 1) basically asks if I want to implement it myself, is that right? 

I am wondering what implementing it exactly means. I assume this means:

- Implementing something to trigger macro events. Any specific idea what that should be? 

- Implementing the actual macro action handler that will parse the strings and set the target variables. I am unsure about how this should ideally be done. My current approach (i.e., switching according to first letter and then "if-elsing" over commands) is quite naive, and using some indexed approach might be a better idea. Still, I am unsure how to do that in order to make it conflict free. Maybe redesigning the language to be more compatible with my current parsing schemes (see last section).

------

What does 2) exactly mean?

I guess it means that you actually want to achieve a strong and well written macro engine based on the fork. I guess it means one of the following - which one?

  a)  you actually want to start porting the fork "as is", but do it part by part. Yet preserve compatibility of the code between stock and fork.
  
  b)  you actually want to start building a new powerful macro engine loosely based on the fork, but avoiding my past bad decisions. Yet preserve compatibility of the code between stock and fork.
 
In either of the cases you will need to start with porting/writing a one-grade-more-powerful macro engine core. I am not sure how to do either in a painfree manner - the only thing that comes to mind atm is actually merging the fork "as is" while decluttering the feature set by disabling/deleting part of the feature set.

-----

On advanced macro engine:

It might be possible to implement smart macros within the current stock firmware now and merge it into fork, and just then care about how to merge fork into the stock firmware. But it might also be more efficient labour-wise to first take care of the macro engine core, and just then start adding new stuff.

If you wish to steer towards the powerful macro engine version (and it is your full right not to!), following roadmap sounds reasonable to me:

- First lets wait till the situation around stock firmware settles down a bit. And ideally around new products too. I probably need you to be able to dedicate some of your time to this, and to be focused when dealing with this.

- In the meanwhile, give me some time to actually merge upstream to the fork. Now the stock & fork are already in a pretty diverged state due chaotic and fast firmware development around modules.

- Then formulate what it is actually that you want to achieve in the long term (maybe having a few calls might be practical). You have been very conservative regarding any changes to the stock firmware for a very long time, so at this time your motivations are still an enigma to me.

- Then lets together go through the language of my firmware, and decide what should be ported directly, what should be ported later and what should be abandoned. Also, lets discuss other related issue at that point. 

- Then the macro engine core would be dealt with. I.e.:
  - The fork would be merged as is, yet immediately slimmed down and refactored.
  - I would probably drop support of my fork at that point. 

- Then smart macros would be either implemented or refactored.

- Finally, rest of the featureset would be ported if desired.

Unfortunately, this would mean that smart macros are not going to happen in near future. Fall of this year at best.

-----

Well. Back to the original question and this issue. I don't know. I think this thing is not a priority issue at the moment. I think you should focus on delivering the hardware, and deal with questions around my fork and this issue when you have time to do it properly.

Or decide to patch it quickly (i.e., implement smart macros on top of current stock firmware) regardless of future compatibility and call it a technical debt.  If the different initiation strategy (e.g., `$` in regular text action for my macros and new action for smart macros) is used, and if the implementation is sufficiently self-contained (maybe implemented in a separate file), then it might be relatively conflict free for the time being. In that case, I probably prefer you to deal with it, since I don't have a clear idea of what should be done. *[This seems to be the currently preferred variant.]*

Or go head first by merging the fork now (that is when I manage to merge the upstream) while slimming down its feature set and continue from there onward chaotically.

-----

One more thought on the design of your smart macro language. It might be a bit more practical to think about parsing strategy before designing the language itself. Namely, I am a bit concerned about the "domain driven" nature of the language, which is nice but seems to ignore parsing strategy. Implementation wise, it might be a bit more practical to make it an imperative LL0 parsable language.

[... cut ...] d then continue recognizing the rest of the string (e.g., which config value is to be set etc.). This way would make the question of compatibility a non issue.

The problem with the current proposal is that it uses an infix notation, either forcing me (and the parser too) to read it from the center (which implementation-wise means dealing with some LR parser, which would need nontrivial infrastructure and memory), or to consider every "path" to be a separate command (which might be impractical).

kareltucek, 2021-05-02:
The bottom line is probably that if you want to start by implementing smart macros and have them implemented soon, then probably I prefer you to do it.

mondalaci, 2021-05-16:
@kareltucek This is not a trivial issue, and I've been thinking about the details for a while. Let me be very specific going forward.

I think it'd be practical for both firmwares to use the yet to be implemented macro command instead of the $ prefix for the sake of consistency and clarity.

As for the syntax, I only used the a=b infix notation as an EBNF-like notation, but the actual syntax would be `set module.trackball.navigationMode.base caret` which would make parsing easy.

For the time being, I'd like to have only the `set` and `printStatus` commands in the stock firmware. I'm reluctant to add the other commands and language elements, but will reconsider later. This way, your firmware would implement the superset of the stock firmware smart macros.

I'd like to have multi-line macro commands where each line would describe a single command. This way, it'd be possible to set multiple variables in a single macro action without creating additional actions per variable.

I'd expect the macro engine to interpret macro commands and call macros when events are triggered. Please feel free to suggest whether to use your codebase for such purposes.

With the introduction of the `set` command, I think it'd be worth to migrate your `set*` commands as parameters of the `set` command and obsolete them.

mondalaci, 2021-05-31:
@kareltucek What do you think would be the best way to go forward with this?

kareltucek, 2021-05-31:
> For the time being, I'd like to have only the set and printStatus commands in the stock firmware. I'm reluctant to add the other commands and language elements, but will reconsider later. 

I get it, yet your further plans and intentions are quite vital for any considerations regarding the macro engine core. 

> I'd expect the macro engine to interpret macro commands and call macros when events are triggered. Please feel free to suggest whether to use your codebase for such purposes.

That's a difficult question. Long story short, the engine can provide you:

- The ability to run macros in parallel. This feature can simplify the initiation sequence for the event triggers, since with that feature, it suffices to fire all the event-triggered macros at the same time.

- Some string handling utilities. Could be partially substituted with standard libs, but when writing it I was not sure how wise it is to link such things as part of the firmware.

- The status buffer.

But apart from that, I think it does not matter much whether to use my codebase or whether to stick with current code for the moment.

Of course, for me unifying the two macro engines is somewhat desirable (especially since the access to stateful information differs between them, which makes maintenance somewhat nontrivial). On the other hand - again all those things discussed in #325...

> @kareltucek What do you think would be the best way to go forward with this?

I think the best way would be to have a few calls and discuss things in a bit wider context and in a bit more realtime fashion. Will get in touch by email...

mondalaci, 2021-06-09:
@kareltucek A quick summary of yesterday's meeting:

- You want to tidy up the layer handling code. I don't feel strongly about this, but I have no objections, so feel free to go ahead.
- I want to have multi-line macro commands.
- Let's use [macro actions](https://github.com/UltimateHackingKeyboard/agent/issues/1534) instead of the `$` notation except for macro events.
- Let's only expose macro events and the `set` and `printStatus` commands first. Feel free to use a preprocessor macro to limit the commands. This way, you can easily expose all the available commands with a flip of a switch in your fork.

mondalaci, 2021-07-04:
@kareltucek Just so you know, [this Agent PR](https://github.com/UltimateHackingKeyboard/agent/pull/1545) exposes macro commands via Agent. When adding a macro command and saving the configuration, currently the latest 8.10.10 firmware throws "Communications error with UHK. Response code: 7" because it doesn't support the newly added macro command type which has yet to be implemented in the firmware.

mondalaci, 2021-11-26:
Resolved by #397.

closed 2021-11-26 by mondalaci

andriysavin, 2021-11-27:
Hi, when to expect some documentation on this?

kareltucek, 2021-11-27:
> Hi, when to expect some documentation on this?

No idea about Laci's plans, but there is the [MACRO_DOCS.md](https://github.com/UltimateHackingKeyboard/firmware/blob/master/MACRO_DOCS.md) file in repository root which describes the full version of the language. Bear in mind that only the printStatus and set commands are enabled by default.

kareltucek, 2021-11-27:
Also, at the moment it is somewhat outdated - the $ notation no longer exists, and commands have to be entered by means of the new macro command action in agent.

mondalaci, 2021-11-27:
@andriysavin Yes, we plan to make easy-to-understand documentation and add IntelliSense support to Agent macro command editor.

stefandeml, 2022-02-28:
Will this issue be re-opened when the work on smart macros will be started?

kareltucek, 2022-02-28:
> Will this issue be re-opened when the work on smart macros will be started?

The work on smart macros is done. The corresponding code is already merged in master and just waits for release.

You can also check out https://github.com/kareltucek/firmware, which contains almost strict superset of the functionality targeted by stock smart macros.

stefandeml, 2022-03-01:
awesome!
I'm interested in changing the mouse speed of the track ball. Any pointer for me to look into for achieving this using smart macros?

mondalaci, 2022-03-01:
The next Agent and firmware release will expose this feature via a quite user-friendly manner. Please subscribe to new releases, and give it a try when released.

kareltucek, 2022-03-01:
> awesome! I'm interested in changing the mouse speed of the track ball. Any pointer for me to look into for achieving this using smart macros?

With the above firmware, create a macro named `$onInit`, the create a text containing (e.g.) `$set module.trackball.speed 2.0`.