UltimateHackingKeyboard/firmware #353: Feature request:  Add modifier key delays to keys bound with modifiers

firngrod, 2021-04-15:
Mappings of keys with modifiers such as Alt+Tab or Ctrl+Shift+Esc using the direct, naïve approach of a key plus mod keys directly in the UI is currently next to useless when using Windows Remote Desktop Protocol.  This is an issue in this time of pandemic-enforced homebound work.  I have observed similar issues when working through TeamViewer, so it seems to not be isolated to the RDP.

From what I can gather, the issue is that the UHK sends the keys "too quickly" and that the RDP program or protocol, or some layer in-between, gets confused about which one was pressed first, or only registers one or some of the keys.  A solution, or at least work-around, would be to delay each key event slightly to allow the RDP or similar program to parse the key events correctly.  From testing with a custom firmware, 5-10ms delay seems to do the trick.

Your stance stance on this, from an earlier report of the same issue, was along the lines of "The UHK adheres to the specs of the keyboard USB protocol and the issue is in the RDP implementation.  We will not do firmware work-arounds for software issues in other products."

While I do agree with the analysis and can see the merrit in the stance, the fact of the matter is that Microsoft is not going to address this for the insignificant minority of their customers who has this issue, and who frankly have no choice but to use their OS for work anyway.  So no fix is forthcoming for the software bug in other products.

As such, this issue does detract greatly from the value of your product for anyone using RDP for their daily work, to a point where I would advise people to avoid your product if they use RDP extensively.  Had I known of this issue, I certainly would have thought twice before purchasing your product, which is a shame because I absolutely love every other aspect of it.

I would implore you to reconsider your previous stance on this issue and address it, such that the full range of features of your keyboard can become available for those of us who have to use RDP for work.

kareltucek, 2021-04-15:
This is a known problem and has been discussed on multiple occasions. Please see this post: https://github.com/UltimateHackingKeyboard/agent/issues/871#issuecomment-716100383

The bottom line is that currently the recommended solution is to use my fork (https://github.com/kareltucek/firmware, for instructions see post above). The allegedly reliable delay is 5ms.

Still, thanks for bringing this forward again! @mondalaci just for the record, my opinion is that the solution should be ported and made available in master. The solution is clean and takes just something like 5 lines of code while using the power of the postponer.

firngrod, 2021-04-15:
@kareltucek
I'm currently running your fork of the firmware for this very reason, and I'm happy with that.

My worry about using your firmware is that at any time, you might decide to stop maintaining your branch, and that would leave the users of your firmware without updates from the main team, such as bug fixes, support for new modules, and the like.  Or that you might not have time to, or decide not to, merge the module support which would leave me effectively unable to use your firmware once the modules for my keyboards arrive.

Also, it's not a strong selling point that the keyboard can work fully as expected with RDP as long as the customer is willing to flash non-official firmware on it.

The discussion from that previous thread ( UltimateHackingKeyboard/agent#871 ) is also what lead me to post this issue as a feature request rather than a bug report, as it is true that the bug is not in the firmware, but rather in the surrounding software, and it would be a feature beyond base specs to have the firmware work around that.

Finally, I would mention that I have seen issues with key combinations not making it through multiple layers of remote sessions (RDP -> TeamViewer) with 5ms keystroke delay.  I've yet to reach any kind of conclusion on whether 10ms is better, but frankly, expecting the firmware to account for such cases is unreasonable.  I would expect no improvement beyond 10ms.

mondalaci, 2021-04-15:
Now that [smart macros](https://github.com/UltimateHackingKeyboard/firmware/issues/351) are planned, I'll gladly merge this to master, and we'll expose this feature as a smart macro variable.

mondalaci, 2021-04-20:
@kareltucek We should make a build of your PR so that @firngrod could test it. Please suggest a `keystrokeDelay` value.

firngrod, 2021-04-20:
I would be happy to test any firmware build you send my way.

I'm running @kareltucek's firmware, and I use 10ms as delay currently.  5ms seems pretty solid for RDP, but I have seen some issues with TeamViewer, which is why I switched to 10ms.  Since then I haven't had any support, so I've had no chance to test it yet.

kareltucek, 2021-04-20:
If this is about testing whether it functions as expected, you (@mondalaci ) can try it out with let's say a 100ms delay :-).

mondalaci, 2021-04-20:
@firngrod Please give a try to 
[uhk-firmware-karel-port-keystroke-delay-10ms-8.10.8.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6346589/uhk-firmware-karel-port-keystroke-delay-10ms-8.10.8.tar.gz)

firngrod, 2021-04-21:
So far not looking good.  I can Alt+Tab, but I can't Ctrl+Shift+Esc or Ctrl+Alt+End.  My immediate impression is that it works for single-modifier keystrokes only?  Except that Super+Shift+Arrow seems to work.

kareltucek, 2021-04-21:
Oops, I see I have messed up the cherrypick. It's working now (yes, this time I actually tested it rather then trusting my omnipotence).

Here is a fixed tarball, if you would be so kind to give it a shot - [uhk-firmware-port_keystroke_delay_v2_10ms-8.10.8.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6349000/uhk-firmware-port_keystroke_delay_v2_10ms-8.10.8.tar.gz)

> Edit:
> I have to unplug and replug the keyboard after flashing the test firmware before I can use the mouse layer again (And probably other functionality). I don't remember having to do that before, and I don't after flashing back @kareltucek's firmware.
> RCtrl seemed to get stuck in it's pressed state after one of the firmware flashes even though I never use that key. Replugging the keyboard however many times didn't help. Hitting RCtrl resolved the issue.
> Replugging the keyboard however many times didn't help. 

Interesting. Sounds like uninitialized memory to me, esp. due to the "Replugging" part.

Have you observed this behaviour at any other time? (I.e., not as an immediate consequence of flashing.)

firngrod, 2021-04-21:
> Oops, I see I have messed up the cherrypick. It's working now (yes, this time I actually tested it rather then trusting my omnipotence).

Happens to the best of us.

Now the key modifiers seem to work well.  It even seems to work through a TeamViewer session on the other end of the RDP (Set up a phony support setup to test it.  Can't vouch for bad connections, but we can't take care of everything, eh?)

For this build I also had to power cycle the keyboard after the firmware flash before I could get any output.  The display shows the proper keymap.  I do not recall having to do that with any other firmware flashes, nor at any other time for any reason that I haven't identified.

I do have an issue where the keyboard will go haywire from external disturbances.  When I turn on my old CRT oscilloscope, the keyboard will almost always output "-i]kkkkkkkkkkkkkkkkkk...." and keep spamming 'k' until I power cycle it.  The oscilloscope is on the same power bar as the computer.  The same happens (rarely) when the floor heating relays switch, the power for which is drawn from just upstream from said power bar.  Both are in close-ish physical proximity to the keyboard (~1m).  A ferrite filter does not seem to eliminate this issue, although it does seem to alleviate it a little.

mondalaci, 2021-04-21:
EMI (electromagnetic interference) can cause issues like the ones you mentioned, although they happen quite rarely according to the feedback we receive. We'll try to improve the UHK in this respect but such issues are usually very hard or impossible to reproduce. In the meantime, workarounds exist, such as connecting your UHK (or rather its host computer) as far from such disturbing appliances as possible.

firngrod, 2021-04-22:
I had also put it down to issues in my setup once I made the connection to the relays switching, which is why I haven't reported it as a bug.  Once I can gather enough energy, I will move where I draw power for my home office setup to the other end of the room and see if that helps.  Otherwise, I'm almost done with my USB-C modification to the keyboard, after which power cycling it using a magnetic cable connection becomes very easy.

mondalaci, 2021-05-02:
Reopening this issue, because even though #358 has been merged, the `keystrokeDelay` [smart macro](https://github.com/UltimateHackingKeyboard/firmware/issues/351) variable has yet to be exposed.

firngrod, 2021-07-14:
I am sorry to be a bother, but what is the current status on this feature?

I just received my modules, so I'm eager to know whether I can expect to use the latest version of the firmware, whether I can just stick with the cherry picked test version (8.10.8), or whether I should run the latest version from @kareltucek.

mondalaci, 2021-07-14:
We're making progress with smart macros according to its issue to expose this feature.

I'm sorry, but I'm a bit lost as to which revision of which repo is best to build for your needs. Karel may be able to help you.

firngrod, 2021-07-14:
That's perfectly understandable.

Things seem to function with this firmware, and 8.10.9 on the modules, with which they shipped.  Would there be any any predictable issues with mixing firmware versions?

mondalaci, 2021-07-14:
Not for the time being. Later on, it may be required that the keyboard halves and modules run the same firmware version.

kareltucek, 2021-07-14:
> I am sorry to be a bother, but what is the current status on this feature?

It is implemented in both stock firmware and in the fork. However, in stock firmware there is currently no way to configure it. 

There have been some bug fixes since the thing was merged into master, and probably some new features too, so upgrading might be a good idea.

My fork has now upstream merged, so it should (hopefully) work just as fine as stock. If you prefer running stock, here is a quick build [uhk-firmware-8.10.10.keystroke_delay_10.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6818018/uhk-firmware-8.10.10.keystroke_delay_10.tar.gz)

firngrod, 2021-07-22:
@kareltucek

That works perfectly.  So did the old keystroke delay build while leaving the modules on the shipped firmware 8.10.9, but now everything has the same firmware, which feels more correct.  Thank you for taking the time to build this!

One note, however:  I had to disconnect the modules to update the left keyboard half, otherwise, it would get stuck 'waiting for the "Left keyboard half"' and then time out.  Disconnecting the modules, updating the two halves, then reconnecting the modules and updating them allowed the full update to go through.

Again, thank you for taking the time to help out!

mondalaci, 2022-04-16:
`keystrokeDelay` is merged to master, and the next firmware release will contain it, so I'm closing this issue.

closed 2022-04-16 by mondalaci

giulianob, 2022-04-16:
> `keystrokeDelay` is merged to master, and the next firmware release will contain it, so I'm closing this issue.

How do we configure this in the next release? Will it be in the agent as well?

mondalaci, 2022-04-16:
I'll write a dedicated blost post about this feature set.

PR #358 (2021-05-02): Port keystroke delay.
Closes #353.