UltimateHackingKeyboard/firmware #361: Some Keybindings not working with 8.10.9

ldriscoll, 2021-04-30:
I have Mod + F bound to L_Ctrl + L_Super + Right Arrow
and Mod + S bound to L_Ctrl + L_Super + Left Arrow.

When I updated to firmware 8.10.9 these bindings stopped working.
Also, Fn + 1, 2, 3, 4, 5, and 6 failed to cycle through the layout mappings. 

I downgraded to 8.10.8, and functionality has been restored.

I'm running the latest agent on Linux (OpenSUSE Tumbleweed) Kernel 5.12.0-1

mondalaci, 2021-04-30:
Please attach your configuration.

ldriscoll, 2021-04-30:
Sorry, had to zip it, GH wouldn't let me upload the JSON: [UserConfiguration.zip](https://github.com/UltimateHackingKeyboard/firmware/files/6408022/UserConfiguration.zip)

mondalaci, 2021-04-30:
I couldn't reproduce any of the mentioned issues with firmware 8.10.9 and your configuration on Linux Mint.

Please test your UHK with another computer and let me know your findings.

ldriscoll, 2021-04-30:
Exactly the same problem if I update using the MBP. I can't switch between keymaps, and the keybindings don't work.
I've tried installing the firmware bundled with the agent, as well as downloading fresh. The same issue.

mondalaci, 2021-04-30:
What happens after saving your configuration when running firmware 8.10.9?

ldriscoll, 2021-04-30:
@mondalaci I have found a workaround. The issue happens if I
1. Have my configuration
2. Update to 8.10.9

The workaround:
1. Export the user configuration
2. Update to 8.10.9
3. Re-import the user configuration.

This seems to work now.

mondalaci, 2021-04-30:
So glad the issue has been resolved for you!

I'd like to fix the root cause of this issue, though. Can you reproduce it by importing your configuration when running firmware 8.10.8 and then updating to firmware 8.10.9?

ldriscoll, 2021-04-30:
I'll give it a go.

ldriscoll, 2021-04-30:
Yup it breaks:
1. Downgrade to 8.10.8
2. Install my configuration
3. Test that it's working correction - it is
4. Upgrade to 8.10.9
5. Test keybindings/Test switching between keymaps - *Both fail*
6. Re-import my configuration
7. Everything works again.

mondalaci, 2021-04-30:
Thank you! We'll look into it.

mhantsch, 2021-06-07:
Something very similar happened to me too after upgrading firmware from 8.10.9 to 8.10.10. Directly after the firmware upgrade finished, the keyboard seemed to work with some default map, as the Capslock key did not activate the mod layer as it should have on my keymap. I then clicked on a key in Agent and "Remap key" (without changing its assignment) to get the "Save to keyboard" button. After "Save to keyboard", everything worked again.

So my fix did not need an export/import of the configuration in Agent, but I needed to re-save it to the keyboard.

rbrinkman, 2021-07-16:
This just happened to me as well. Replication steps:

1. Used Agent 1.5.14 app image on Linux to flash firmware 8.10.9 (upgrading from 8.10.1 firmware, previous Agent version was probably 1.5.8 since I last updated in December)
2. Observed that the base layer on the left keyboard half was reset to the stock Qwerty PC map, while the right half retained my custom map. Agent's UI showed my map but the keyboard was not using it. I had removed the keymap switching from the function layer, so I cannot confirm whether other keyboard layouts were stuck too.
3. No attempts to save key mappings to the keyboard from Agent 1.5.14 were successful. The message on Agent indicated success but no keymapping saved.
4. Opened an old Agent version (1.2.5) on my Mac (different computer), which also displayed my keymappings despite the keyboard not having them. However, remapping the keys and saving the settings worked on this version. I did not reflash the firmware.
5. Keyboard is currently running 8.10.9 firmware with a keymap saved by Agent 1.2.5.

Also of note:
Agent 1.5.14's Configuration History does not show an entry for these latest changes. Its last entry is from my previous update in Dec 2020.

mondalaci, 2021-08-16:
@ldriscoll I was able to reproduce this issue the other day, but not this time. When reproducing the issue, please let me know which Agent version you're running (preferably the latest one), your operating system, and which modules are attached to your UHK.

@rbrinkman Your help is also much welcome. I have the same questions for you as for @ldriscoll. Please try to reproduce the issue with the latest Agent version and include your configuration.

ldriscoll, 2021-08-16:
I've done the update using the latest agent/firmware, and everything seems to have gone well. Thanks

mondalaci, 2021-08-16:
@ldriscoll I'm happy for you, but can you possibly reproduce the issue with an older Agent version? I'm slightly worried, because I don't know what happened, and it may resurface again in the future.

ldriscoll, 2021-08-16:
Using Agent 1.5.12:
1. Attempt to reinstall 8.10.10 - keyboard works as expected
2. Downgrade to 8.10.9 - Keyboard works as expected
3. Upgrade to 8.10.10 - keyboard works as expected
4. Downgrade to 8.10.8 - keyboard works as expected
5. Upgrade to 8.10.9 - keyboard works as expected (this is the situation that failed before)
6. Upgrade to 8.10.10 - keyboard works as expected

Using agent 1.5.15
1. Downgrade to 8.10.8 - keyboard works as expected
2. **REMOVE TRACKBALL - Upgrade to 8.10.9 - reproduced issue.**
3. Reinstall Trackball - Upgrade to 8.10.10 - Keyboard now works again, no need for config to be uploaded.

mondalaci, 2021-08-17:
@ldriscoll I must be missing something because I couldn't reproduce the issue based on your description. Before you started to reproduce the issue using Agent 1.5.15, which modules were you using? Only the trackball module? Also were you using your configuration that you shared above?

ldriscoll, 2021-08-17:
Yes, I only have the trackball; and I've not changed my config. It seems to only be an issue when there are no modules attached to the keyboard.

mondalaci, 2021-08-17:
@ldriscoll Thanks, your feedback has been very useful!

According to my findings, this issue only occurs after flashing firmware 8.10.9 or 8.10.10 (which is currently the latest version) and when none of the key cluster, trackball, and trackpoint modules are attached. (The presence of the touchpad doesn't make a difference.) Power cycling the UHK makes it load the correct keymap, so it's only an issue directly after flashing the firmware.

Can you confirm my findings?

ldriscoll, 2021-08-17:
@mondalaci 
Confirmed:
1. Remove trackball module
2. Using agent 1.5.15 Downgrade to 8.10.8
3. Verify keybindings work
4. Upgrade to 8.10.10
5. Keybindings don't work
6. Unplug keyboard from USB/Reconnect Keyboard to USB
7. Keybindings work again

rbrinkman, 2021-08-17:
@mondalaci Yes I believe your findings are correct. When I conducted the botched upgrade above, I did not have my key module yet (it was shipping). Confirmation testing below:

1. Used Agent 1.5.14 to flash firmware 8.10.9 with no modules attached - left half of keyboard lost its keymap
2. Re-saving keymap separately did work this time
3. Used Agent 1.5.15 to flash firmware 8.10.10 with no modules attached - left half of keyboard lost its keymap
4. Unplugged/replugged keyboard, which restored the map (confirms that this is only a problem directly after flashing)
5. Used Agent 1.5.15 to flash firmware 8.10.10 *with key module attached* - both halves of the keyboard retained their maps, but *the key module lost its mapping*
6. Unplugged/replugged the key module but this did not fix the mapping
7. Re-saved the mapping, which correctly saved the key mapping to the key module

mondalaci, 2021-08-20:
I've bisected the commits between firmware versions 8.10.8 and 8.10.9 and found that 45c027de8053c3f50a8932e4ab600b832bce3451 introduced the issue. Interestingly, at that point, the issue surfaced regardless of whether the left or right firmware was flashed.

The next 75766c4e5a139af2f5fe93d7b79f75082f774e36 commit made the issue only surface when the left half (or non-touchpad modules) got flashed which explains why it wasn't noticed during development because usually only the right half gets flashed when developing the firmware.

I'll dig deeper and will try to come up with a fix.

mondalaci, 2021-08-21:
Further investigation revealed that whichever module (left keyboard half, key cluster, trackball, or trackpoint module) gets flashed lastly during the firmware update process is the one affected by this issue. So if the modules are not connected and the left keyboard half gets updated lastly, it'll use the default keymap. If a trackball module is connected and gets updated lastly during the firmware update process, it'll use the default keymap.

The cause of this issue is probably the bootup delay of the affected module. The right half boots up first, it doesn't detect the lastly updated module, which hasn't booted up yet, so it doesn't actualize its mappings. Amusingly, this issue is probably the duplicate of #348.

PR #497 (2022-07-03): Reload keymap when module is swapped.
Closes #361, #348.
closed 2022-07-03 by mondalaci