UltimateHackingKeyboard/firmware #364: Touchpad module oscillation

mondalaci, 2021-05-02:
The touchpad module oscillates in Agent.

[image]

I noticed this issue a while ago. The right keyboard half keeps querying the touchpad over I2C, and the touchpad NAKed from time to time which caused the above phenomenon in Agent. I set MANUAL_CONTROL to 1 and MODE_SELECT to Active mode in bb9bb29 which resolved the issue.

Afterward, I had a chat with Azoteq to improve the tracking accuracy of the touchpad and changed some values in a2af6d994015cc3a2dd4f9b91ca1e004ea7df95a which apparently reintroduced the NAKs. The following values were changed in the Trackpad Hardware Setup values of the Prox Hardware Settings section:

1. Pass Length: 101 -> 010
2. UpLength: 101 -> 010
3. VTrip: 0011 -> 1000
4. Opamp Bias: 2.5 uA -> 5.0 uA

According to my findings, changing either 1. or 2. negates the effect of the `MANUAL_CONTROL=1` and `MODE_SELECT=Active mode` settings.

We have to figure out which settings to use to maximize tracking accuracy and get rid of the NAKs.

TristanoTenaglia, 2021-05-31:
Having the same issue. An added note is that that the message under the firmware tab "Touchpad is detected..." is also oscillating. Thanks!

misterwrong, 2021-06-12:
I have the same issue, both on Mac OS Catalina and Arch Linux.

mhantsch, 2021-06-12:
I had that too, but I thought it may be an electrical contact problem between module and keyboard, so I rubbed the contacts a little and it seems the problem went away.

misterwrong, 2021-06-12:
> I rubbed the contacts a little and it seems the problem went away.
If dirty contacts were the problem, I'd expect the device to function intermittently as well, but that's not the case. I'd also expect the pulsating in the UI to be less regular.

mhantsch, 2021-06-12:
@misterwrong Well, I'm still observant to see under which conditions the issue may reoccur.

ehdoyle, 2021-06-15:
I'm seeing the same symptoms - Ubuntu 20.04. and the 1.5.14 Linux version of agent, with updated firmware from that agent release.
Although I have to commend support for being really quick in responding to my email and directing me here. I figured it would be a day, rather than an hour before I heard anything back.

YokaiRick, 2021-06-15:
Got my touchpad today too. It works fine, but within the agent i have the same problem on archlinux too
I use the AppImage (tried it on two different PCs)

brokenflea, 2021-06-16:
same issue with the touchbad i received yesterday, MacOS Big Sur (11.4) running on 1.5.14 Agent.

dante-12, 2021-07-17:
same issue with both the trackball & touchpad module. The modules are almost non-functional. Agent 1.5.14, Firmware 8.10.9. Windows 10.

mondalaci, 2021-07-17:
@dante-12 What do you mean exactly by saying that "the modules are almost non-functional"? The touchpad is functional, but its tap action can't be changed from left click until this gets fixed. We've received much positive feedback regarding the trackball.

dante-12, 2021-07-17:
@mondalaci It hints me to try again since I have two UHKs (Windows & Mac). Trackball works perfectly on Mac but Windows. On Windows, I was unable to make the mouse cursor move mostly (can move very rarely). Then, I did clean a contact metal parts on UHK for Windows, then it looks working fine last a few minutes. The oscillation issue is also gone. Thanks for your response.

brokenflea, 2021-07-18:
just checking in, is there a fix for the oscillation on touchpad or an eta to when it'll become available ?

mondalaci, 2021-08-07:
There's no ETA, but I'll prioritize such issues as things settle up, which may take some months.

mattgodbolt, 2021-08-07:
> There's no ETA, but I'll prioritize such issues as things settle up, which may take some months.

Thanks for the update. The touchpad is an expensive curiousity until we can configure it, and right now we can't configure it as it keeps appearing and disappearing in the UI... so a timely fix would be lovely :) I appreciate there's lots to do though!

Slabity, 2021-08-11:
I'm glad I'm not the only one having this issue.

If the underlying firmware issue will take a while to fix, would it be possible to allow the user to configure their modules without them being connected to the keyboard? That would allow us to configure the touchpad module (or any other module) without dealing with this particular issue.

mondalaci, 2021-08-12:
@Slabity You can export your configuration, edit the relevant mapping, and import your configuration as a workaround.

YokaiRick, 2021-08-16:
Oh, good to know. I'll try it. Thanks ^^

mondalaci, 2021-08-28:
@kareltucek The workaround I suggested is useless until #349 gets fixed, indeed.

Although it's unnecessary to have a module attached to edit its configuration, the current user interface is cleaner than any alternative I can think of, which allows the edition of any module regardless of whether it's attached. We've shipped thousands of modules, and nobody has complained about this limitation so far. Even if users knew that the possibility is technically there, I don't think it'd make a difference.

kareltucek, 2021-08-28:
Hmm, didn't think about user interface. If I was to put it somewhere, I would probably go with a cogwheel centered right below layer bar. On click, it would open a popover that would have clickable options, one per line (reading `auto` `keycluster` `trackball` `trackpoint` `touchpad`) that could be toggled by a click. Clicking `auto` would disable everything else and similarly clicking any module would disable `auto`. Clicking any right module would disable any other right module. 

But if no-one has complained, it's indeed not worth the trouble.

SaswatPadhi, 2022-02-08:
I have 8.10.12 firmware and looks like this problem still exists

brokenflea, 2022-02-08:
issue still exists, my module is a dead paperweight since I bought it.

mondalaci, 2022-02-08:
@brokenflea The oscillation in Agent is annoying, but your module should be useable regardless. What makes your touchpad module useless?

brokenflea, 2022-02-08:
> @brokenflea The oscillation in Agent is annoying, but your module should be useable regardless. What makes your touchpad module useless?

Since the module oscillates there’s no way I’ve been able to get the agent to do something as simple as tap the touchpad for left click or right click or double click. The only thing I’m able to do with the module is move the cursor around which defeats the purpose of having the module in the first place.

famuvie, 2022-02-08:
For what is worth, the **connection of a second left-side module** stops the oscillation issue and allows editing the functionality of the module.
If you have one, it can be used as a workaround in the meanwhile.

[image]

kareltucek, 2022-02-08:
Does it? As far as I know, left click is still hard-coded in latest stock firmware release, so you cannot remap it anyways and there is nothing to be gained by the workaround at the moment.

(Also, feel free to try https://github.com/kareltucek/firmware - it contains new touchpad code. Left click can be remapped via Agent, UserConfig.json or via a macro command, right click can be remapped only via a macro command. Navigation modes can be reassigned or remapped by macro commands.)

famuvie, 2022-02-08:
You are right. I had not actually try to remap anything. I was referring to the oscillation issue.

bhaeussermann, 2022-02-08:
@famuvie By default - without any remapping / configuration - and the way I've been using my touchpad is:

- swiping moves the cursor,
- tap emits a left click and
- MOD + swiping scrolls.

I often use it in combination with the Mouse layer for eg. right-click or click and drag actions.

mondalaci, 2022-09-03:
@kareltucek Can you still reproduce this issue? With the git master firmware and Agent, I can't reproduce it anymore.

kareltucek, 2022-09-03:
Yes, but only when left module or left half is disconnected. 🤨

mondalaci, 2022-09-03:
@kareltucek Interesting finding!

I can only trigger the bug when disconnecting the left half. Then I can only untrigger the bug by reconnecting the left half and the key cluster module.

Can you dig deeper? I think we're onto something.

kareltucek, 2022-09-03:
Sure!

mondalaci, 2022-10-21:
Resolved by #534. The newly released 9.1.1 firmware version contains the fix.

closed 2022-10-21 by mondalaci