UltimateHackingKeyboard/firmware #368: CTRL+C starts to open Calculator on Windows with firmware 8.10.9

mschwartau, 2021-05-06:
I'm using Windows 10 Pro (Latest version, 20H2). After installing the newest firmware when pressing CTRL (left) + C the windows calculator is opened. This does not happen if I use the keyboard of my laptop or the right CTRL key. 

Furthermore after flashing the firmware with  8.10.8, everything was back to normal. So CTRL (left) + C just copies the selected text and does not open the windows calculator.

I tried it multiple times (upgrading to 8.10.9 using agent 1.5.13) and reverting to 8.10.8, it was reproducible.

My config is as follows:
[UserConfiguration.json.txt](https://github.com/UltimateHackingKeyboard/firmware/files/6433715/UserConfiguration.json.txt)

I exported the config using firmware 8.10.8 and 8.10.9 and compared them but there was no difference.

mondalaci, 2021-05-07:
Please re-save your configuration in Agent after updating to firmware 8.10.9 and let me know if it fixed your issue.

mondalaci, 2021-05-07:
Thanks for letting us know! This is a duplicate of #361, and we'll look into the root cause.