UltimateHackingKeyboard/firmware #374: USB 2.0 Compliance: HID GetProtocol and SetProtocol requests unsupported

abcminiuser, 2021-05-20:
HID compliance failure: 

```INFO Starting GET/SET protocol tests for interface : 0
INFO GET/SET protocol request is optional. Skipping tests
INFO Starting GET/SET protocol tests for interface : 1
FAIL (HID: 1.1.10) HID devices must support a valid SetProtocol(Report) request.
FAIL (HID: 1.1.11) HID devices must support a valid GetProtocol request.
FAIL (HID: 1.1.12) HID devices must support a valid SetProtocol(Boot) request.
FAIL (HID: 1.1.11) HID devices must support a valid GetProtocol request.
FAIL (HID: 1.1.10) HID devices must support a valid SetProtocol(Report) request.
FAIL (HID: 1.1.11) HID devices must support a valid GetProtocol request.
INFO Starting GET/SET protocol tests for interface : 2
INFO GET/SET protocol request is optional. Skipping tests
INFO Starting GET/SET protocol tests for interface : 3
INFO GET/SET protocol request is optional. Skipping tests
INFO Starting GET/SET protocol tests for interface : 4
FAIL (HID: 1.1.10) HID devices must support a valid SetProtocol(Report) request.
FAIL (HID: 1.1.11) HID devices must support a valid GetProtocol request.
FAIL (HID: 1.1.12) HID devices must support a valid SetProtocol(Boot) request.
FAIL (HID: 1.1.11) HID devices must support a valid GetProtocol request.
FAIL (HID: 1.1.10) HID devices must support a valid SetProtocol(Report) request.
FAIL (HID: 1.1.11) HID devices must support a valid GetProtocol request.```

The GetProtocol and SetProtocol requests must be supported on HID interfaces that advertise boot protocol functionality (i.e. `USB_HID_SUBCLASS_BOOT`, `USB_HID_PROTOCOL_KEYBOARD`/`USB_HID_PROTOCOL_MOUSE`). We are non-compliant in two respects: we indicate boot protocol support but don't sent boot protocol compatible reports, and we don't support the required protocol switching requests.

Quick fix for now is to remove the boot protocol values from the descriptors; this should prevent some BIOS and other applications from getting extremely confused when trying to use the device in boot protocol mode. Long term fix is to actually add boot protocol support to the keyboard and mice interfaces.

abcminiuser, 2021-05-20:
Note: This is currently worse than it would appear on the surface: a bug in the Kinetis code interprets SetProtocol requests as `SetIdle`, and `GetProtocol` requests don't fail the request if the user application fails the command. That results in the host setting and getting boot protocol values on the interface "successfully", without actually altering the layout of the reports sent from the device to the host.

mondalaci, 2021-05-20:
It's worth mentioning that the Kinetis SDK has been obsoleted by the MCUXpresso SDK, which implements the same API.

The migration, however, would be quite tricky as we've patched our KSDK fork quite a bit, and the same changes would have to be applied to the MCUXpresso SDK. Also, I think the chances are slim that the MCUXpresso SDK fixed such USB-related issues.

In any case, it'd be worth reporting SDK-related issues to NXP, for which we'd have to upgrade to the latest MCUXpresso SDK.

PR #378 (2021-05-22): Add support for additional mandatory HID class requests.
Fixes #374.

This implements the remaining HID class requests, several of which are mandatory for HID class interfaces that implement boot protocol functionality (the restricted subset of HID that is used by limited hosts, such as PC BIOS).

1) Add `SetProtocol`/`GetProtocol`:
Only mandatory for the boot protocol keyboard/mouse interfaces, as they advertise boot protocol support in the configuration descriptors. These should initially report 1 (report protocol) but should be settable to 0 (boot protocol) by the host on demand. 

Happily  [... cut]
closed 2021-05-22 by mondalaci