UltimateHackingKeyboard/firmware #380: USB XInput interface

mondalaci, 2021-06-02:
It'd be cool to transform the UHK to a gamepad for which an XInput interface should be implemented. In specific, we should make the trackpoint module configurable to behave like a gamepad joystick. I've exchanged emails with @steamraven and @abcminiuser in this respect, and I'd like to continue the discussion here.

benedekkupper, 2022-03-13:
So let's get closer to some actual specification of desired functionality here. Which controls of a gamepad are required? Which SW components are producing the data that is to be represented as gamepad actions? What transformation is needed on the data before it is sent over USB?

mondalaci, 2022-03-14:
@benedekkupper We want to expose every XInput control. Please provide a list of the controls, and we'll move forward from that point.

mondalaci, 2022-03-26:
@benedekkupper Let me be more specific. For testing the XInput USB interface, we could hardcode several keys to behave as XInput controls, like the keys of the Mouse layer. For example, Mouse+{i, j, k, l} could move the joystick via XInput. Alternatively, we could use the trackpoint module to do the same. We only need the XInput interface exposed in the firmware, and we'll implement the transformation of the data and integration with Agent from that point.

We're happy to assist you to hook to the above actions for testing purposes. It'd be worth starting with a hardcoded mapping scheme for development purposes, but first, we need the list of controls that can be implemented via XInput.

benedekkupper, 2022-04-05:
So I'll need some clarification on whether we specifically want an XInput simulated, or just an HID gamepad. For a differentiation of features, see [this link](https://docs.microsoft.com/en-us/windows/win32/xinput/xinput-and-directinput).

**XInput** is a completely custom USB protocol, and requires the entire USB device to act as an XBOX controller. See some details [here](https://www.partsnotincluded.com/understanding-the-xbox-360-wired-controllers-usb-data/) and [here](https://espruino.microco.sm/api/v1/files/11349d480e49ef1d74b3b76f600b016e7a91a0e6.pdf).

We can easily implement an HID device that is picked up by the **DirectInput** driver, for the possible HID usage codes see the [Microsoft reference](https://docs.microsoft.com/en-us/windows/win32/xinput/directinput-and-xusb-devices#gamepad). I believe this link should be your starting reference point to understand what controls can be exposed meaningfully as an HID gamepad. (In theory tools such as [DsHidMini](https://github.com/ViGEm/DsHidMini) could be used to mask the HID interface(s) as an XInput device, but this would require further investigation.)

mondalaci, 2022-04-06:
@benedekkupper Thanks for the links!

I strongly prefer XInput because the legacy DirectInput is not recommended by Microsoft anymore, and I assume the latter will be unsupported by newer games.

We should implement the XInput Gamepad interface. I'm unsure which controls are analog and what's their range. I assume the left and right sticks are analog, but there may be others.

benedekkupper, 2022-09-05:
### Progress so far
I have based my work on available online sources and a Wireshark capture of the Wooting One keyboard, which has XInput support without any driver installation.

I'm developing the proof-of-concept with my own USB library for simplicity. This setup consists of a basic keyboard interface and an XInput one - to make sure that the XInput driver isn't hardcoded to any specific interface index, or endpoint addresses. I have copied the configuration descriptor elements themselves (which are quite well documented [here](https://www.partsnotincluded.com/understanding-the-xbox-360-wired-controllers-usb-data/)) and tested the enumeration. Windows failed to recognize the XInput interface. Since all the open-source libraries I found were (re)using the USB VID:PID of a real XBox controller, but the Wooting One manages to use its own IDs, I suspected that something's missing.

Looking at the Wireshark capture shows some strange messages: the Info only says `Unknown type 7f`, and the raw data from USBPCAP also seems very strange, definitely not showing the real USB transfer. I did spot the `XUSB20` string in the last response, which reminded me of the Microsoft OS descriptors. As I have this implemented in my library, I could quickly enable it and use this `XUSB20` as CompatID. Voilá, the XInput interface is recognized! Some further testing and research has led me to find that this ID is documented by Microsoft in the [1.0 version](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-os-1-0-descriptors-specification), but funny enough the enumeration only works if I use version 2.0 of MS OS descriptors.

This solution will only allow automatic enumeration on Windows, other platforms will need their own solutions (e.g. [patching](https://github.com/paroj/xpad/blob/master/xpad.c) VID:PID for Linux).

The next step was sending some actual input data through this interface. Using the report layout described in the first link, I could verify that the data is accepted and interpreted correctly, using https://gamepad-tester.com/ website. (There is a Windows tool called joy.cpl that only works partially, for some reason in some directions it doesn't detect the cursor movement.)

### Up next
The MS OS descriptor provision will need to be added to the NXP USB stack so the interface can be enumerated. Then the interface logic itself will be added (new endpoints, descriptors). Focusing on the input side first, I'm not sure how much use we can get out of LED and rumble feedback information from the host.

### Rant
Microsoft (co-)authors the HID specification to create a very flexible and generic input device protocol. This is so nice that it is ported to many other transport layers, e.g. Bluetooth, I2C, etc, and supported by all major OS-es. Then this same Microsoft comes and throws out HID in favor of a non-standardized, closed spec, rigid XInput protocol, and discontinuing (or planning to discontinue) support for HID-based gamepad devices. Due to the lack of standardization, most (cheap) third-party gamepad controllers are reusing the USB VID:PID of the XBox controllers, leading to [painful compatibility issues](https://github.com/paroj/xpad/commit/bc5a06d22fe927857255e7f4433d701df3f4969c). This is all without considering that XInput is also somehow replacing HID over Bluetooth for controllers...

mondalaci, 2022-09-07:
@benedekkupper Thanks for sharing your findings! Your write-up is both interesting and painful to read at the same time, and I feel for you.

Any ways around adding our own VID:PID to the kernel? It's quite an absurd requirement.

To make matter worse, I planned to use distinct PIDs per product, but it can take quite a while for users to use new kernels that contain our VID:PIDs, so maybe we should use the same PID for every product going forward. Or any better ideas?

What about Mac support?

On second thought, it'd be useful to support legacy DirectInput eventually for compatibility with older applications, but it's not a priority.

benedekkupper, 2022-09-07:
My proposal for dealing with all these compatibility issues is this:
The FW presents both an XInput and an HID DirectInput gamepad interface. When the FW detects that the host is interrogating the MS OS descriptors (which only Windows OS should be doing), it sets the active gamepad to XInput, otherwise the HID gamepad interface is used. The HID report descriptor can be defined in a way that it matches the predefined XInput report layout. So the report data assembly logic can be uniform, the FW only routes the reports to different endpoints depending on the selection flag.

mondalaci, 2022-09-07:
Genious fallback mechanism, I love it!

benedekkupper, 2022-09-28:
I have done a lot of further investigation into the HID gamepad interface (which I will summarize later on), the funny thing is that Microsoft actually has an HID extension driver called "XINPUT HID Filter Driver" (xinputhid.INF), which only binds to Microsoft produced USB devices automatically. So it's possible to use HID on Windows as well, but the user would need to do a manual swapping of the device driver in Device Manager.
Anyway I was pursuing the original plan as laid out above, but I've hit a roadblock, if I enable the use of the MS OS descriptor on the UHK, Windows fails to enumerate the device entirely. The only saving grace is that I can use a Mac / Linux computer to reflash it to a working firmware. The progress can be checked [on this branch](https://github.com/benedekkupper/UHK-firmware/tree/usb-gamepad). At this point I'd need some way to debug the firmware, but unfortunately I only have a fully assembled unit with me.

mondalaci, 2022-09-29:
Thanks for keeping us up to date!

Very interesting findings. Let me contact you by email shortly about additional gear to be provided to you.

mondalaci, 2022-10-26:
Resolved by #535.

closed 2022-10-26 by mondalaci