UltimateHackingKeyboard/firmware #381: Key Chatter Often Munges Characters

StitiS, 2021-06-07:
Hey Team,

After about 1 year on the previous SW, I updated to the latest Agent (1.5.14) and FW (8.10.9) last week.  I've noticed key chatter is an issue now, with errors popping up once every couple minutes.  I've seen double letters on "cc", "tt", "uu", "mm", "aa", as well as letters popping up twice but out of sequence.  Something like "thingsg".  I've tried with a couple profiles, as well as a couple computers and there isn't any improvement or changes.

I tried down-reving the FW, but it errored out.  Maybe I'm doing something wrong, or there are limits on reverting or how far back I can go?

mondalaci, 2021-06-07:
Hi there! You should be able to flash earlier firmware versions. Let's figure out first whether this issue is related to firmware versions. Please try to downgrade your firmware, and let us know if you can't for some reason along with the firmware update log.

mondalaci, 2021-07-23:
Closing this issue due to the lack of replies. I find it unlikely that this issue is firmware-related, and it's worth checking [this guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys).

closed 2021-07-23 by mondalaci