UltimateHackingKeyboard/firmware #383: Trackpoint module pointer jumping

mondalaci, 2021-06-29:
When the trackpoint module is used, the pointer sometimes jumps. That is, its position suddenly changes. According to my experience, the relative distance of the jump is constant. I think the PS/2 protocol parser of the trackpoint is buggy which results in this issue.

ewancoder, 2021-09-08:
I'm experiencing a similar (but maybe different) issue. Trackpoint generally works good and I can move it and scroll articles with it. However when using Xodo Windows 10 application to read PDF files and trying to scroll down, trackpoint jumps back up sometimes as far as the point where I started scrolling from, or higher, making it nearly impossible to scroll the document. Basically the page just flies up and down all the time. Example of this can be found here: https://www.dropbox.com/s/g2jllgd1p8kyv17/trackpoint%20scroll%20issue.mkv?dl=0

mondalaci, 2021-09-08:
If other applications, like your web browser, exhibit this issue, it's likely a UHK related issue. Otherwise, it's an application-specific issue.

ewancoder, 2021-09-08:
I didn't manage to reproduce it anywhere except Xodo application so far. However I don't experience this issue with separate mouse, with mouse mode of the UHK or with ThinkPad laptop trackpoint in Xodo. I'll try to use some other applications and see.

kareltucek, 2021-09-08:
@mondalaci, have any reliable way to confirm your suspicion? 

I am wondering if the acceleration acts some part in it. Maybe not causing it, but might be amplifying the issue?

mondalaci, 2021-09-08:
@kareltucek I don't have a reliable way to confirm my suspicion, but scrolling does behave weirdly on some webpages where the scroll handler is overwritten. I think the scrolling mode of modules emits an unusually large number of scroll events in a given timeframe that some applications don't handle properly.

kareltucek, 2021-09-08:
Interseting observation. I haven't noticed such behaviour myself, with trackball module. (I use the scrolling mode with right hand modules exclusively.)

I guess I should be using trackpoint and touchpad modules more for sake of testing...

ewancoder, 2021-09-09:
Not sure if it's helpful but I just set acceleration to 0 using @kareltucek firmware and the issue still persists. (I have baseSpeed 0.4 and speed 0.7). The issue is also present with the official firmware.

ewancoder, 2021-09-10:
I've found that it is reproduced in Windows 10 list of audio devices (I have a lot of audio devices and I can scroll them down or up):
[image]

mondalaci, 2021-09-10:
@ewancoder A screencast would be very useful featuring affected and unaffected applications.

ewancoder, 2021-09-10:
Here I first scroll it slowly and then faster. When you press on trackpoint harder it seems to escalate the issue. In settings / start menu windows it also repeats but not while scrolling the list, just at the end when you already scrolled it all the way up and down it jumps a bit: https://www.dropbox.com/s/kmjlmh4ykc26w1h/trackpoint%20scroll%20issue%202.mkv?dl=0

mondalaci, 2021-09-10:
Can you reproduce this issue with both 8.10.9 and 8.10.10 firmware versions?

ewancoder, 2021-09-10:
I'll try to do this in couple hours.

ewancoder, 2021-09-10:
Yes it does reproduce on 8.10.9 as well as 8.10.10.

kareltucek, 2021-09-11:
@ewancoder can you try with this firmware? 
[uhk-firmware-8.10.10-no_accel.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7147303/uhk-firmware-8.10.10-no_accel.tar.gz)

ewancoder, 2021-09-11:
The issue is still present on this firmware as well. I'm starting to wonder maybe it's a hardware issue with my trackpoint if you guys can't reproduce it. But then it would probably show up in all kinds of applications..

kareltucek, 2021-09-11:
I admit that I am quite unsure of what we are dealing with. 

Laszlo claims to have seen some sort of trackpoint jump issue, as well as you @ewancoder  and @linfeng. 

Stock firmware still runs unpatched acceleration driver which may theoretically divide by zero time to time and produce ugly results (in case it is called twice in one millisecond - which is unlikely). However, you @ewancoder can reproduce the problem even with firmware that has the acceleration driver removed, but only in some apps.

Last time I saw similar problem with my setup, it was with high acceleration exponent on an unpatched firmware. After patching it, the problem went away for good.

On the other hand, @linfeng can reproduce the problem with a patched acceleration driver, but claims that the no_accel firmware works fine for him fine.

All three of you claim to have problems with trackpoint only and there are no similar reports for other modules, which is at least suspicious. After all - if this was a quirk of specific applications, all three modules should produce similar results.

llinfeng, 2021-09-13:
> On the other hand, @llinfeng (typo with github username fixed) can reproduce the problem with a patched acceleration driver, but claims that the no_accel firmware works fine for him fine.

@kareltucek I'd like to take back [what I was reporting](https://github.com/kareltucek/firmware/issues/64#issuecomment-917336233). The jump happened on the third day with the `no_accel` firmware.

> I'm starting to wonder maybe it's a hardware issue with my trackpoint

@ewancoder I am hopeful that a future firmware update can solve the jump problem. I am dead-heart in love with the trackpoint business, and have collected four trackpoint modules in total. I have reproduced the jump on both old and new trakcpoint modules, with both the stock firmware (v8.10.10) and the modified `no_accel` firmware. (By "old", I am referring to the first module shipped this August under my crowdsourced order from 2018. The buttons on this "old" trackpoint module are a bit louder than the newer ones I ordered from the UHK store.)

kareltucek, 2021-09-13:
Yep, sorry for the late reply. 

>  I think the PS/2 protocol parser of the trackpoint is buggy which results in this issue.

You mean the code in `trackpoint/src/module.c`? Guess I will have a look then.

---

In the meantime, @llinfeng @ewancoder I have put together a firmware with some logging. If it's not too much to ask, then please:

1) Download and flash [uhk-firmware-8.10.10-tp_hunt_v1.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7156834/uhk-firmware-8.10.10-tp_hunt_v1.tar.gz) 
2) Create a macro with a text action containing `printStatus` text.
3) When you see "ERR" light up on the LED display, open some text editor and use the `printStatus` macro.

It checks for short delays between two accel driver invocations (<4 ms) and for high cursor deltas (over 100 in any direction).

(The firmware is based off the fork, so @llinfeng your TP macros will work. Also, the firmware is now patched to interpret any text actions as command actions, so try not to use any regular text action macros.)

mondalaci, 2021-09-13:
> > I think the PS/2 protocol parser of the trackpoint is buggy which results in this issue.
> 
> You mean the code in `trackpoint/src/module.c`? Guess I will have a look then.

Yes, exactly, and a fix would be much appreciated!

llinfeng, 2021-09-13:
>(The firmware is based off the fork, so @llinfeng your TP macros will work. Also, the firmware is now patched to interpret any text actions as command actions, so try not to use any regular text action macros.)

@kareltucek Noted, and thank you for building it off the fork. I just applied the firmware and will post back here what I catch when the cursor jumps again.

> When you see "ERR" light up on the LED display, open some text editor and use the printStatus macro.

Just to clarify, when the event does happen, is it advised that one should use **another** mouse to open the text editor? Or, it doesn't matter?

kareltucek, 2021-09-13:
>  will post back here what I catch when the cursor jumps again

You may get some logs even when cursor does not jump. Those are of interest too.

> Or, it doesn't matter?

Doesn't matter.

llinfeng, 2021-09-14:
> You may get some logs even when cursor does not jump. 

@kareltucek I only got the key to work for once. Please advise if I was setting the macros wrong. The key only worked for once immediately after flashing the UHK unit with the `tp_hunt` firmware. I did see a much larger vertical jump for the trackpoint, but there is no "ERR" text at the LED panel. When this happened, pressing the key mapped with the macro does not generate any text in a plain text editor.

[image]

I only have three macros defined as below. I don't think any of them are fancy enough to be counted as "regular text action macros".
[image]

llinfeng, 2021-09-14:
@kareltucek `ERR` did show up when the mouse jumped :) Here are the texts I got.
```
High delta encountered:
- x 249
- - y 0
- - d 19
- - m 4
- - pm 4
- Low 
LFF ETIME DIFF ENCOUNTERED:
- x 2
- - 
- d 2
- - m 4
- - pm igh delt4
- HNCOUELTA ntered:
- - x 0
- - 50
- - d 10
- - m 4
- - p4
-  necountered x 0
- - y 250
- - d 
- m 4
- - pm 4
-  encoured:
- - x 0
- - y 250
- - d 10m 4
- - pm 4
- 
```

Though, without the `ERR` light, the macro does nothing.

ewancoder, 2021-09-14:
@kareltucek unfortunately I won't be able to do any kinds of investigation for a week, I will have my hands back on my UHK only on next Monday

kareltucek, 2021-09-14:
> Though, without the `ERR` light, the macro does nothing.

That's fine. Just means that there is nothing waiting in the buffer to be printed.

> @kareltucek `ERR` did show up when the mouse jumped :) Here are the texts I got.

Nice. The text is a bit garbled - probably because the used editor does not handle fast input. But the value 250 supports Laszlos hypothesis, since it is general PS2 ACK message.

> @kareltucek unfortunately I won't be able to do any kinds of investigation for a week, I will have my hands back on my UHK only on next Monday

No problem, thanks for info!

---------------

Here is another build [uhk-firmware-8.10.10-tp_hunt_v2.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7160686/uhk-firmware-8.10.10-tp_hunt_v2.tar.gz). Logging is still present and has a little decreased trashold, but hopefully it won't be needed. 

I have altered the PS2 parser to check message format and reset itself if the format does not meet criteria, so it should hopefully no longer interpret trackpoint's reset sequence as streaming mode data... So, be sure to flash your trackpoint module and keep track of which of your units have been flashed.

Also, it is question why does the TP module interrupt streaming mode on its own - I assume it gets resetted for some reason. Maybe power issue?

llinfeng, 2021-09-14:
@kareltucek New log, with `tp_hunt_v2`. `ERR` showed up minutes after I flashed a separate set of UHK+TP in my office.

```
ForceSpeed: 0: 0: High delta encountered:
ForceSpeed: 0: 0: - x 0
ForceSpeed: 0: 0: - y 254
ForceSpeed: 0: 0: - d 8
ForceSpeed: 0: 0: - m 4
ForceSpeed: 0: 0: - pm 4
ForceSpeed: 0: 0: High delta encountered:
ForceSpeed: 0: 0: - x 0
ForceSpeed: 0: 0: - y 255
ForceSpeed: 0: 0: - d 7
ForceSpeed: 0: 0: - m 4
ForceSpeed: 0: 0: - pm 4
```

llinfeng, 2021-09-14:
@kareltucek From the same keyboard, `ERR` shows up again when I was moving the cursor downward. The cursor was sent up in a jumpy manner.
```
ForceSpeed: 0: 0: High delta encountered:
ForceSpeed: 0: 0: - x 0
ForceSpeed: 0: 0: - y 243
ForceSpeed: 0: 0: - d 9
ForceSpeed: 0: 0: - m 4
ForceSpeed: 0: 0: - pm 4
```

llinfeng, 2021-09-14:
And, one more `ERR`, when trying to move the cursor rightward.
```
High delta encountered:
- x 255
- - y 4294967295
- - d 8
- - m 4
- - pm 4
- 
```

kareltucek, 2021-09-15:
All high numbers around the reset, resend and error handling messages (`0xff, 0xfe, 0xfc`). I assume that they are off a few numbers because deltas accumulate and get summed together.

Here is another build with more aggresive sanitation. Not sure if this is the ideal approach to the problem, but I don't have a better idea. Again - trackpoint code is updated, so please be sure to flash it too.

[uhk-firmware-8.10.10-tp_hunt_v3.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7165834/uhk-firmware-8.10.10-tp_hunt_v3.tar.gz)

llinfeng, 2021-09-15:
> Again - trackpoint code is updated, so please be sure to flash it too.

Does the following suffice what you said above? I kept the trackpoint module on the UHK keyboard when flashing the `tp_hunt_v3` firmware.
[image]

Anyways, I need to rewind back to the `tp_hunt_v2` firmware as the trackpoint module now becomes very hard to use. It rarely responds to gentle pressures for refined movement, and gets stuck here and there when moving continously with heavier force.

kareltucek, 2021-09-15:
Yeas, should suffice.

> Anyways, I need to rewind back to the `tp_hunt_v2` firmware as the trackpoint module now becomes very hard to use. It rarely responds to gentle pressures for refined movement, and gets stuck here and there when moving continuously with heavier force.

I see. Negative values are served straight in two's complement, so they overlap with the message range. Will have to think about it a bit more.

llinfeng, 2021-09-15:
@kareltucek I feel like I am logging more jumps with the `tp_hunt_v2` firmware, compared to `tp_hunt_v1`. As of this morning, I am catching two of them in an hour. Moving forward, I will collect the texts from `ERR` events into the following file on Dropbox.
* [tp_hunt_logs.txt](https://www.dropbox.com/s/1btw5sgte8e4mzc/tp_hunt_v2_logs.txt?dl=0)

If there is a new firmware to test, I will log them in the same file as well.

kareltucek, 2021-09-15:
> Moving forward, I will collect the texts from ERR events into the following file on Dropbox.

I think those collected so far suffice. Nothing new to learn there.

kareltucek, 2021-09-15:
@llinfeng here is another version [uhk-firmware-8.10.10-tp_hunt_v4.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7172871/uhk-firmware-8.10.10-tp_hunt_v4.tar.gz) (I have refined logging mechanism so it should now give exact values rather than approximate, and optimized control flow a bit - so in case of race conditions it might behave significantly better.)

Please, give it a try and again collect any logs.

llinfeng, 2021-09-15:
@kareltucek I just caught a new jump with `tp_hunt_v4`. Gladly, the trackpoint now moves smoothly.
```
- b 24
- x 0
- y 244
- e 0
```

 Future logs will be kept here: https://www.dropbox.com/s/ztg5o3zxqopti51/tp_hunt_v4_logs.txt?dl=0

kareltucek, 2021-09-20:
@llinfeng I am sorry to say so, but the longer I am looking at these logs, the more sure I am that something is wrong with them. Specifically, they look like the `byte1` lines belong to different x/y deltas. (16/32 in `byte1` mark negative sign for x/y, which are furthermore negated on output, therefore (e.g.) 24 should come with positive x (strictly nonzero, since there is no "negative zero" in two's complement)) I am not sure how that is possible

kareltucek, 2021-09-20:
@llinfeng I am sorry to say so, but the longer I am looking at these logs, the more sure I am that something is wrong with them (for which I have of course only myself to hold responsible). Specifically, they look like the byte1 lines belong to different x/y deltas. (16/32 in byte1 mark negative sign for x/y, which are furthermore negated on output, therefore (e.g.) 24 should come with positive x (strictly nonzero, since there is no "negative zero" in two's complement). Furthermore, byte1 should always contain 8 bit.) The only explanation seems to be a race happening on data exchange, which is however improbable to happen every single time. 

Anyway, even despite the weird inconsistency of byte1, they refute my assumption about stray PS2 messages - the values are too random for that. So at the moment, I think it is more probable that the PS2 parser interrupts are getting delayed, resulting in corrupted `byte1` and effectively flipped bits of the x/y deltas.

As a result, for v5, I have:
- bulletproofed data exchhange.
- decreased priority of keyscan interrupt - this way, PS2 GPIO reads should have highest priority and therefore should hopefully not be delayed by anything.
- optimized out a few function calls and refactored the code to interrupt on falling edge of the signal only. (In case you ancounter any tp freezes, please let me know.)

So please give a try to [uhk-firmware-8.10.10-tp_hunt_v5.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7195759/uhk-firmware-8.10.10-tp_hunt_v5.tar.gz). 

@mondalaci in case this fix does not help, is it feasible to rewrite the parser in a blocking manner? E.g., would blocking implemenentation break I2C or some other interrupt-reliant behaviour?

llinfeng, 2021-09-20:
@kareltucek Here is the first logged event with the `v5` firmware. 

<details>
  <summary>Click to expand.</summary>

```
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967232
ForceSpeed: 0: 0: - y 42$(^&@&%:ForceSorced: 0ed:: - 0 0
F 8
ForceSpeed: 0: 0: - x 4294967227
ForceSpeed: 0: 0: - y 4294967274
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967223
ForceSpeed: 0: 0: - y 4294967274
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967218
ForceSpeed: 0: 0: - y 4294967273
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0:( 0: - xF($(^&pe
For: 0peed e 0: 0orce- y 4ed$(^&@&@
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967216
ForceSpeed: 0: 0: - y 4294967271
ForceSpeed: 0: 0 0: E )
 _ BCESorce: 0ed:: - 0Forc 42peed216
 0ceS B *
:orceSPEED: 02: 0: -0
F@($ES216
F 0eSpeed: 0:F): - y 4294967270
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294ed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294
```
</details>

-----

I just noticed that there is a new version `5b`. I will follow up in a separate reply for additional logs.

kareltucek, 2021-09-20:
Fortunately, those are all false positives - I have also decresed trashold a bit (from 80 to 63) - apparently too much.  

Please, continue with [uhk-firmware-8.10.10-tp_hunt_v5c.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7197952/uhk-firmware-8.10.10-tp_hunt_v5c.tar.gz) .

llinfeng, 2021-09-20:
Here is the log for `5b`. I will be flashing `5c` in a moment.

<details>
  <summary>Click to expand.</summary>
  
  ```
TimeTag: 2021-09-20 14:59:58, in the office. The firmware was flashed at 2021-09-20 14:32:47.
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967293
ForceSpeed: 0: 0: - y 4294967232
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967293
ForceSpeed: 0: 0: - y 4294967230
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967293
ForceSpeed: 0: 0: - y 4294967230
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 8
ForceSpeed: 0: 0: - x 4294967292
ForceSpeed: 0: 0: - y 4294967232
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 24
ForceSpeed: 0: 0: - x 11
ForceSpeed: 0: 0: - y 4294967231
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 24
ForceSpeed: 0: 0: - x 11
ForceSpeed: 0: 0: - y 4294967231
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 24
ForceSpeed: 0: 0: - x 10
ForceSpeed: 0: 0: - y 4294967229
ForceSpeed: 0: 0: - e 0
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 24
ForceSpeed: 0: 0: - x 11
ForceSpeed: 0: 0: - y
  ```
</details>

It also made my day when I first learned about this `Click to expand` business.

llinfeng, 2021-09-21:
> Please, continue with uhk-firmware-8.10.10-tp_hunt_v5c.tar.gz . (Which has _threshold_ set to 126.)

@kareltucek I guess `126` is too large? I saw the cursor jumped multiple times and nothing is caught. For the most recent event, no `ERR` text when the cursor jumps vertically.

kareltucek, 2021-09-21:
> @kareltucek I guess `126` is too large?

All events logged so far (which were logged with 100 and 80 thrasholds), were far larger (had the 128 bit set in the binary representation). However, this source seems to have gone away with the last version.

I do not think that simply increasing trashold will yield new information - current logging looks for a variety of problems other than just size of deltas, and does not report anything, so we would most likely see just a bunch of regular reports and not know what to do about them anyway.

kareltucek, 2021-09-21:
I don't think I can do much more about the parser. 

Here are two more versions:

- [uhk-firmware-8.10.10-tp_hunt_v6a.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7206410/uhk-firmware-8.10.10-tp_hunt_v6a.tar.gz) This version has short delay inserted at the beginning of the PS2 interrupt, since some sources recommend to sample values in the middle of clock window. It may improve matters, or make them worse.
- [uhk-firmware-8.10.10-tp_hunt_v6b.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7206487/uhk-firmware-8.10.10-tp_hunt_v6b.tar.gz) If v6a does not help, this firmware simply tries to suppress any sudden changes of deltas. In such case, it produces a `128` error code. If the original problem reappears, it will produce codes `254` or `255`. Other codes (<128) mark other data inconsistencies.

So please:
- give a try to v6a and if it does not help, move to v6b
- with v6b, log any errors for a day or two. 128 are not very interesting - you can easily trigger it intentionally with a strong swipe. Hopefully, no other errors should pop up, and no more jumps should appear. If jumps persist, then there is some entirely different problem.

llinfeng, 2021-09-24:
Log for `v6a`, caught for the first time after 2 full days of testing. @kareltucek Is this still a false positive log? I'll keep using `v6a` for now.

<details>
  <summary>Click to expand.</summary>
  
  ```
TimeTag: 2021-09-23 22:04:17, noticed the ERR upon reconnecting the left half. [The rubber foot at the bottom left corner fell off. Glued it back. :) ]
---
- b 56
- x 174
- y 208
- e 0

  ```
</details>

kareltucek, 2021-09-24:
Does it mean that there were no jumps during last two days, or does it mean that just no logs were produced?

The event indeed looks like one of those logged with v4, but since it is connected to reconnecting the left half, I would rather ignore it altogether.

llinfeng, 2021-09-24:
> Does it mean that there were no jumps during last two days, or does it mean that just no logs were produced?

There were jumps, but none was logged with `v6a`. Though, my vague sense is that the jumps get smaller in width/length. The jumps are also getting way less frequent compared to, say, `v4` where things were logged crazy. `v4` was also a truthful logger, in the sense that when the mouse cursor jumped, the `ERR` is shown on the left half.

kareltucek, 2021-09-24:
> Though, my vague sense is that the jumps get smaller in width/length. 

Yes, this is quite probable. 

The way I see it, there are two types of jumps:

- those caused by corrupted sign bits in byte1. Those result in ~250 pixel jumps since the sign bits effectively negate entire delta value. I think these might be caused by delayed interrupt handlers.
- those caused by corrupted delta values. Those would be less than 128 in most cases (except bits overflowing between bytes). I am however not sure how this could happen. These would be most likely be caused by an interrupt getting called too early - maybe wrong detection of signal edge? But in that case, entire bit sequence would get shifted, which does not seem to be the case, since in that case you would get reports with zero x/y and `- e 3/4` (bad parity bit).

(You have not been ignoring reports with x/y = 0 that would happen without any jump, right?)

> The jumps are also getting way less frequent compared to, say, v4 where things were logged crazy. v4 was also a truthful logger, 

I am wondering whether this observation is just a result of type 1 errors going away, or whether I have actually introduced a new problem. 

I have added the check again after the 6va delay -> please give a try to [uhk-firmware-8.10.10-tp_hunt_v7.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7227172/uhk-firmware-8.10.10-tp_hunt_v7.tar.gz)

kareltucek, 2021-09-24:
I have had some luck. Storing for future reference, and making a few adjustments -> [uhk-firmware-8.10.10-tp_hunt_v7b.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7228744/uhk-firmware-8.10.10-tp_hunt_v7b.tar.gz)

<details>
<p>

```
v7
: 0: 0: ---
: 0: 0: - b 24
: 0: 0: - x -517
: 0: 0: - y 918
: 0: 0: - e 0
---
- b 8
- x 0
- y 0
- e 254
--
- b 24
- x 0
- y 0
- e 254
---
- b 8
- x 0
-y 0
- e 254

halfway between v7 v7b
---
- b 24
- x 0
- y 0
e 38
---
- b- x 0
- y 0- e 36
---
- b 8
- x 0
- y 0
- e 4
---- b 8
-x 0
- y 0
- e4
---
- b 8
- x 0
- y0
- e 6
---
- b 40
- x 0
- y 0
-e 38
```
</p>
</details>

mondalaci, 2021-09-25:
@kareltucek Rewriting the PS/2 protocol parser in a blocking manner would hardly result in an improvement. PS/2 runs at a baud rate of 10 kHz to 16.7 kHz, and the KL03 MCU runs at 48 Mhz, so it's able to handle PS/2 without breaking a sweat. A busy loop would unnecessarily increase power consumption, and I'm confident that using interrupts is a better approach.

Can you reproduce the issue? If so, a logic analyzer would be very helpful, and we can purchase one for you.

kareltucek, 2021-09-25:
> Can you reproduce the issue?

Not much. I see some artifact once every few days...

llinfeng, 2021-09-26:
@kareltucek Here is what I just caught using `v7b`. See [here](https://www.dropbox.com/s/1ohuyqix4ku06y3/tp_hunt_v7b.txt?dl=0) for the full log.

<details>
  <summary>Detailed log</summary>
  
  ```
ForceSpeed: 0: 0: ---
ForceSpeed: 0: 0: - b 24
F_ORCESpeorce 0:ED: 0 x 0: - e 166
---
- b 8
- x 0
- y 0
- e 166
---
- b 24
- x 0
- y 0
- e 166
---
- b1 8
- x - by 0
 0
66
-0
- e 132
---
- b 8
- x 0
- y 0
- e 132
---
- b 8
- x 0
- y 0
- e0
 132
--- e 132--
- b 8
- x 0
- y 0
- e 132
---
- b 8
- x 0
- y8
- - e - y 0
- e 132
---
- b 8
- x 0
- y 0
- e 132
---
- b 8
- x 0
- y 0
- e 132
---
- b132
-- - b 8
- x 0
- y 0
- e 132
---
- b 8
- x 0
- y 0
- e 132
---
- b 2
---

 8
- x 0
- y 0
- e 132
---
- b 8
- x 0
- y 0
- e 134
---
- b 24
- x 0
- y 0
- e 166
--- e 16 8
---
- b - x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
x 0
- y 0
- e 134
---
- b 40
- x 0
- y 0
- e 166
  ```
</details>

ewancoder, 2021-09-27:
Lately I'm having all kinds of issues with my UHK. The freezing/jittering issue was not fixed by disabling turning off devices in device manager, it appeared again. Also when I'm using my trackpoint I did notice that it jumps to random location from time to time. Furthermore, it freezes from time to time and sometimes right mouse button click is triggered even if I don't touch the keyboard (or possibly when I move the trackpoint or when I click left mouse button on the keycluster module). Sometimes it also misses a key so I mistype something as a result. I'm not sure if this is all triggered by a trackpoint module or by the keyboard itself or by my motherboard / faulty usb ports etc. I've tried it with two different usb 3.0 ports, no devices are connected to usb 2.0 ports. I didn't try it with another machine yet. Right now when using left mouse button on keycluster module it actually clicked right mouse button btw.

I was away for a while, could yo please suggest what I can do to gather some logs etc that might help troubleshoot these issues? (there it skipped 'u' key in the you wword, and here it typed w multiple times). Not sure what is happening and why, it was never like this previously, fels like something has degraded.

ewancoder, 2021-09-27:
It looks like right mouse click and missing keys might be because of keycluster module, I disconnected it and so far didn't have any issues.

kareltucek, 2021-09-27:
Thanks @llinfeng ! 

Looks like there are still some delayed interrupt calls. I think I will rework logging so that we can actually see the timing. Current solution is very simple and unfortunately does not yield very detailed info.

-----

@ewancoder this sounds like a problem specific to your configuration. Start by general troubleshooting:
- flash the keyboard and halves again to the latest official firmware. Make sure that Agent actually flashes all the units - wait to see in the log that every connected module is flashed.
- make sure that your modules and all the steel guides hold firmly in place. If some of the modules is unresponsive on I2C, it may make the keyboard itself unresponsive.
- regarding the right button, make sure that you have it mapped consistently on all layers.
- do specific keys misbehave? Could it be simply a faulty switch? 
- if you can, try with another computer. This might also be a power issue.

> (should I use v7b?)

I don't see a point in it. This is more general problem.

ewancoder, 2021-09-27:
I've tried flashing the official firmware it still gave me the same results. Everything feels firm. However for now I'm using the keyboard with trackpoint module only, without keycluster, for half a day, and I didn't have any issues with missing keys or right mouse button presses so I'd say it's coming from the keycluster module. I still have trackpoint issues though: 1) occasional jumping to random location, which I think you are troubleshooting in this thread 2) when scrolling page tries to jump back all the time in some apps like windows sound devices list or Xodo reader which I did a screencast of recently. I'm not sure about freezing/jittering mouse issue because it didn't appear after I disconnected keycluster module but it's a very rare issue for me so I'll wait for a week to see if it happens again

kareltucek, 2021-09-27:
> ...keycluster...

I have no idea of what might be wrong. But as Laci suggests, next step is to try reproducing it with different computer. 

> when scrolling page tries to jump back all the time in some apps like windows sound devices list or Xodo reader which I did a screencast of recently

This is strange. Not sure how, but It would be a good idea to find out why just some software is affected.

> 1) occasional jumping to random location, which I think you are troubleshooting in this thread

It is hard to tell whether it is actually the same issue though, and from what you write, it sounds like the smallest issue.

Anyway, please wait till I get back with next debug firmware iteration.

mondalaci, 2021-09-27:
@ewancoder Your UHK exhibits an unusually large number of issues, and chances are they're mostly power-related. [This guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys) may help to narrow down the cause of the issue.

What do you mean by saying that your trackpoint module freezes?

Please try to reproduce the mentioned issues on another computer.

ewancoder, 2021-09-28:
By freezing I refer to the other issue that was closed here already, when mouse cursor starts stuttering when you move it, jumping from place to place with huge delays instead of moving smoothly. Yeah maybe it's my hardware I dunno. I'll try to reproduce all these issues on a laptop when I have time.

kareltucek, 2021-09-28:
@llinfeng  @ewancoder here is [uhk-firmware-8.10.10-tp_hunt_v8.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7245606/uhk-firmware-8.10.10-tp_hunt_v8.tar.gz). 

Please, give it a try and:

- gather any logs.
- let me know whether jumps persist.

ewancoder, 2021-09-30:
Jumps do not occur for me so far on v8, maybe it's too early to say though (one full day of trackpoint usage).

[46 more replies]

PR #419 (2021-11-14): Trackpoint jumps fix
Closes #383 .
closed 2021-11-14 by mondalaci