UltimateHackingKeyboard/firmware #384: Plugging in external monitor causes UHK to become unresponsive when split

laxu, 2021-07-12:
Using:
* UHK v1
* Firmware 8.10.10

So this is a bit of a weird one. I recently replaced the power supply for my old ASUS PG278Q and when that display is plugged in and powered, UHK will become unresponsive or it will cause the keys to lag and start dropping inputs or repeating them.

It seems that if the keyboard is connected together the issue goes away. During this time my other peripherals work fine.

Could this be caused by #74 ? I have not replaced those resistors. The power supply I got is some generic one with matching specs to the one that came with the display but I wonder if it's bad quality and causes some kind of power issues that then affects the UHK.

laxu, 2021-07-12:
Yeah it definitely seems to be caused by that generic power supply. As long as it's plugged into my monitor, does not even need to have the monitor connected and powered it will cause issues with the UHK. I guess I will just return it.

mondalaci, 2021-07-12:
The UHK is relatively sensitive to noisy power, and there's not much we can do about it at the moment, unfortunately. This issue is hard to reproduce, and we'll work with our contractors to improve our design in this respect. In the meantime, high-quality power supplies should fix the issue, and it's also worth using a straight (non-spiral) cable that is as short as needed.

closed 2021-07-12 by mondalaci

ftes, 2022-01-28:
I'm also experiencing connectivity issues after switching to a 3rd party power adapter for my Macbook.

mondalaci, 2022-01-28:
@ftes Please try to narrow down the cause based on points 1 to 9 of [this guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys), and let me know your findings.

ftes, 2022-01-28:
Thank you, step 7 fixed it.
Noisy power supply seems to be the issue. I did change that recently.