UltimateHackingKeyboard/firmware #385: Trackpoint module momentarily unresponsive when used in strokes

hsszyman, 2021-07-12:
When using the trackpoint module in a stroke-like fashion (rather than maintaining constant contact with the point, contact is reapplied with direction) resulting in some strokes not producing movement ( usually not the first stroke )

mondalaci, 2021-07-12:
Does the trackpoint module momentarily disappear in Agent when this issue happens?

hsszyman, 2021-07-12:
nope it stays connected

mondalaci, 2021-07-12:
Can you reproduce a similar issue when only using the mouse keys of the mouse layer, and the trackpoint is connected but not used?

hsszyman, 2021-07-12:
mouse layer doesn't seem to be doing this either. this only happens when the pressure to the trackpoint is applied in strokes and it seems inconsistent when it happens

mondalaci, 2021-07-12:
Can you make a screencast that shows this issue?

hsszyman, 2021-07-12:
https://drive.google.com/file/d/1PeOBUR9TqI9eKQzkHAgXtGAO9gfeccPA/view?usp=drivesdk

hsszyman, 2021-07-12:
each stroke is happening at the same time, but you should see pauses in the cursor intermittently. That's when the cursor is unresponsive the the movement when placing the finger back on the nub

mondalaci, 2021-07-12:
Thanks for sharing. I can notice the issue. Please give a shot to [this guide](https://ultimatehackingkeyboard.com/repair/fastening-module-steel-guide-screws), and let me know whether it fixes the issue.

hsszyman, 2021-07-12:
just tried it. didn't seem to change anything. Module also doesn't feel loose

mondalaci, 2021-07-12:
Then we shall replace your module. Please shoot an email to support@ultimatehackingkeyboard.com and we'll arrange it.

Thank you for your patience and understanding!

closed 2021-07-12 by mondalaci