UltimateHackingKeyboard/firmware #386: Key cluster module trackball can have an effect after release

laxu, 2021-07-18:
I have experienced an interesting issue after using the little trackball on the key cluster module. Basically what seems to happen is that despite no longer scrolling on the trackball, it will still have that effect and can affect the next key pressed.

As an example: I scroll a webpage using the trackball. I stop scrolling. I then press CTRL and I can see the website zoom in, as if you used CTRL+scroll wheel. I can consistently replicate this. So it seems like the trackball commands seem to keep triggering after it has stopped. This only happens once so it will not keep zooming but just sends one CTRL+scroll up or down command.

Using firmware 8.10.10 and latest Agent.

kareltucek, 2021-07-18:
> I can consistently replicate this.

Interesting, I can't reproduce it. 

If you think it might be keymap-specific, would you share your UserConfig?

mondalaci, 2021-07-19:
I can't reproduce this issue either.

@laxu Please try to reproduce it on multiple computers and let us know your findings.

laxu, 2021-07-19:
I don't think it's keymap specific because if I swap between two keymaps that have CTRL mapped to different keys I can get it to happen on both. 

I can map any key to CTRL and have this effect happen.

I can also get this to happen if I scroll with the key cluster trackball, then type a few letters and again press CTRL. 

Interestingly no other modifier seems affected, only CTRL. Or they just don't have any modifier + scroll action that would produce a visual difference.

mondalaci, 2021-07-20:
@laxu Can you reproduce this on other computers?

laxu, 2021-07-21:
My setup has the UHK and my mouse connected to a CableMatters USB switch which is connected to both my PC and my Mac. So I can try the same setup on both.

I cannot replicate the issue on my Macbook Pro 2019. I will have to see if this is a Windows only issue or if something else in my system could cause this.

mondalaci, 2021-07-21:
@laxu Please disconnect your mouse and try to reproduce the issue. If it still persists, connect your UHK directly to your Windows computer without the CableMatters USB switch. Chances are the switch causes the issue.

laxu, 2021-08-01:
Ok I figured it out. This was somehow caused by a software called X-Mouse Button Control which allows mouse button remapping. I've been using this with my Logitech mouse so I don't have to install their awful software. 

I need to see if updating that or changing some configuration setting can disable it for UHK.

closed 2021-08-01 by laxu