UltimateHackingKeyboard/firmware #388: Firmware flash of key cluster interrupted

autoaliased, 2021-07-26:
In a beautiful demonstration of Murphy's Law, my keyboard cable got snagged and briefly disconnected while the firmware flash for the key cluster module was in progress.

Everything else still works (left and right keyboard halves, and the trackball module), but the key cluster is no longer recognized in Agent and does not light up when connected.  Is there any way to save it, or is it totally bricked?

mondalaci, 2021-07-26:
Fear not! the latest Agent 1.5.15 version has a module unbricking mode which you can activate by right-clicking on the Flash Firmware button of the Firmware page, and your module shall be unbricked.

Please follow up on how it went.

autoaliased, 2021-07-26:
Success!!!  (Phew!!)  Nice feature addition!  :)

Thanks for your help, Laszlo.  And thanks for a beautiful product!

closed 2021-07-26 by autoaliased