UltimateHackingKeyboard/firmware #389: Keycluster module keys occasionally stop working

mhantsch, 2021-07-28:
Strange behaviour, not sure how to best test / report this. It has happened twice in the past week, so I am beginning to think it's not just a one-off fluke.

Right keyboard half firmware: 8.10.10
Left keyboard half firmware: 8.10.10
Key cluster firmware: 8.10.10
Touchpad is detected, and its firmware is implemented by the right keyboard half.

Agent version: 1.5.15

UHK60 v1 with touchpad and keycluster modules. When I used my UHK this morning, I noticed the keys on the keycluster are not working. They don't do anything. The scroll wheel (ball) still works. The keys on the regular keyboard halves work fine. All layers work. But they keycluster keys don't do anything. No mouse clicks, no key presses.

If I unplug the keycluster and plug it back in, no change.
If I unplug the complete left half and plug it back in, no change.
If I unplug the touchpad and plug it back in, no change.

The first time this happened, I unplugged the complete UHK from the USB port and re-connected it, then the key cluster worked again.
The second time this incident happened, I opened Agent, and re-saved the keyboard configuration into the keyboard, now the key cluster works again.

Any suggestions what else I should try / do if it happens again?

mondalaci, 2021-07-28:
Please [this guide](https://ultimatehackingkeyboard.com/repair/fastening-module-steel-guide-screws) a shot, and let me know whether it works.

mhantsch, 2021-07-28:
> Please give [this guide](https://ultimatehackingkeyboard.com/repair/fastening-module-steel-guide-screws) a shot

I am happy to try this, but I am confused.

Why should those screws change anything? The issue was not solved by unplugging/replugging the key cluster, so it cannot be a contact problem. Re-saving the configuration from Agent solved it; so that sounds more like a software issue?

Also, why would the scroll ball still work, and only the keys fail to do anything?

mondalaci, 2021-07-28:
The connection of your modules may be unreliable due to a loose mechanical connection which may mess up the I2C stack that can result in all kinds of issues. That's my theory for the time being, and I'll dig deeper if the issue persists.

mhantsch, 2021-07-28:
OK, understood. I'll reseat the connection and observe whether the issue reoccurs.

mondalaci, 2021-08-21:
Closing this issue due to the lack of replies. Feel free to follow up, and I'll reopen it if needed.

closed 2021-08-21 by mondalaci