UltimateHackingKeyboard/firmware #390: 8.10.10 firmware causes problems with kvm switch

AnsonT, 2021-08-03:
After upgrading from 8.10.9 to 8.10.10, UHK isn't recognized after switching between 2 PCs on a USB KVM switch.

Starting UHK Agent will restore functionality.

mondalaci, 2021-08-03:
@abcminiuser Thanks, great idea!

For the sake of the easiest testing, I created a release for every commit between firmware version 8.10.9 and 8.10.10 which may cause this issue.

[uhk-firmware-8.10.9-01-21594d1.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923453/uhk-firmware-8.10.9-01-21594d1.tar.gz)
[uhk-firmware-8.10.9-02-55fa6bd.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923455/uhk-firmware-8.10.9-02-55fa6bd.tar.gz)
[uhk-firmware-8.10.9-03-3c0924b.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923456/uhk-firmware-8.10.9-03-3c0924b.tar.gz)
[uhk-firmware-8.10.9-04-412d935.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923457/uhk-firmware-8.10.9-04-412d935.tar.gz)
[uhk-firmware-8.10.9-05-5d46451.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923458/uhk-firmware-8.10.9-05-5d46451.tar.gz)
[uhk-firmware-8.10.9-06-5702bd4.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923459/uhk-firmware-8.10.9-06-5702bd4.tar.gz)
[uhk-firmware-8.10.9-07-191e915.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923460/uhk-firmware-8.10.9-07-191e915.tar.gz)
[uhk-firmware-8.10.9-08-6fc356e.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923461/uhk-firmware-8.10.9-08-6fc356e.tar.gz)
[uhk-firmware-8.10.9-09-0f1f4e9.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923462/uhk-firmware-8.10.9-09-0f1f4e9.tar.gz)
[uhk-firmware-8.10.9-10-f6a8135.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6923463/uhk-firmware-8.10.9-10-f6a8135.tar.gz)

The first number after the version number means the chronological order and the second is the commit id.

@AnsonT @jfieber Please test the commits, guys, and let me know where things go bad.

AnsonT, 2021-08-03:
The issue started on uhk-firmware-8.10.9-08-6fc356e.tar.gz

mondalaci, 2021-08-03:
@AnsonT Thanks for your feedback!

@abcminiuser According to the above feedback, 6fc356e31dd6b8d1cc11de326901109746c21874 is the culprit. Any ideas why?

AnsonT, 2021-08-05:
This check-in introduced a 4sec delay in UsbBasicKeyboardCheckIdleElapsed(), but I don't know the USB protocol to understand the purpose.

abcminiuser, 2021-08-07:
It's a (multiple of) 4ms delay, for the idle report period -- the timer functions report in microseconds rather than milliseconds, so I have to do a x1000 multiplier.

The idle period is a part of the HID specification that allows a host to configure the device to automatically re-send reports after a configured interval, even if the report contents has not changed. This is used by simple BIOS and other applications, which will set an idle period for automatic key repeats.

By default, the idle period is zero for each interface, meaning infinity (only changes in reports are sent to the host). Idle periods are configurable in 4ms units, from 0 (meaning infinite period) up to a raw value of 255 (1020ms).

I see one mistake I've done there; the millisecond-to-microsecond idle period is stored in a `uint16_t` instead of an `uint32_t` which may overflow. It's probably saner to instead use milliseconds as the unit and divide the timer function's result by 1000, but in any case this likely isn't the root cause of the current compatibility problems.

mondalaci, 2021-08-07:
Just merged #393 which may fix the issue, and built [uhk-firmware-8.10.10-1.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/6949310/uhk-firmware-8.10.10-1.tar.gz)

Please give it a try, guys, and let me know whether the issue still persists.

jfieber, 2021-08-08:
Yes, it takes somewhere between immediately and about 30 seconds before the the system becomes responsive to keyboard input, switching between two Macs with macOS 11.5.1 with the "ugreen" USB switch I linked above.  The delay is quite inconsistent. There is no delay in the device being recognized as plugged in, but something goes amiss with getting everything hooked up. I'm not exactly sure what to look for in the system logs, but just searching for "UHK" shows elapsed time (15 seconds in this case) between the kernel noticing it plugged in to when (loginwindow, apparently) recognizes it and wires it up, after which keyboard input is recognized by the system. 

[image]

abcminiuser, 2021-08-08:
Does the delay occur when plugging the keyboard in directly to the Mac, rather than via the switch? A delay like that sounds like a control request timing out.

I've worked with the Mac APIs over the last year quite a lot, and calling them "barely functional and entirely un-debuggable memory leaking piles of junk" would be speaking a little to favourable of them.

jfieber, 2021-08-08:
Back to the 8.10.9, the mysterious trip to la-la-land doesn't happen.

[image]

Will reflash with 8.10.10-1 again and see if the delay happens when plugging in, no switch involved…

jfieber, 2021-08-08:
Yes, the inconsistent delay also happens with 8.10.10-1 when just plugging the keyboard directly in with no intermediary hubs or switches.

abcminiuser, 2021-08-08:
Reproduced, and I believe I've found the issue.

Could you please try https://www.fourwalledcubicle.com/files/temp/uhk-firmware-8.10.10-macfix.tar.gz and see if this resolves the issue for you?

jfieber, 2021-08-08:
Sadly, does not fix it for me. Tested on two different macs.

jfieber, 2021-08-08:
I can confirm @AnsonT's experience that the problem starts with uhk-firmware-8.10.9-08-6fc356e.tar.gz.

AnsonT, 2021-08-08:
My other observation is that the trackpad module does not have this delay, but I guess the mouse HID is quite different.

abcminiuser, 2021-08-08:
Ok, dug into this some more, but have run out of time for today. I bisected the patch's changes, and it seems that it can be "fixed" by changing:

```usb_status_t UsbBasicKeyboardCheckIdleElapsed()
{
    uint16_t idlePeriodMs = ((usb_device_hid_struct_t*)UsbCompositeDevice.basicKeyboardHandle)->idleRate * 4; // idleRate is in 4ms units.
    if (!idlePeriodMs) {
        return kStatus_USB_Busy;
    }

    bool hasIdleElapsed = (Timer_GetElapsedTimeMicros(&usbBasicKeyboardReportLastSendTime) / 1000) > idlePeriodMs;
    return hasIdleElapsed ? kStatus_USB_Success : kStatus_USB_Busy;
}```

To:

```usb_status_t UsbBasicKeyboardCheckIdleElapsed()
{
    uint16_t idlePeriodMs = ((usb_device_hid_struct_t*)UsbCompositeDevice.basicKeyboardHandle)->idleRate * 4; // idleRate is in 4ms units.
    if (!idlePeriodMs) {
        return kStatus_USB_Busy;
    }

    bool hasIdleElapsed = (Timer_GetElapsedTimeMicros(&usbBasicKeyboardReportLastSendTime) / 1000) > idlePeriodMs;
    return kStatus_USB_Busy;
}```

I.e. still doing the idle period test, but always reporting that the idle period has *not* elapsed. This isn't a real fix - I lack the debugging hardware setup to investigate further why this is causing an issue (if @mondalaci or someone else with a debugger/serial UART could add some logging, that would be very helpful). My guess is that the idle period is expiring far quicker than expected for some reason, which is causing the controller to get bogged down trying to send continuous reports - or it's related to the issue below.

The other part I noticed and am perplexed by is the ` UsbBasicKeyboardResetActiveReport()` function (and all the other reset report functions in their respective HID drivers). That's called *before* the report change check, and it zeroes out the current HID report. That shouldn't be done as it will break the idle period report repeats -- @mondalaci do you know the story behind this one? All the HID reports should be retained and sent to the host verbatim either when their contents change, or the idle period elapses. Clearing out the report every time will result in the idle period elapsing and sending an empty report to the host, rather than a repeat of the previous report.

I can see why the dual buffer reports is needed (to support the USB controller DMAing out the data to the host while the firmware prepares the next buffer) but there's a fundamental mismatch going on with the HID spec here. Each report should contain the current key/mouse/etc. states to reflect the current status, so that GetReport requests or idle period retransmissions work. To implement macro functions, you'd need to emulate a key release between two emulated key presses of the same key.

mondalaci, 2021-08-08:
@AnsonT The modules have nothing to do regarding this issue because they don't speak directly over USB.

@abcminiuser I'd be happy to make a build with the suggested workaround, but 1. I'm unsure whether I should apply them on top of [your latest commit](https://github.com/abcminiuser/uhk-firmware/commit/2235119543b94294fb35f1bdb420f7065e380d4d), and 2. whether I should apply the change for all the 5 interfaces.

As for the story behind the USB code, I originally used a KSDK example project which implemented a keyboard+mouse composite USB device. According to my knowledge, the quality of the KSDK codebase leaves a lot to be desired, so I wouldn't be surprised by anything.

I'm unsure what to log and would be glad to buy debugging hardware for you.

VboxNick, 2021-08-28:
+1, the same issue after upgrading the firmware to 8.10.10 with Agent 1.15.15. Device: UHK v1.

abcminiuser, 2021-08-29:
Sorry for the very late response everyone; I've been under some extreme work deadlines and frankly have been on the cusp of burning out.

Re: the above, you can apply that "fix" (breaking the idle period timer) to all the interfaces, on the latest HEAD. That should at least fix the compatibility and Mac init delay issues, at the expense of also breaking HID compliance again.

To fix the issues at hand, we need to check that the set idle timers per-interface actually expire at the correct intervals. The host may set an idle period in 4ms units per-interface, which should then trigger the firmware to re-send the last HID report at that interval if nothing has changed (with changes being sent immediately). This is used by BIOS and other limited hosts for key repeat.

A more fundamental review of the way HID reports are generated is also needed. Each interface should retain the previously sent report (for idle period timeouts) and only switch to the alternate buffer when a change has been made, so that the "current" report can be sent to the host in response to GET REPORT requests or idle period timeouts. Currently it looks like the firmware always zeros and swaps the buffer in advance of any changes, which breaks this.

Ideally, each interface has a "current" and "next" buffer (also acceptable, a queue of reports ready to be sent to the host). Each time a change has occurred, you mark it as the "current" buffer and send it to the host. Any time the idle period elapses or the host sends a GET REPORT request, send that current buffer back to the host. In the meantime, you can fill out pending changes in the "next" buffer(s) ready for dispatch once the previous report has been sent. This only gets tricky when you have a large amount of rapid state changes you want to communicate with the host (like for macros) as you would need to wait until the report buffer is read by the host before you generate and send off the next report containing the next few macro keypresses.

mondalaci, 2021-09-09:
@abcminiuser Thank you so much for helping us despite your current situation!

I've applied [your commit](https://github.com/abcminiuser/uhk-firmware/commit/2235119543b94294fb35f1bdb420f7065e380d4d) on the latest HEAD, and built [a new release](https://github.com/UltimateHackingKeyboard/firmware/files/7136814/uhk-firmware-usb-interface-workaround-8.10.10.tar.gz). Please test it, folks!

@kareltucek Any ideas on how to fix the issues mentioned in the last three paragraphs of @abcminiuser's last comment?

kareltucek, 2021-09-09:
> @kareltucek Any ideas on how to fix the issues mentioned in the last three paragraphs of @abcminiuser's last comment?

Well, restructuring the code so that report buffers are swapped before USB_DeviceHidSend rather than after should not be hard. Regarding rest of the comment, as well as UHK's USB stack, I have to admit that I am quite lost.

mondalaci, 2021-09-09:
@kareltucek My USB skills aren't exactly on par with @abcminiuser, either. Whatever improvement you can implement in this respect would be much appreciated.

kareltucek, 2021-09-09:
Allright, will take a deeper look at it.

jfieber, 2021-09-13:
> I've applied [your commit](https://github.com/abcminiuser/uhk-firmware/commit/2235119543b94294fb35f1bdb420f7065e380d4d) on the latest HEAD, and built [a new [...]

Sadly, this build does not solve the problem.

UnknownFreak, 2021-12-07:
I got this when I connected my UHK (v8.10.12) after an update today when I got home from office.
[image]
Apart from that usb devices were acting strange until I plugged the keyboard out of my kvm switch.
I did downgrade to my latest known version to work (which is v8.10.7) and I have no issues again.
I also tried v8.10.9 and it seem to work fine as well. (and that's what I am running at the moment)
I have not tried the "workaround" build yet, but will do so tomorrow if I get the time.

mikhey, 2021-12-08:
I have a [USB switch](https://www.amazon.com/gp/product/B00Y3CTEG2) to which I have a Windows machine (work) & my MacBook Pro connected.  I got this switch mainly because I was tired of pulling the USB from one dock and into another.

I recently upgraded the firmware and I noticed the a lag until the keyboard is detected when switching back to my MacBook Pro.  The Windows machine seems to work just fine - no lag.  I have the trackpoint and key cluster modules, both of which work without any lag when I switch to my Mac. 

To check when I can type, I press Mod+A which enables caps lock - if I see that light toggle on/off - I know I can start to type again.  It's "okay" now, but in 3 months, probably not so much.

On both machines it's connected via a Dell monitor USB hub, then to a Thunderbolt 3 dock

MacBook Pro -> OWC Thunderbolt 3 Dock -> Dell Monitor USB3 Hub -> Switch -> UHK
Work PC -> Thunderbolt Dock TB16 -> Dell Monitor USB3 Hub -> Switch -> UHK

benedekkupper, 2021-12-11:
Hello everyone, let me put my two cents to this discussion. As it seems the problem stems from adding GET_IDLE, SET_IDLE, GET_PROTOCOL and SET_PROTOCOL HID commands to the FW. I haven't examined the code itself yet, first I just want to talk about the merit of the SET_IDLE functionality itself. As it's interpreted in the current FW implementation, it allows the host to request the device to repeat the last report periodically without the contents having changed.

Well, that's stupid. It wastes the bandwidth of the USB bus, potentially preventing other interface endpoints from being serviced. Not to mention other transfer media that have HID supported. Here's what the Bluetooth HID profile spec says about SET_IDLE:

> SET_IDLE is deprecated and should not be sent by Bluetooth HID Hosts. If a
Bluetooth HID Host sends SET_IDLE, it shall only send it with an Idle Rate
parameter value of 0. This may be done to ensure that legacy Bluetooth HID
devices turn off their repeat function.
SET_IDLE was used to set a specific Idle rate of a Bluetooth HID device and is
deprecated.
NOTE: Though SET_IDLE was originally intended to enable HID keyboards (both
Bluetooth and USB) to provide key auto-repeat functionality for simple HID host
devices, this function is not provided by the Bluetooth HID Profile. Key auto-repeat
functionality must be provided by host software if desired.

The last sentence is the key in my opinion. Any sensible OS would implement auto-repeating by itself, instead of putting this responsibility on an external component, with potentially questionable reliability.

My proposal is to follow the Bluetooth spec's idea, namely that the SET_IDLE requests are either always rejected, or rejected when the duration isn't 0 (corresponding to infinity), and GET_IDLE always returns 0.

mondalaci, 2021-12-13:
@benedekkupper Thank you for elaborating on the suspected cause of the issue! Your explanation makes perfect sense, and please proceed with the suggested fix.

To recap, after the fix is implemented, the following should work:
- The UHK should be able to [wake up OSX](https://github.com/UltimateHackingKeyboard/firmware/issues/219). This is currently working, and it was the original issue that called for the USB-related changes that unfortunately caused more harm than good.
- KVMs should work with the UHK.
- The "USB controller resources exceeded" error on Windows should be a non-issue.
- The high CPU usage on Macs should be a non-issue.
- Ideally, the UHK should pass the USB HID compliance tool.

mondalaci, 2021-12-16:
Everybody affected by this issue, please test [uhk-firmware-8.10.12-usb-fix.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7725857/uhk-firmware-8.10.12-usb-fix.tar.gz), which is based on #432, and let us know your findings.

jfieber, 2021-12-17:
@mondalaci On a quick test, that appears to address the issues I was having with a USB switch, and general slowness in being fully recognized by macOS.

closed 2021-12-21 by mondalaci

jwr, 2022-01-24:
Are there any roadblocks to releasing this in an official firmware update? Us MacOS users are still burning about 10% CPU when the UHK is connected (#428), all day, and would really appreciate the fix.

mondalaci, 2022-01-24:
@jwr Yes, the next release will contain quite a few changes. Please use [uhk-firmware-8.10.12-usb-fix.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7725857/uhk-firmware-8.10.12-usb-fix.tar.gz) in the meantime.

jwr, 2022-01-25:
Thanks so much — that firmware gave me back my 10% CPU and made me very happy :-)

mikhey, 2022-02-24:
Figured I'd check back to see if there were any updates.  I *still* have this problem only when switching back to my MacBook Pro.  When I switch to my Windows machine, no problemo it just works.  On my Mac, I have to wait for roughly 15sec before the keyboard becomes usable.  However, the the trackpoint & keycluster modules are instantly available, I just can't type.  This is getting increasingly frustrating.  I picked up the USB switch so I wouldn't have to physically move the connector from machine to machine to reduce "wear & tear".  I'm still on MacOS BigSur 11.6.4 (20G417).

kareltucek, 2022-02-24:
@mikhey so have you tried the above "usb-fix" firmware?

mikhey, 2022-02-24:
Yep, just had to wait for a length of time (roughly 8 secs) before I could type this reply.  What's strange is it's only on my Mac that I have this issue. 

Connection to my MBP:
MacBook Pro -> OWC Thunderbolt 3 Dock -> Dell Monitor USB3 Hub -> Switch -> UHK

```
USB 3.1 Bus:

  Host Controller Driver:	AppleUSBXHCIAR
  PCI Device ID:	0x15d4 
  PCI Revision ID:	0x0002 
  PCI Vendor ID:	0x8086 

Hub:

  Product ID:	0x8440
  Vendor ID:	0x0451  (Texas Instruments)
  Version:	1.10
  Speed:	Up to 5 Gb/s
  Location ID:	0x01300000 / 1
  Current Available (mA):	900
  Current Required (mA):	0
  Extra Operating Current (mA):	0

Hub:

  Product ID:	0x8442
  Vendor ID:	0x0451  (Texas Instruments)
  Version:	1.00
  Serial Number:	FD030841612F
  Speed:	Up to 480 Mb/s
  Location ID:	0x01100000 / 2
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0

HD Webcam C615:

  Product ID:	0x082c
  Vendor ID:	0x046d  (Logitech Inc.)
  Version:	0.11
  Serial Number:	4D9ECBA0
  Speed:	Up to 480 Mb/s
  Location ID:	0x01120000 / 3
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

4-Port USB 2.0 Hub:

  Product ID:	0x5411
  Vendor ID:	0x0bda  (Realtek Semiconductor Corp.)
  Version:	1.21
  Speed:	Up to 480 Mb/s
  Manufacturer:	Generic
  Location ID:	0x01110000 / 36
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0

UHK 60 v1:

  Product ID:	0x6122
  Vendor ID:	0x1d50
  Version:	1.02
  Speed:	Up to 12 Mb/s
  Manufacturer:	Ultimate Gadget Laboratories
  Location ID:	0x01111000 / 37
  Current Available (mA):	500
  Current Required (mA):	150
  Extra Operating Current (mA):	0

Composite Device:

  Product ID:	0x82ff
  Vendor ID:	0x0451  (Texas Instruments)
  Version:	2.00
  Speed:	Up to 480 Mb/s
  Location ID:	0x01150000 / 5
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0

```

mondalaci, 2022-02-24:
@mikhey Please test the [uhk-firmware-9.0.0-5afd245.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/8136621/uhk-firmware-9.0.0-5afd245.tar.gz) firmware.

mikhey, 2022-02-24:
SWEET!! That did the trick!  Thank you so much!  It hasn't been a big deal the last few weeks, as I haven't had to switch around so much, but now I'm back to doing that more regularly.  I'm glad I checked in!

mikhey, 2022-03-05:
Just wanted to follow up.  I'm wondering if this may be a symptom of the 4P4C RJ11 going bad.  Today, I connected my laptop to my dock which is connected to pretty much everything else, and the left side refused to work.  I attempted to reload the firmware but to no avail (until I removed the cord and physically put the sides together w/o the modules).  

I picked up a temporary replacement cord and it seems to be working.  I just wanted to put that out there as a possibility, I'm planning on rolling back to the 8.0 firmware at some point to put my newfound theory to the test.

mondalaci, 2022-03-06:
@mikhey Your UHK may not receive enough power via your dock. Please disable its backlighting and try to reproduce the issue.

In any case, your issue is not related to this one, so any further problems, please send us an email to support@ultimatehackingkeyboard.com

kyoshiro, 2022-06-20:
> @mikhey Please test the [uhk-firmware-9.0.0-5afd245.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/8136621/uhk-firmware-9.0.0-5afd245.tar.gz) firmware.

This firmware works for me on MacOS 12 to reduce the cpu usage from rnb 17% to 0.9% (com.apple.AppleUserHIDDrivers)

PR #432 (2021-12-21): Attempt to fix #390
Fixes #390.

I was set out to implement my [suggestion](https://github.com/UltimateHackingKeyboard/firmware/issues/390#issuecomment-991797166), but on the way I found several other issues that I think should be fixed. I myself don't have any way to verify the functionality, I can only be certain that this branch builds. I can also imagine that removing the SET_IDLE support isn't even necessary to get the issue fixed, that's why that change is the last commit.