UltimateHackingKeyboard/firmware #391: Add more layers

mondalaci, 2021-08-04:
This issue is the firmware side of the [Add more layers Agent issue](https://github.com/UltimateHackingKeyboard/agent/issues/617).

Currently, it is possible to add 8 more layers to CurrentKeymap, but not 9 because then we run out of RAM. But adding 8 more layers would not be possible by simply increasing the layer count with Karel's soon-to-be-merged firmware fork because it uses additional RAM.

I think the cleanest solution would be to restructure CurrentKeymap as

```
#define KeyCountPerModule_RightKeyboardHalf 35
#define KeyCountPerModule_LeftKeyboardHalf 35
#define KeyCountPerModule_LeftModule 6
#define KeyCountPerModule_RightModule 3

key_action_t KeymapSlot_RightKeyboardHalf[LayerId_Count][KeyCountPerModule_RightKeyboardHalf];
key_action_t KeymapSlot_LeftKeyboardHalf[LayerId_Count][KeyCountPerModule_LeftKeyboardHalf];
key_action_t KeymapSlot_LeftModule[LayerId_Count][KeyCountPerModule_LeftModule];
key_action_t KeymapSlot_RightModule[LayerId_Count][KeyCountPerModule_RightModule];

key_action_t CurrentKeymap[SLOT_COUNT] = {
    KeymapSlot_RightKeyboardHalf,
    KeymapSlot_LeftKeyboardHalf,
    KeymapSlot_LeftModule,
    KeymapSlot_RightModule,
};
```

The compiler throws an error for the above code so it's surely incorrect, but you get the idea. If possible, it'd be great to have an array of variable-sized arrays, but I'm not sure if it's possible, and my C is not good enough to implement this off the top of my head.

kareltucek, 2021-08-04:
It might suffice to just change the 64 to 35 for all modules.

Setting  MAX_KEY_COUNT_PER_MODULE  =  35 and layer count to 12 on the fork yields:

```
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:       65552 B       463 KB     13.83%
          m_data:       64760 B        64 KB     98.82%
        m_data_2:       41920 B      65280 B     64.22%
        m_noinit:          16 B        255 B      6.27%
```

I am using 1024 bytes for the status buffer and 4096 bytes for runtime macros, so there is some 3 KB reserve in case we desperately need it.

--------------------------------

> but I'm not sure if it's possible

I don't think so. `KeymapSlot_RightKeyboardHalf` is a pointer type, hence the compiler errors. Still if you make it an array of pointers, C will treat them as one-dimensional arrays.

I guess you would have to construct manually a two dimensional array of pointers (pointing to the third-dimension arrays) manually.

mondalaci, 2022-03-19:
This is already implemented in master and will be included in the next release.

closed 2022-03-19 by mondalaci