UltimateHackingKeyboard/firmware #392: Slow scrolling on key-cluster breaks scrolling

m42e, 2021-08-06:
Having the Key cluster and settings set to Mac Defaults, scrolling slowly using the small trackball on the key cluster does stop working after some seconds.
After hitting another key or moving the trackball on the right hand side module it works again, as expected.

Let me know if I can add some more useful information

closed 2021-08-06 by m42e

mondalaci, 2021-08-06:
Is this still a valid issue? If not, how did you fix it?

m42e, 2021-08-06:
Well i really only use scrolling in the browser in a jupyter environment, and I tried it on both laptops. I have rushed into creating this ticket, as it turned out it occurs only in the jupyter environment. 

So in conclusion I think it is not a valid ticket.