UltimateHackingKeyboard/firmware #398: Touchpad drag and drop

mondalaci, 2021-08-14:
Most touchpads support drag and drop via double-tapping and holding their surface. I'll try to implement the same for the touchpad module, but according to my memory, the touchpad controller IC doesn't drag and drop the standard way, and its drag and drop implementation is rather clumsy. Also, it's questionable how useful drag and drop would be for the touchpad module since its surface area is rather small.

linduxed, 2021-09-11:
> Also, it's questionable how useful drag and drop would be for the touchpad module since its surface area is rather small.

While I agree that long distance dragging will likely be rather difficult with the touchpad, I would argue that there's one reason for supporting it: highlighting a couple of words in text.

I can only speak for myself, but with most other touchpads supporting the double-tap-then-hold feature, *muscle memory* gets disrupted when one can't double-tap then hold to highlight a region of text.

I don't mean to influence your project priorities with this comment, just wanted to give a reason for why I think it would be worth implementing.

nilehmann, 2021-10-08:
I'm trying the touchpad module now for the first time, and I'm really missing double-tap-then-hold feature for the exact same reason @linduxed mentioned: highlighting text.

matthewgrossman, 2021-11-11:
I've also seen in some amount of mice a "hold setting" thats lets you 
- hold mod+shift, or some keyboard combo
- moving the mouse will begin dragging whatever is under the cursor 
- you stay in "dragging mode" until you release the keyboard combo

This fixes the problem around the mousepad being too small, as you can re-center your thumb and continue dragging elsewhere

mondalaci, 2021-11-11:
@matthewgrossman It's already possible to hold Mouse+F according to the default keymap to achieve similar behavior.

matthewgrossman, 2021-11-11:
ah excellent, thanks! This is preferable to me than the double-tap fwiw, sorry for the noise

mondalaci, 2022-09-04:
Done in master and will be included in the next firmware release.

closed 2022-09-04 by mondalaci