UltimateHackingKeyboard/firmware #399: Inaccurate module attached state

mondalaci, 2021-08-15:
When attaching or detaching a module, it can temporarily or permanently affect the attached/detached state of other modules, including the left keyboard half. According to my experience, only the key cluster module is permanently affected sometimes. This bug is easy to notice in Agent, and with get-device-state.ts

kareltucek, 2022-07-03:
Is this still relevant?

mondalaci, 2022-07-03:
Yes, it is. There's a minor and major issue.

The minor issue is that when attaching or detaching a module, other modules or the left keyboard half often momentarily disappears.

The major issue is that when detaching the key cluster, sometimes it stays attached in Agent.

These issues are reproducible not only with Agent, but with `agent/packages/usb/get-device-state.ts`, too.

mondalaci, 2022-07-04:
Major issue beautifully solved.

As for the minor issue, the I2C bus is not designed to be hot-swappable. When attaching or detaching a module, it likely interferes with the actual communication of the bus and makes other modules disconnect as a side-effect. As a workaround, maybe the updated connection state of the modules affected by hot-swapping shouldn't be exposed via USB for a short period, like about half a second, only afterward.

On a related note, it may be worth checking the various error counters of modules, such as `arbitrationLost` and `timeout` by running `watch agentDir/packages/usb/get-i2c-health.ts` to more explicitly detect which modules are affected by the above side-effect.

kareltucek, 2022-07-04:
Still, it should be a basic decency for the KSDK library to report failed transactions flawlessly.

Unless we want to actually figure out cause of the issue, I would rather not clutter the code by workarounds as long as there is not a strong reason to do so.

mondalaci, 2022-07-05:
Does the KSDK not report failed transactions with `arbitrationLost` or `timeout` status when the minor issue happens?

In any case, I strongly feel that the user experience should be as good as possible, and flashing parts of the UHK unrelated to the currently attached/unattached modules is disturbing and doesn't instill great confidence in the product.

Please dig deeper if necessary, and let's fix this.

kareltucek, 2022-07-05:
> Does the KSDK not report failed transactions with arbitrationLost or timeout status when the minor issue happens?

Not sure, but the schedulerCallback chain has to be interrupted somewhere.

Admittedly, I see no reason why it should be assumed that a transfer is scheduled when KSDK returns an error right away.

kareltucek, 2022-07-05:
Well, I see, KSDK indeed returns arbitration lost right away, but the scheduler can't do anything with it.

> In any case, I strongly feel that the user experience should be as good as possible, and flashing parts of the UHK unrelated to the currently attached/unattached modules is disturbing and doesn't instill great confidence in the product.

Why should a brief disconnect during a hotswap affect flashing?

mondalaci, 2022-11-21:
@kareltucek Can you fix the minor issue part? Clearly not a showstopper, but it'd be nice if flashing couldn't be seen.

kareltucek, 2022-11-21:
I don't think this is a priority at the moment...

PR #644 (2023-07-20): Make fw report a module as connected if it was connected in last 500ms.
Closes #399.
closed 2023-07-20 by mondalaci