UltimateHackingKeyboard/firmware #400: Mini Trackball accidental scroll

juneidysoo, 2021-08-21:
There is slight issue I have experienced with the key cluster module in regards to the mini trackball.

Sometimes as I was typing some keys, a slight scroll was registered. I have a feeling the vibration of typing the key caused the mini trackball sensor to register it as a scroll movement.

This is especially annoying when I type command in my terminal because scroll would go back/forth through the command history and that can be dangerous.

Video of how I reproduced it: https://nextcloud.awesomest.me/s/XZSTEGeSr3Jxydg/download

From my experience, the issue is more likely to happen moment after I have used the mini trackball.

mondalaci, 2021-08-21:
@kareltucek Haven't you implemented a countermeasure for this issue? Maybe we've just talked about it.

In any case, I can only reproduce this when intentionally trying, and it hasn't been reported yet. I think this issue is highly dependent on the vibration of various setups, so it'd probably be wise to leave the default behavior as it is to not reduce mini trackball resolution artificially, and it'd be worth exposing a relevant variable via smart macros.

kareltucek, 2021-08-21:
> @kareltucek Haven't you implemented a countermeasure for this issue? Maybe we've just talked about it.

At the moment, ticking and axis locking is only implemented in the caret/media mode.

mondalaci, 2021-08-21:
What do you mean by ticking?

kareltucek, 2021-08-21:
I mean the thing you've just talked about :)

I.e., the condition that two hardware ticks should be required to trigger first event, and then proceed 1:1. Generally the mechanism that should define how hardware ticks map to events we send.

mondalaci, 2021-08-21:
Gotcha! :)

Do you have a specific behavior on your mind? For example, requiring two subsequent ticks in the same direction for an event in a given timeframe, then only requiring one tick per event until timing out.

kareltucek, 2021-08-21:
> Gotcha! :)
> 
> Do you have a specific behavior on your mind? For example, requiring two subsequent ticks in the same direction for an event in a given timeframe, then only requiring one tick per event until timing out.

I think that makes sense.

However, I am not sure about how the thing should be incomporated. I don't like the idea of duplicating current code (for caret/media modes), but current code is not sufficiently abstracted atm, and is quite hard to separate due to local variables. Another part is that i am not sure whether axis locking should be applied to scrolling (my opinion is: probably not). 

So... implementation wise, I will probably first deal with the missing variables of smart macros. That should simplify the matter. And then think about this again. Anyway, this ticket is "sort of" duplicate of #359.

closed 2021-09-06 by juneidysoo

mondalaci, 2021-09-06:
The root cause of this issue was hardware-related, so a related firmware feature may be unjustified.

wbolster, 2022-02-21:
i also experience accidental scrolls on the left keyboard half due to the mini trackball detecting movement. even when typing normally (not slamming) this happens regularly, up to the point that i sometimes remove the key cluster altogether out of annoyance.

it gets especially annoying when this happens when using the control key: many applications zoom in/out using ctrl-mousewheel, so simply typing a control makes the whole screen messed up.

this issue is closed, but has has anything be done about this? or can i do something about it myself?

one thing that would be a perfectly fine solution is to ignore any mini trackball movements in the short timeframe, e.g. 250ms, after registering a key press (or release) event.

i am running the latest firmware (8.10.12).

wbolster, 2022-02-21:
oh btw i can reproduce this without pressing any keys: tapping the frame or palm rest without actually pressing any keys makes web browsers scroll up and down randomly

kareltucek, 2022-02-21:
Yes, the proposed (firmware-based) solution was implemented in the mentioned PR, which was just merged independently. 

The solution has just not made it yet into a release. Feel free to take a look at [kareltucek/firmware](https://github.com/kareltucek/firmware) - it features the relevant code, and I *think* it should be enabled by default.

-----------------

Still, if the problem is as severe as you claim, then it is probably the hardware-related issue and it is advisable to shoot an email to laszlo (https://ultimatehackingkeyboard.com/contact-us).

wbolster, 2022-02-21:
thanks 🙏🏼 i guess i will just wait it out for a while.

tapping the casing or palm rests doesn't happen accidentally, while typing is normal operation and can perhaps be addressed purely in software/firmware indeed